fsaris/EspHome-AwoX-BLE-mesh-hub #4: MQTT disconnecting every few seconds

ItsEcholot, 2023-02-21:
Hello

This probably isn't an issue with this component but still wanted to ask if someone experiences the same issue.
MQTT seems to reconnect every few seconds which then obviously leads to the Light entity being unavailable for a short amount of time:

```
[13:36:45][W][mqtt:264]: MQTT Disconnected: TCP disconnected.
[13:36:45][I][mqtt:176]: Connecting to MQTT...
[13:36:45][I][mqtt:216]: MQTT Connected!
[13:37:27][W][mqtt:264]: MQTT Disconnected: TCP disconnected.
[13:37:27][I][mqtt:176]: Connecting to MQTT...
[13:37:28][I][mqtt:216]: MQTT Connected!
```

ItsEcholot, 2023-02-21:
Could very well be.
Changing the ESPHome configuration to the following BLE settings (reducing BLE scan window) has improved things a tiny bit but it still is happening although less regularly.
So who knows if this actually helps or is just coincidentally.
```
esp32_ble_tracker:
  scan_parameters:
    window: 15ms
    interval: 320ms
    active: false
```

I have the model `EGLO-RGB-TW-IPSU` and the ID is I think `41`? But not sure.

fsaris, 2023-02-21:
Did you enable the debug sensor?

Disabling them also makes the mqtt more stable.

Remove at least
```
device:
      name: "Device Info"
```

ItsEcholot, 2023-02-21:
No, I disabled most of the debug stuff, for future reference this is my current config:

```yaml
esphome:
  name: esphome-awox
  friendly_name: esphome-awox

esp32:
  board: esp32dev
  framework:
    type: arduino
    version: dev

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "---redacted---"

ota:
  password: "---redacted---"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: "---redacted---"
  output_power: 14db
    
esp32_ble_tracker:
  scan_parameters:
    window: 15ms
    interval: 320ms
    active: false

awox_mesh:
  mesh_name: !secret awox_mesh_name
  mesh_password: !secret awox_mesh_password

external_components:
  source: github://fsaris/EspHome-AwoX-BLE-mesh-hub

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_user
  password: !secret mqtt_password
  log_topic:

fsaris, 2023-02-26:
I disabled some devices in device_info.h to reduce the memory use

ItsEcholot, 2023-02-27:
Thanks, I updated my esp32 and tried your changes.
Sadly it doesn't seem to change the disconnects frequency, it still looses MQTT connection regularly.
Did it improve things for you?

michalpulda, 2023-05-04:
Hello,

I was also experiencing constant MQTT disconnections, so I commented out pretty much everything I found absolutely necessary for my use case. After commenting out HA API, it works very well.

Here's my YAML config:

```
substitutions:
  name: "esp32u-ble-awox"
  friendly_name: "ESP32U BLE Awox"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"

esp32:
  board: esp32dev
  framework:
    type: arduino
    # Set to fixed version due to build issues https://community.home-assistant.io/t/esphome-could-not-find-package-json/499026/8
    version: 2.0.6 #https://github.com/espressif/arduino-esp32/releases

wifi:
  ssid: !secret
  password: !secret
  use_address: !secret

  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "${friendly_name} hotspot"
#    password: !secret

#captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
#api:
#  encryption:
#    key: !secret

ota:
  password: !secret

mqtt:
  broker: !secret
  username: !secret
  password: !secret

external_components:
  source: github://fsaris/EspHome-AwoX-BLE-mesh-hub
  # - source:
  #     type: local
  #     path: components

esp32_ble_tracker:
  scan_parameters:
    active: false

awox_mesh:
  mesh_name: !secret
  mesh_password: !secret

switch:
  - platform: restart
    name: "${friendly_name} Restart"

binary_sensor:
  - platform: status
    name: "${friendly_name} Status" 
```

Running it on ESP32U with external antenna and it seems to work more reliably compared to the original HA integration.

Thanks!

fsaris, 2023-05-05:
Ah, it was probably the `api` config. That seems to conflict with mqtt.
Glad it works now

closed 2023-05-05 by fsaris

ItsEcholot, 2023-05-05:
Can confirm, removing the `api` section has fixed this for me as well.
Unbelievable I never thought of trying this myself, thanks!