fsaris/EspHome-AwoX-BLE-mesh-hub #5: Howto use this custom component

fsaris, 2023-02-26:
Dear Frans,
first of all, thanks for your efforts. Also, I have an Olimex ESP32-POE-EA device that I wanted to use as a bluetooth proxy especially for EGLO lamps. However, at the moment I have not been able to establish a connection with your old integration, but I would be happy to help you test the MQTT version. I tried to flash it but couldn't really get past the compilation. How could I help you? 

Thanks in advance!

_Originally posted by @4D4M-Github in https://github.com/fsaris/home-assistant-awox/issues/67#issuecomment-1437646073_

fsaris, 2023-02-26:
@4D4M-Github where did you exactly get stuck?

1. could you build the code on your device?
2. Did you find the mesh credentials (see the README.md)
3. did you get some logging in the esphome console output?

4D4M-Github, 2023-03-28:
Hi Frans,

sorry for the late reply, had a massive HA crash

1.: Yes, I have since managed to translate
2.: Yes, I have managed to find it
3.: Yes:

```
INFO Successfully compiled program.
INFO Resolving IP address of awox-ble-mesh-hub.local
INFO  -> 192.168.50.206
INFO Uploading /data/awox-ble-mesh-hub/.pioenvs/awox-ble-mesh-hub/firmware.bin (1442256 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from awox-ble-mesh-hub.local using esphome API
WARNING Can't connect to ESPHome API for awox-ble-mesh-hub.local: Error resolving IP address: [Errno -5] No address associated with hostname
INFO Trying to reconnect to awox-ble-mesh-hub.local in the background
INFO Successfully connected to awox-ble-mesh-hub.local
[15:57:39][I][app:102]: ESPHome version 2023.3.2 compiled on Mar 28 2023, 15:55:23
[15:57:39][C][logger:293]: Logger:
[15:57:39][C][logger:294]:   Level: DEBUG
[15:57:39][C][logger:295]:   Log Baud Rate: 115200
[15:57:39][C][logger:296]:   Hardware UART: UART0
[15:57:39][C][esp32_ble:214]: ESP32 BLE:
[15:57:39][C][esp32_ble_tracker:591]: BLE Tracker:
[15:57:39][C][esp32_ble_tracker:592]:   Scan Duration: 300 s
[15:57:39][C][esp32_ble_tracker:593]:   Scan Interval: 320.0 ms
[15:57:39][C][esp32_ble_tracker:594]:   Scan Window: 30.0 ms
[15:57:39][C][esp32_ble_tracker:595]:   Scan Type: ACTIVE
[15:57:39][C][esp32_ble_tracker:596]:   Continuous Scanning: True
[15:57:39][C][ethernet:166]: Ethernet:
[15:57:39][C][ethernet:284]:   IP Address: 192.168.50.206
[15:57:39][C][ethernet:285]:   Hostname: 'awox-ble-mesh-hub'
[15:57:39][C][ethernet:286]:   Subnet: 255.255.255.0
[15:57:39][C][ethernet:287]:   Gateway: 192.168.50.1
[15:57:39][C][ethernet:292]:   DNS1: 1.1.1.1
[15:57:39][C][ethernet:293]:   DNS2: 192.168.50.1
[15:57:39][C][ethernet:300]:   MAC Address: 24:4C:AB:02:33:0F
[15:57:39][C][ethernet:305]:   Is Full Duplex: YES
[15:57:39][C][ethernet:310]:   Link Speed: 100
[15:57:39][C][ethernet:169]:   Power Pin: 12
[15:57:39][C][ethernet:171]:   MDC Pin: 23
[15:57:39][C][ethernet:172]:   MDIO Pin: 18
[15:57:39][C][ethernet:173]:   Type: LAN8720
[15:57:39][C][mdns:108]: mDNS:
[15:57:39][C][mdns:109]:   Hostname: awox-ble-mesh-hub
[15:57:39][C][ota:093]: Over-The-Air Updates:
[15:57:39][C][ota:094]:   Address: awox-ble-mesh-hub.local:3232
[15:57:39][C][ota:097]:   Using Password.
[15:57:39][C][api:138]: API Server:
[15:57:39][C][api:139]:   Address: awox-ble-mesh-hub.local:6053
[15:57:39][C][api:141]:   Using noise encryption: YES
[15:57:39][C][mqtt:065]: MQTT:
[15:57:39][C][mqtt:067]:   Server Address: 192.168.50.5:1883 (192.168.50.5)
[15:57:39][C][mqtt:068]:   Username: 'mqttclient'[redacted]
[15:57:39][C][mqtt:069]:   Client ID: 'awox-ble-mesh-hub-244cab02330c'[redacted]
[15:57:39][C][mqtt:071]:   Discovery prefix: 'homeassistant'
[15:57:39][C][mqtt:072]:   Discovery retain: YES
[15:57:39][C][mqtt:074]:   Topic Prefix: 'awox-ble-mesh-hub'
[15:57:39][C][mqtt:076]:   Log Topic: 'awox-ble-mesh-hub/debug'
[15:57:39][C][mqtt:079]:   Availability: 'awox-ble-mesh-hub/status'
[15:57:53][D][AwoxMesh:057]: Total devices: 3
[15:57:53][D][AwoxMesh:059]: Available device A4:C1:38:B7:0D:81 => rssi: -53
[15:57:53][D][AwoxMesh:059]: Available device A4:C1:38:72:7D:BD => rssi: -58
[15:57:53][D][AwoxMesh:059]: Available device A4:C1:38:A3:DB:2E => rssi: -78
[15:57:53][I][AwoxMesh:063]: Try to connect A4:C1:38:B7:0D:81 => rssi: -53
[... 20 lines cut ...]
[15:58:31][I][esp32_ble_client:064]: [0] [A4:C1:38:B7:0D:81] 0x00 Attempting BLE connection
[15:58:31][I][MeshDevice:173]: CONNECTING
[15:58:32][I][esp32_ble_client:196]: [0] [A4:C1:38:B7:0D:81] Connected
[15:58:32][I][MeshDevice:177]: CONNECTED
[15:58:32][I][mesh_device:151]: Connected....
[15:58:32][D][mesh_device:249]: combine mesh name + password: [redacted]
[15:58:32][D][mesh_device:235]: Listen for notifications
[15:58:32][D][mesh_device:242]: Enable notifications
[15:58:32][W][esp32_ble_client:215]: [0] [A4:C1:38:B7:0D:81] esp_ble_gattc_get_descr_by_char_handle error, status=10
[15:58:32][D][esp32_ble_tracker:246]: Starting scan...
[15:58:32][D][AwoxMesh:045]: scan end
[15:58:32][E][mesh_device:195]: Device authentication error: known mesh credentials are not excepted by the device. Did you re-pair them to your Awox app with a different account?
[15:58:32][I][mesh_device:201]: [0] [A4:C1:38:B7:0D:81] response 00001110
[15:58:32][I][esp32_ble_client:081]: [0] [A4:C1:38:B7:0D:81] Disconnecting.
[15:58:32][I][MeshDevice:153]: DISCONNECTING
[15:58:32][D][AwoxMesh:057]: Total devices: 3
[15:58:32][D][AwoxMesh:059]: Available device A4:C1:38:B7:0D:81 => rssi: -49
[15:58:32][D][AwoxMesh:059]: Available device A4:C1:38:72:7D:BD => rssi: -61
[15:58:32][D][AwoxMesh:059]: Available device A4:C1:38:A3:DB:2E => rssi: -75
[15:58:32][I][AwoxMesh:063]: Try to connect A4:C1:38:B7:0D:81 => rssi: -49
[15:58:32][I][esp32_ble_client:064]: [0] [A4:C1:38:B7:0D:81] 0x00 Attempting BLE connection
[15:58:32][I][MeshDevice:173]: CONNECTING
```
and so on.
The mesh credentials are correct.

oxigen-dev, 2023-04-12:
Hi @fsaris ,

First, thanks for working on this!

Unfortunately I was not able to connect to the lamps (I've tried with 2 EGLO lamps). I'm not sure whether I have the same issue as @4D4M-Github. Could you maybe check it? Thanks!

Here you can find the logs:

```
INFO Reading configuration /config/egloconfig.yaml...
INFO Starting log output from eglo.local using esphome API
INFO Successfully connected to eglo.local
[22:44:38][I][app:102]: ESPHome version 2023.3.2 compiled on Apr 12 2023, 20:27:41
[22:44:38][C][wifi:504]: WiFi:
[22:44:38][C][wifi:362]:   Local MAC: [redacted]
[22:44:38][C][wifi:363]:   SSID: [redacted]
[22:44:38][C][wifi:364]:   IP Address: 192.168.1.115
[22:44:38][C][wifi:366]:   BSSID: [redacted]
[22:44:38][C][wifi:367]:   Hostname: 'eglo'
[22:44:39][C][wifi:369]:   Signal strength: -64 dB ▂▄▆█
[22:44:39][C][wifi:373]:   Channel: 5
[22:44:39][C][wifi:374]:   Subnet: 255.255.255.0
[22:44:39][C][wifi:375]:   Gateway: 192.168.1.1
[22:44:39][C][wifi:376]:   DNS1: 192.168.1.1
[22:44:39][C][wifi:377]:   DNS2: 0.0.0.0
[22:44:39][C][logger:293]: Logger:
[22:44:39][C][logger:294]:   Level: DEBUG
[22:44:39][C][logger:295]:   Log Baud Rate: 115200
[22:44:39][C][logger:296]:   Hardware UART: UART0
[22:44:39][C][esp32_ble:214]: ESP32 BLE:
[22:44:39][C][esp32_ble_tracker:591]: BLE Tracker:
[22:44:39][C][esp32_ble_tracker:592]:   Scan Duration: 300 s
[22:44:39][C][esp32_ble_tracker:593]:   Scan Interval: 320.0 ms
[22:44:39][C][esp32_ble_tracker:594]:   Scan Window: 30.0 ms
[22:44:39][C][esp32_ble_tracker:595]:   Scan Type: PASSIVE
[22:44:39][C][esp32_ble_tracker:596]:   Continuous Scanning: True
[22:44:39][C][captive_portal:088]: Captive Portal:
[22:44:39][C][mdns:108]: mDNS:
[22:44:39][C][mdns:109]:   Hostname: eglo
[22:44:39][C][ota:093]: Over-The-Air Updates:
[22:44:39][C][ota:094]:   Address: eglo.local:3232
[22:44:39][C][api:138]: API Server:
[22:44:39][C][api:139]:   Address: eglo.local:6053
[22:44:39][C][api:143]:   Using noise encryption: NO
[22:44:39][C][mqtt:065]: MQTT:
[22:44:39][C][mqtt:067]:   Server Address: 192.168.1.102:1883 (192.168.1.102)
[22:44:39][C][mqtt:068]:   Username: [redacted]
[22:44:39][C][mqtt:069]:   Client ID: [redacted]
[22:44:39][C][mqtt:071]:   Discovery prefix: 'homeassistant'
[22:44:39][C][mqtt:072]:   Discovery retain: YES
[22:44:39][C][mqtt:074]:   Topic Prefix: 'eglo'
[22:44:39][C][mqtt:076]:   Log Topic: 'eglo/debug'
[22:44:39][C][mqtt:079]:   Availability: 'eglo/status'
[22:44:50][I][AwoxMesh:071]: Failed to connect XX:XX:XX:XX:XX:XX => rssi: -58
[22:44:50][I][esp32_ble_client:081]: [0] [XX:XX:XX:XX:XX:XX] Disconnecting.
[22:44:50][I][MeshDevice:153]: DISCONNECTING
[22:44:50][D][AwoxMesh:057]: Total devices: 1
[22:44:50][D][AwoxMesh:059]: Available device XX:XX:XX:XX:XX:XX => rssi: -57
[22:44:50][I][AwoxMesh:063]: Try to connect XX:XX:XX:XX:XX:XX => rssi: -57
[22:44:50][I][esp32_ble_client:064]: [0] [XX:XX:XX:XX:XX:XX] 0x00 Attempting BLE connection
[22:44:50][I][MeshDevice:173]: CONNECTING
[22:45:00][I][MeshDevice:157]: IDLE
[22:45:00][D][mesh_device:142]: [0] [XX:XX:XX:XX:XX:XX] ESP_GATTC_DISCONNECT_EVT, reason 256
[22:45:00][W][esp32_ble_client:134]: [0] [] Connection failed, status=133
[22:45:00][I][MeshDevice:157]: IDLE
[22:45:00][D][AwoxMesh:057]: Total devices: 1
[22:45:00][D][AwoxMesh:059]: Available device XX:XX:XX:XX:XX:XX => rssi: -54
[22:45:00][I][AwoxMesh:063]: Try to connect XX:XX:XX:XX:XX:XX => rssi: -54
[22:45:00][I][esp32_ble_client:064]: [0] [XX:XX:XX:XX:XX:XX] 0x00 Attempting BLE connection
[22:45:00][I][MeshDevice:173]: CONNECTING
[... 107 lines cut ...]
[22:48:00][I][AwoxMesh:063]: Try to connect XX:XX:XX:XX:XX:XX => rssi: -57
[22:48:00][I][esp32_ble_client:064]: [0] [XX:XX:XX:XX:XX:XX] 0x00 Attempting BLE connection
[22:48:00][I][MeshDevice:173]: CONNECTING
[22:48:20][I][AwoxMesh:071]: Failed to connect XX:XX:XX:XX:XX:XX => rssi: -57
[22:48:20][I][esp32_ble_client:081]: [0] [XX:XX:XX:XX:XX:XX] Disconnecting.
[22:48:20][I][MeshDevice:153]: DISCONNECTING
[22:48:20][D][AwoxMesh:057]: Total devices: 1
[22:48:20][D][AwoxMesh:059]: Available device YY:YY:YY:YY:YY:YY => rssi: -87
[22:48:20][I][AwoxMesh:063]: Try to connect YY:YY:YY:YY:YY:YY => rssi: -87
[22:48:20][I][esp32_ble_client:064]: [0] [YY:YY:YY:YY:YY:YY] 0x00 Attempting BLE connection
[22:48:20][I][MeshDevice:173]: CONNECTING
[22:48:30][I][MeshDevice:157]: IDLE
[22:48:30][D][mesh_device:142]: [0] [YY:YY:YY:YY:YY:YY] ESP_GATTC_DISCONNECT_EVT, reason 256
[22:48:30][W][esp32_ble_client:134]: [0] [] Connection failed, status=133
[22:48:30][I][MeshDevice:157]: IDLE
[22:48:30][D][AwoxMesh:057]: Total devices: 1
[22:48:30][D][AwoxMesh:059]: Available device YY:YY:YY:YY:YY:YY => rssi: -87
[22:48:30][I][AwoxMesh:063]: Try to connect YY:YY:YY:YY:YY:YY => rssi: -87
[22:48:30][I][esp32_ble_client:064]: [0] [YY:YY:YY:YY:YY:YY] 0x00 Attempting BLE connection
[22:48:30][I][MeshDevice:173]: CONNECTING
[22:48:50][I][AwoxMesh:071]: Failed to connect YY:YY:YY:YY:YY:YY => rssi: -87
[22:48:50][I][esp32_ble_client:081]: [0] [YY:YY:YY:YY:YY:YY] Disconnecting.
[22:48:50][I][MeshDevice:153]: DISCONNECTING
[22:49:00][I][MeshDevice:157]: IDLE
```

fsaris, 2023-04-13:
@oxigen-dev could it be that you are out of reach of the lights? The scanner finds 2 devices but doesn't look powerful enough to connect and that could be due to the distance between esp and device

4D4M-Github, 2023-04-13:
@fsaris One of my lights is 4 meters from the esp without any obstruction. I don't think it's quite the same problem, because I have a problem with the authentication. If I change the password I get the same result.

oxigen-dev, 2023-04-13:
> @oxigen-dev could it be that you are out of reach of the lights? The scanner finds 2 devices but doesn't look powerful enough to connect and that could be due to the distance between esp and device

@fsaris The ESP was in the same room with one of the ceiling lamps, let's say only 2 meters away. The other one was 10 meters away so this could be an issue with the second one.

Should I create a new issue as my problem seems to be a different one?

Thanks!

oxigen-dev, 2023-04-14:
OK I've bought a different device, flashed the software and this issue didn't appear :). Some of my lamps are now an unknown device, some seems to be still available for hours although they are switched off etc. but for a first step I'm happy with the current situation :)

closed 2023-08-08 by fsaris