fsaris/EspHome-AwoX-BLE-mesh-hub #6: Build fails when framework is espidf

gCode72, 2023-03-01:
Hello
The attached config file builds and runs on the shelly 1 plus pretty well, without the add of the awox line in the end.

When adding the awox part, taken from https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/blob/main/awox-ble-mesh-hub.yaml, the the build crashes.

See ERROR_1 below.

When I replace the board framework type to arduino, then it's building, but of corse the image doesn't run on the shelly.

This happens on an iMac 10.15.7.

I tried also in esphome addon in HA and on a macbook. There I got this error in build.

See ERROR_2 below.

Is board framework type esp-idf not supported, or am I doing something wrong?

ERROR_1:
esphome run shelly_1_plus_bt_mqtt.yaml
INFO Reading configuration shelly_1_plus_bt_mqtt.yaml...
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO0 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing shelly-plus-1 (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 5.2.0)

HARDWARE: ESP32 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /Volumes/DatenHD/Giampiero/Documents/espHome/.esphome/build/shelly-plus-1/sdkconfig.shelly-plus-1
-- App "shelly-plus-1" version: 2023.2.4
-- Adding linker script /Volumes/DatenHD/Giampiero/Documents/espHome/.esphome/build/shelly-plus-1/.pioenvs/shelly-plus-1/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/soc/esp32/ld/esp32.peripherals.ld
-- Components: __pio_env app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs src tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/__pio_env /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/app_trace /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/app_update /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/asio /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/bootloader /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/bootloader_support /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/bt /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/cbor /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/cmock /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/coap /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/console /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/cxx /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/driver /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/efuse /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp-tls /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp32 /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_adc_cal /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_common /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_eth /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_event /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_gdbstub /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_hid /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_http_client /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_http_server /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_https_ota /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_https_server /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_hw_support /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_ipc /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_lcd /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_local_ctrl /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_netif /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_phy /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_pm /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_ringbuf /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_rom /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_serial_slave_link /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_system /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_timer /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_websocket_client /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esp_wifi /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/espcoredump /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/esptool_py /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/expat /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/fatfs /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/freemodbus /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/freertos /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/hal /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/heap /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/idf_test /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/ieee802154 /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/jsmn /Volumes/DatenHD/Giampiero/.platformio/packages/framework-espidf/components/json /Volum [... cut ...]
* Web  > https://registry.platformio.org/search?q=header:AES.h
*
*************************************************************

 #include <AES.h>
          ^~~~~~~
compilation terminated.
*** [/data/shelly-plus-1/.pioenvs/shelly-plus-1/src/esphome/components/awox_mesh/mesh_device.o] Error 1
========================= [FAILED] Took 129.03 seconds =========================

[shelly_1_plus_bt_mqtt_1.txt](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/10864621/shelly_1_plus_bt_mqtt_1.txt)

fsaris, 2023-03-01:
I only tested with the Arduino platform because then the crypto (AES) dependency is correctly resolved. 

Could be that it's something missing in my component config but didn't yet tried to get it running on esp-idf.

It uses the same crypto dependency as https://esphome.io/components/sensor/dsmr.html not sure if that works with esp-idf.

Could be a starting point to get it to work

fsaris, 2023-03-01:
If I find some time I will look if I can use mbedtls/aes.h

Like in https://github.com/MarkEvens/AES-Encryption-ESP-IDF/blob/main/main/main.c

gCode72, 2023-03-04:
Hi Frans, that would be very kind. Thanks!

gCode72, 2023-04-23:
> If I find some time I will look if I can use mbedtls/aes.h
> 
> Like in https://github.com/MarkEvens/AES-Encryption-ESP-IDF/blob/main/main/main.c

Hi there, any news on this?

fsaris, 2023-04-23:
Sorry not yet

fsaris, 2023-08-20:
@gCode72 could you test the dev branch? The arduino requirement for encryption is removed.

You can use this in the YAML to install the dev version:

```
external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub
      ref: dev
```

gCode72, 2023-08-21:
Hi Frans,
Thanks for working on that. 
I got still build errors. Here is the log. I think my config file is correct.

Log:
ESPHome 2023.5.4
INFO Reading configuration shelly_eingang_plus_bt.yaml...
INFO Cloning https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub@dev
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO0 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing shelly-plus-1-eingang (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.3.0)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
HARDWARE: ESP32 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40404.0 (4.4.4) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Generating assembly for certificate bundle...
Dependency Graph
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/adc/adc_sensor.o
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/api/api_connection.o
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/api/api_pb2.o
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/api/api_pb2_service.o
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/api/api_server.o
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/api/list_entities.o
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/api/proto.o
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/api/subscribe_state.o
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/api/user_services.o
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/awox_mesh/awox_mesh.o
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/awox_mesh/mesh_device.o
src/esphome/components/awox_mesh/awox_mesh.cpp:1:9: warning: #pragma once in main file
 #pragma once
         ^~~~
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/binary_sensor/automation.o
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/binary_sensor/binary_sensor.o
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/binary_sensor/filter.o
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/bluetooth_proxy/bluetooth_connection.o
src/esphome/components/awox_mesh/mesh_device.cpp: In member function 'virtual bool esphome::awox_mesh::MeshDevice::gattc_event_handler(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':
src/esphome/components/awox_mesh/mesh_device.cpp:139:10: error: enumeration value 'ESP_GATTC_REG_EVT' not handled in switch [-Werror=switch]
   switch (event) {
[... 69 lines cut ...]
   sprintf(value, "%02X%02X%02X", hex1, hex2, hex3);
   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.o
src/esphome/components/awox_mesh/mesh_device.cpp: In member function 'void esphome::awox_mesh::MeshDevice::handle_packet(std::__cxx11::string&)':
src/esphome/components/awox_mesh/mesh_device.cpp:72:18: error: 'sprintf' writing a terminating nul past the end of the destination [-Werror=format-overflow=]
   sprintf(value, "A4:C1:%02X:%02X:%02X:%02X", part3, part4, part5, part6);
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/awox_mesh/mesh_device.cpp:72:10: note: 'sprintf' output 18 bytes into a destination of size 17
   sprintf(value, "A4:C1:%02X:%02X:%02X:%02X", part3, part4, part5, part6);
   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from src/esphome/components/awox_mesh/mesh_device.h:8,
                 from src/esphome/components/awox_mesh/mesh_device.cpp:6:
src/esphome/core/log.h:114:18: error: 'mesh_id' may be used uninitialized in this function [-Werror=maybe-uninitialized]
[... cut ...]
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/gpio/output/gpio_binary_output.o
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/json/json_util.o
Compiling .pioenvs/shelly-plus-1-eingang/src/esphome/components/logger/logger.o
cc1plus: some warnings being treated as errors
*** [.pioenvs/shelly-plus-1-eingang/src/esphome/components/awox_mesh/mesh_device.o] Error 1
================================================================================ [FAILED] Took 25.29 seconds ================================================================================

fsaris, 2023-08-23:
@gCode72 having the device functioning as switch, Bluetooth proxy and mesh hub will probably be to much. 

It's known that the `api:` and `mqtt:` can result in rebooting of the device due to memory issues when running the mesh hub.

But that isn't the reason it doesn't build at all :)

gCode72, 2023-08-23:
Thanks for the hints. 
I will take that into account!

fsaris, 2023-08-23:
@gCode72 got it running now with framework `esp-idf` 

Only logging is not working properly when connected by cable but that's probably some missing config on my side. Logging over wifi works properly

gCode72, 2023-08-24:
Hi Frans,
It's compiling now, great job!
Although I could not get it run on the device. After removing the api config entry (left only mqtt config) the device crashes with error about logger missing (sorry, did not save the log).
Have to get if out of the wall first and flashed it by cable before I can retry.

Thanks so far!!

michalpulda, 2023-09-13:
Hello,
I have compiled the firmware using the `esp-idf` framework and it works well. I can't really say if there is any difference comparing it to the original `arduino` framework. They say the `esp-idf` framework should work better/faster with bluetooth, though. Will come back in a few days to let you know.
Thanks!

gCode72, 2023-09-14:
Hello @michalpulda,
Great to hear that. Thank you for contributing! 
You mean, you have flashed your esp32 device, and it works? What kind of device do you use? 
I tried to flash a Shelly 1 Plus. But I can't really get it to work. Permanent reboot. At that point, I'am not sure if the Shelly is capable for all that.
Could you maybe share your config file?

Regards!

michalpulda, 2023-09-15:
Hi @gCode72, I'm using regular esp32 devkit (vroom). The config file is rather simple, nothing fancy:

```
substitutions:
  name: "esp32u-ble-awox"
  friendly_name: "ESP32U BLE Awox"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"

esp32:
  board: esp32dev
  framework:
    #type: arduino
    type: esp-idf

wifi:
  ssid: !secret wifi-ssid
  password: !secret wifi-pass
  use_address: 192.168.xxx.xxx

logger:

ota:
  password: !secret ota-pass

mqtt:
  broker: 192.168.xxx.xxx
  username: !secret mqtt_user
  password: !secret mqtt_password

external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub
      ref: dev # for esp-idf testing

esp32_ble_tracker:
  scan_parameters:
    active: false

awox_mesh:
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password
  address_prefix: A4:C1:38:56:C1:CD # bulb right next to the ESP32
  device_info:
  -  product_id: 0x49
     device_type: DIM
     name: Bulb A60 Warm
     model: ESMLm_w9w
     manufacturer: EGLO
     icon: mdi:lightbulb-outline
  -  product_id: 0x33
     device_type: RGB
     name: Spot 170
     model: EGLOSpot 170/w
     manufacturer: EGLO
     icon: mdi:light-recessed
  -  product_id: 0x32
     device_type: RGB
     name: Spot 120
     model: EGLOSpot 120/w
     manufacturer: EGLO
     icon: mdi:light-recessed
  -  product_id: 0x51
     device_type: RGB
     name: TriSpot 85
     model: ETriSpot_85
     manufacturer: EGLO
     icon: mdi:light-recessed
  -  product_id: 0x85
     device_type: PLUG
     name: EGLO PLUG PLUS
     model: SMPWBm10GEa
     manufacturer: EGLO
     icon: mdi:power-socket-de

switch:
  - platform: restart
    name: "${friendly_name} Restart"

binary_sensor:
  - platform: status
    name: "${friendly_name} Status"
```

dreyks, 2023-09-24:
can confirm, managed to build and flash using the config above. my device was already using esp-idf for other stuff and adding awox nearly doubled the firmware size. it took almost all the flash space of my Lilygo stick and took me a couple of tries to push it through the timeouts while doing OTA, but now everything works 👍

gCode72, 2023-09-27:
Hello everyone,
I finally also get it work. Now it's working pretty good. Light become unavailable from time to time in HA.

But for me the issue regarding esp-idf framework is fixed.

Again, great work @fsaris.
Thanks all who have contributed testing.

Regards!

fsaris, 2023-09-28:
Code is merged to the main branch

closed 2023-09-28 by fsaris