fsaris/EspHome-AwoX-BLE-mesh-hub #9: Add new device

oxigen-dev, 2023-04-14:
Hi @fsaris ,

Unfortunately 3 of my devices were not recognized, could you please add them? I've pasted the MQTT payloads. For all of the lamps one can set the brightness, color, temperature. In case you need any additional information, please let me know. Thanks!

```
schema: json
name: Unknown device type
unique_id: awox-XX:XX:XX:XX:XX:XX-light
icon: mdi:lightbulb-help-outline
state_topic: awox-ble-mesh-hub/22980/state
command_topic: awox-ble-mesh-hub/22980/command
availability:
  - topic: awox-ble-mesh-hub/22980/availability
  - topic: awox-ble-mesh-hub/status
availability_mode: all
color_mode: true
brightness: true
brightness_scale: 255
supported_color_modes:
  - brightness
device:
  identifiers:
    - esp-awox-mesh-22980
    - XX:XX:XX:XX:XX:XX
  name: Unknown device type
  model: t
  manufacturer: AwoX
  via_device: ------
platform: mqtt
```

```
schema: json
name: Unknown device type
unique_id: awox-YY:YY:YY:YY:YY:YY-light
icon: mdi:lightbulb-help-outline
state_topic: awox-ble-mesh-hub/3590/state
command_topic: awox-ble-mesh-hub/3590/command
availability:
  - topic: awox-ble-mesh-hub/3590/availability
  - topic: awox-ble-mesh-hub/status
availability_mode: all
color_mode: true
brightness: true
brightness_scale: 255
supported_color_modes:
  - brightness
device:
  identifiers:
    - esp-awox-mesh-3590
    - YY:YY:YY:YY:YY:YY
  name: Unknown device type
  model: t
  manufacturer: AwoX
  via_device: -----------
platform: mqtt
```

```
schema: json
name: Unknown device type
unique_id: awox-ZZ:ZZ:ZZ:ZZ:ZZ:ZZ-light
icon: mdi:lightbulb-help-outline
state_topic: awox-ble-mesh-hub/23338/state
command_topic: awox-ble-mesh-hub/23338/command
availability:
  - topic: awox-ble-mesh-hub/23338/availability
  - topic: awox-ble-mesh-hub/status
availability_mode: all
color_mode: true
brightness: true
brightness_scale: 255
supported_color_modes:
  - brightness
device:
  identifiers:
    - esp-awox-mesh-23338
    - ZZ:ZZ:ZZ:ZZ:ZZ:ZZ
  name: Unknown device type
  model: t
  manufacturer: AwoX
  via_device: -------
platform: mqtt
```

fsaris, 2023-04-27:
Looks like some info is missing in the message you posted.

The model value should show a product id.

It should also be visible in the UI on the device details screen

oxigen-dev, 2023-04-28:
> Looks like some info is missing in the message you posted.
> 
> The model value should show a product id.
> 
> It should also be visible in the UI on the device details screen

@fsaris  Ooops I only have this...

[image]

Any ideas? :)

MineTech13, 2023-05-04:
Have the same Issue my EPanel 600 (HW1.0) FW2.2.6 is also just registering by the model "t"

voxain, 2023-05-06:
Same thing here. Got some of the Fueva-C models

fsaris, 2023-05-07:
Can you check the log in ESPHome en look for productID

voxain, 2023-05-07:
I somehow can't get this to log anymore, but the product ID was 0x40 - It's a square Fueva-C / EGLO Surface. Below I have attached the MQTT discovery log if that is helpful (Replaced the MAC Address just to be safe).
```yaml
schema: json
name: Unknown device type
unique_id: awox-A4:C1:38:XX:XX:XX-light
icon: mdi:lightbulb-help-outline
state_topic: esphome-web-305a2c/19134/state
command_topic: esphome-web-305a2c/19134/command
availability:
  - topic: esphome-web-305a2c/19134/availability
  - topic: esphome-web-305a2c/status
availability_mode: all
color_mode: true
brightness: true
brightness_scale: 255
supported_color_modes:
  - brightness
device:
  identifiers:
    - esp-awox-mesh-19134
    - A4:C1:38:XX:XX:XX
  name: Unknown device type
  model: t
  manufacturer: AwoX
  via_device: 30c6f7305a2c
platform: mqtt
```

oxigen-dev, 2023-05-09:
> Can you check the log in ESPHome en look for productID

Hi @fsaris ,

With this idea I was able to get the product ids. You can find a PR here: #16 

Did you have a chance to check this (#12)? Thanks!

fsaris, 2023-05-09:
> > Can you check the log in ESPHome en look for productID
> 
> Hi @fsaris ,
> 
> With this idea I was able to get the product ids. You can find a PR here: #16
👍🏼 

> Did you have a chance to check this (#12)? Thanks!

Not yet, but it's one of the next things on my list

voxain, 2023-05-09:
What's the process of adding a product in the list? Do I have to find out anything that needs to be added? Because in #16 it just looks like @oxigen-dev uncommented the line of the product - I'm guessing it's not that easy?

oxigen-dev, 2023-05-09:
> What's the process of adding a product in the list? Do I have to find out anything that needs to be added? Because in #16 it just looks like @oxigen-dev uncommented the line of the product - I'm guessing it's not that easy?

You need to get the product id from the esp log, then search in the file (what was in my PR) for your product id. If it is there, you only need to uncomment it :). I've personally forked the repo, did the modification, tested it with my lamps and then I've created the PR.

voxain, 2023-05-09:
Well that is fantastic news, I will have to try tomorrow. Was a little bummed to see RGB not work - though any particular reason they're not simply all uncommented already? Since they've already been "discovered".

voxain, 2023-05-09:
That sounds wonderful. You are really out here doing god's work, I can't believe that theres so little open-source projects for EGLO lights since they're about the only affordable competition to Philips Hue. 
Thank you for your great work 🙏