fsaris/EspHome-AwoX-BLE-mesh-hub #11: Eglo Plug Plus is always unknown

Dillton, 2023-04-19:
Hello @fsaris its been some time.

First let me thank you for this new solution to control Awox devices.
The first feel is very good in compare of old Awox integration so very good job. :)

Now to the problem I have Eglo Plug Plus device which worked fine in old integration but with ESP the plug is recognized but always in unknown state.
[image]

I have checked your list of supported/activated devices and should be this one so should work just fine:
this->add_device(DEVICE_TYPE_PLUG, 0x85, "EGLO PLUG PLUS", "SMPWBm10GEa", MANUFACTURER_EGLO, "mdi:power-socket-de");

Do you have any idea what could be wrong?
I can provide also some logs if you could guide me a bit, I am newbie to ESP devices :)

FYI:
I was forced by HAOS yesterday update to migrate to this new solution because bluetooth in HAOS 10 is completely broken.

fsaris, 2023-04-20:
Can you please provide the logs from ESPHome. That should tell us more

Dillton, 2023-04-20:
Hi, I hope this is the log you need.
[logs_awox-ble-mesh-hub_logs.txt](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/11284590/logs_awox-ble-mesh-hub_logs.txt)

fsaris, 2023-04-27:
Looks like the mqtt client is reconnecting constantly. Could point to a memory issue or connection issues.

Do you have only this component running on the esp?

Dillton, 2023-04-27:
Yes only Awox hub is running on this ESP module.

Regarding the mqtt reconnecting I think I fixed this by using latest arduino framework and there was also bigger ESPHome update recently. I can't see any constant reconnection in logs now maybe once per hour or so.

Anyway I don't think this is related to the problem as only the plug is in unknown state but all 8 lights working like a charm.

Dillton, 2023-04-28:
Hi, there is also MQTT info for the device if its somehow helpful.
EGLO PLUG PLUS (switch.eglo_plug_plus)
MQTT discovery data:
Topic: homeassistant/switch/awox-A4C13821E89A/config
Payload
schema: json
name: EGLO PLUG PLUS
unique_id: awox-A4:C1:38:21:E8:9A-switch
icon: mdi:power-socket-de
state_topic: awox-ble-mesh-hub/26778/state
command_topic: awox-ble-mesh-hub/26778/command
availability:
  - topic: awox-ble-mesh-hub/26778/availability
  - topic: awox-ble-mesh-hub/status
availability_mode: all
color_mode: true
supported_color_modes:
  - onoff
device:
  identifiers:
    - esp-awox-mesh-26778
    - A4:C1:38:21:E8:9A
  name: EGLO PLUG PLUS
  model: SMPWBm10GEa
  manufacturer: EGLO
  via_device: 0cb815c3aed0
platform: mqtt
Subscribed topics:
awox-ble-mesh-hub/26778/availability
1 most recently received message(s)
Received 15:29:35
QoS: 0, Retained
Payload: online
awox-ble-mesh-hub/status
1 most recently received message(s)
Received 15:29:35
QoS: 0, Retained
Payload: online
awox-ble-mesh-hub/26778/state
1 most recently received message(s)
Received 15:29:35
QoS: 0, Retained
Payload: state: 'OFF'
color_mode: color_temp
brightness: 0
color_temp: 153
color:
  r: 0
  g: 0
  b: 0

michalpulda, 2023-05-04:
Hello, I'm also experiencing the same issue. As mentioned in [another issue](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/issues/4#issuecomment-1534603320), I don't have problem with MQTT disconnections and it works pretty solid so far.

Please see bellow VERBOSE logs. Turning off and on the plug showed following erorrs:
```
[14:14:50][E][json:101]: JSON parse error: InvalidInput
...
[14:15:14][E][json:101]: JSON parse error: InvalidInput
```

[logs_esp32u-ble-awox_logs.txt](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/11397340/logs_esp32u-ble-awox_logs.txt)

Thanks!

fsaris, 2023-05-05:
Do you have the `api:` enabled? If so please give it a try with the api disabled

michalpulda, 2023-05-05:
> Do you have the `api:` enabled? If so please give it a try with the api disabled

I have the `api:` disabled.

Input type for the switch is not what I would expect to be for plug (see the image bellow). It's two _buttons_ and not a _switch_. Isn't the problem connected with this?

[image]

fsaris, 2023-05-05:
This is the state when HA doesn't now yet if the device is on or off

michalpulda, 2023-05-05:
> This is the state when HA doesn't now yet if the device is on or off

I didn't know that, thank you. The plug works fine with the original integration, though. Is there any way I could help any further? Thanks!

fsaris, 2023-05-07:
> Hello, I'm also experiencing the same issue. As mentioned in [another issue](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/issues/4#issuecomment-1534603320), I don't have problem with MQTT [...]

Do you maybe also have the mqtt message logs of this failing device (especially the outgoing)? I don't have a plug myself so I'm not able to reproduce.

michalpulda, 2023-05-08:
Hello,

First, this is how it looks like when calling ON and OFF to a spot light that works well:

[image]
[image]

Now, when I do the same for the plug:

[image]

The only difference I can see is that the call is in different format:

[image]
vs.
[image]

Hope this helps!

(For my future reference: this is a prompt I used via the SSH addon: `mosquitto_sub -h host_address -u user_name -P password -d -v -t '#'`)

fsaris, 2023-05-08:
ah, that log helped. I pushed a update. That should make it possible to control the plug

michalpulda, 2023-05-09:
Works great, thank you!

michalpulda, 2023-05-12:
@fsaris After updating to the last version (based on [this issue](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/issues/15)), the plug doesn't work anymore. Looking at the log, it show exactly the same as before:
 
> [image]
> 
> The only difference I can see is that the call is in different format:
> 
> [image] vs. [image]

fsaris, 2023-05-12:
Please add the needed device info to your yaml and it should work again

michalpulda, 2023-05-20:
Done and it's working. Thank you!