fsaris/EspHome-AwoX-BLE-mesh-hub #12: Devices are shown as Online even after turned off

oxigen-dev, 2023-04-23:
Hi @fsaris ,

After some days I've realized that most of the time if I turn off a lamp it stays Online in HA. As it is Online I can _change the brightness_ but meanwhile nothing appears in the log. Could you please check it? Thanks!

```
[07:52:15][D][esp32_ble_tracker:246]: Starting scan...
[07:52:15][D][AwoxMesh:045]: scan end
[07:52:45][D][esp32_ble_tracker:246]: Starting scan...
[07:52:45][D][AwoxMesh:045]: scan end
[07:53:15][D][esp32_ble_tracker:246]: Starting scan...
[07:53:15][D][AwoxMesh:045]: scan end
```

fsaris, 2023-04-27:
Looks like it isn't able to find your devices. It should report all devices it can connect to

oxigen-dev, 2023-04-27:
> Looks like it isn't able to find your devices. It should report all devices it can connect to 

@fsaris In the example above, none of my lamps were turned on. This is what we can see in the logs. Besides this, I had some lamps Online in HA. It seems that there was no MQTT message when I've turned off my lamps. Do you think the handling of this can be improved?

Thanks!

fsaris, 2023-04-28:
Oh, non of the lights on. Mmm, that's a little tricky because now it uses the messages in the mesh to determine if one of the lights is offline. But if there is no mesh at all that's a little different.
I didn't want to mark the lights as offline when not connected to the mesh because it could be a temporary hickup. But maybe it should mark all as offline when it couldn't connect to the mesh anymore

oxigen-dev, 2023-04-28:
> Oh, non of the lights on. Mmm, that's a little tricky because now it uses the messages in the mesh to determine if one of the lights is offline. But if there is no mesh at all that's a little [...]

This sounds like a good idea! Would you be so kind to do the modification? I'd be happy to test it! Thanks!

fsaris, 2023-05-09:
@oxigen-dev can you test latest version? Should set the devices to "not-available" when lights are all offline

oxigen-dev, 2023-05-09:
> @oxigen-dev can you test latest version? Should set the devices to "not-available" when lights are all offline

Sure I'll test it then give feedback. Thanks!

oxigen-dev, 2023-05-11:
It seems to be fine until now :). I'll give feedback again in a few days.

oxigen-dev, 2023-05-11:
@fsaris Now I was having a lamp which was turned off for 30 mins but still seems to be Online in HA :(

fsaris, 2023-05-11:
@oxigen-dev can you provide some logs from esphome? Do the other lights still respond?

fsaris, 2023-05-11:
@oxigen-dev if you enable the light again for a few seconds and turn it off again will then report as offline in HA?

oxigen-dev, 2023-05-11:
> @oxigen-dev if you enable the light again for a few seconds and turn it off again will then report as offline in HA?

@fsaris I've turned it on for a minute then turned it off and now it is showed correctly as offline.

oxigen-dev, 2023-05-11:
> @oxigen-dev can you provide some logs from esphome? Do the other lights still respond?

After this issue, I've turned on a different light and it was working well.
If necessary, I'll figure it out how I can log to a file then I can share the content. As I took a look in the log via WiFi, nothing was there, only some information about the working lamp.

oxigen-dev, 2023-05-12:
Today's experience: if multiple lamps were turned on and I've turned off only some of them, then with some unkown probability they went offline in HA, some of them stayed online. As all of the lamps were turned off and the mesh was not available, all of them went offline. Unfortunately I don't have any logs for this. Maybe next time :)

oxigen-dev, 2023-08-08:
Hi @fsaris ,

I still have this issue, could you please reopen? Since my last comment I've tried to use a different device with external antenna but this didn't help much. I've also tried to create some logs but the issue didn't appear then. Maybe I'll do the logging for some days.

fsaris, 2023-08-09:
@oxigen-dev okay reopened ticket, please provide some logs

oxigen-dev, 2023-08-11:
> @oxigen-dev okay reopened ticket, please provide some logs

Hi @fsaris ,

I've managed to create some logs (~6.5 hours). I have multiple EGLO ceilings (7 were used while creating the logs) and 1 Eglo Smart Bulb.
Please note that there is a 2 hours difference between the timestamp in the ESPHome log and the timestamp in Home Assistant (ESPHome+2h=HA).

In today's test most of the lamps availability were showed correctly, except for the Bathroom Lamp.
What I can see in HA:
[image]
So the last time it became Online was 7:59:32 PM. I'm sure that at 8:13 PM it was already turned off.
Additional information to that is that normally this Bathroom lamp is not in range for the ESP (not even with the one I'm currently using with an external antenna) and it can be only reached through a MESH network using the lamp in the Hall.

This situation with my other lamps are usually the same (some of them are showed Online even they are switched off) but today the only issue I've had was with the Bathroom ceiling, others were working fine. I'll leave the logger running in case you need some additional logs.

<details> 
  <summary>OPEN THE LOG</summary>

```
[32mINFO ESPHome 2023.7.1[0m
[32mINFO Reading configuration awox-ble-mesh-hub.yaml...[0m
[32mINFO Starting log output from /dev/ttyUSB1 with baud rate 115200[0m
[15:07:34][D][sensor:094]: 'Uptime Sensor': Sending state 570.34998 s with 0 decimals of accuracy
[15:08:07][D][esp32_ble_tracker:245]: Starting scan...
[15:08:07][D][AwoxMesh:045]: scan end
[15:08:34][D][sensor:094]: 'Uptime Sensor': Sending state 630.34998 s with 0 decimals of accuracy
[15:09:34][D][sensor:094]: 'Uptime Sensor': Sending state 690.34998 s with 0 decimals of accuracy
[15:10:34][D][sensor:094]: 'Uptime Sensor': Sending state 750.35699 s with 0 decimals of accuracy
[15:11:34][D][sensor:094]: 'Uptime Sensor': Sending state 810.34998 s with 0 decimals of accuracy
[15:11:35][D][AwoxMesh:057]: Total devices: 1
[15:11:35][D][AwoxMesh:059]: Available device BATHROOM => rssi: -86
[15:11:35][I][AwoxMesh:063]: Try to connect BATHROOM => rssi: -86
[15:11:35][I][mesh_device:475]: Publish connected to mesh device - offline
[15:11:35][I][esp32_ble_client:064]: [0] [BATHROOM] 0x00 Attempting BLE connection
[15:11:35][I][MeshDevice:175]: CONNECTING
[15:11:55][I][AwoxMesh:071]: Failed to connect BATHROOM => rssi: -86
[15:11:55][I][esp32_ble_client:081]: [0] [BATHROOM] Disconnecting.
[15:11:55][I][MeshDevice:155]: DISCONNECTING
[15:11:55][I][mesh_device:475]: Publish connected to mesh device - offline
[15:11:56][D][AwoxMesh:057]: Total devices: 1
[15:11:56][D][AwoxMesh:059]: Available device BATHROOM => rssi: -87
[15:11:56][I][AwoxMesh:063]: Try to connect BATHROOM => rssi: -87
[15:11:56][I][mesh_device:475]: Publish connected to mesh device - offline
[15:11:56][I][esp32_ble_client:064]: [0] [BATHROOM] 0x00 Attempting BLE connection
[15:11:56][I][MeshDevice:175]: CONNECTING
[15:12:05][I][MeshDevice:159]: IDLE
[15:12:05][D][mesh_device:144]: [0] [BATHROOM] ESP_GATTC_DISCONNECT_EVT, reason 0
[15:12:05][W][esp32_ble_client:134]: [0] [BATHROOM] Connection failed, status=133
[15:12:05][I][MeshDevice:159]: IDLE
[15:12:13][D][esp32_ble_client:048]: [0] [BATHROOM] Found device
[15:12:13][I][MeshDevice:167]: DISCOVERED
[15:12:13][D][esp32_ble_tracker:213]: Pausing scan to make connection...
[15:12:13][I][MeshDevice:171]: READY_TO_CONNECT
[15:12:13][I][esp32_ble_client:064]: [0] [BATHROOM] 0x00 Attempting BLE connection
[15:12:13][I][MeshDevice:175]: CONNECTING
[15:12:14]lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
[15:12:14]lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
[15:12:14]lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
[15:12:15]lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
[15:12:15][I][MeshDevice:159]: IDLE
[15:12:15][D][mesh_device:144]: [0] [BATHROOM] ESP_GATTC_DISCONNECT_EVT, reason 62
[15:12:15][I][mesh_device:475]: Publish connected to mesh device - offline
[15:12:15][W][esp32_ble_client:134]: [0] [] Connection failed, status=133
[15:12:15][I][MeshDevice:159]: IDLE
[15:12:15][D][esp32_ble_tracker:245]: Starting scan...
[15:12:15][D][AwoxMesh:045]: scan end
[15:12:15][D][AwoxMesh:057]: Total devices: 1
[15:12:15][D][AwoxMesh:059]: Available device BATHROOM => rssi: -89
[15:12:15][I][AwoxMesh:063]: Try to connect BATHROOM => rssi: -89
[15:12:15][I][mesh_device:475]: Publish connected to mesh device - offline
[... 1973 lines cut ...]
[21:05:34][D][sensor:094]: 'Uptime Sensor': Sending state 22050.34961 s with 0 decimals of accuracy
[21:06:34][D][sensor:094]: 'Uptime Sensor': Sending state 22110.34961 s with 0 decimals of accuracy
[21:07:34][D][sensor:094]: 'Uptime Sensor': Sending state 22170.34961 s with 0 decimals of accuracy
[21:08:34][D][sensor:094]: 'Uptime Sensor': Sending state 22230.35156 s with 0 decimals of accuracy
[21:09:33][D][esp32_ble_tracker:245]: Starting scan...
[21:09:33][D][AwoxMesh:045]: scan end
[21:09:34][D][sensor:094]: 'Uptime Sensor': Sending state 22290.35547 s with 0 decimals of accuracy
[21:10:34][D][sensor:094]: 'Uptime Sensor': Sending state 22350.34961 s with 0 decimals of accuracy
[21:11:34][D][sensor:094]: 'Uptime Sensor': Sending state 22410.34961 s with 0 decimals of accuracy
[21:12:34][D][sensor:094]: 'Uptime Sensor': Sending state 22470.35156 s with 0 decimals of accuracy
[21:13:34][D][sensor:094]: 'Uptime Sensor': Sending state 22530.34961 s with 0 decimals of accuracy
[... cut ...]
[21:15:48][D][mesh_device:350]: online status report: mesh: 8479, on: 1, color_mode: 0, transition_mode: 0, w_b: 64, temp: 25, c_b: 50, rgb: B8A1FF 
[21:15:48][I][mesh_device:415]: 8479: ON temp: 25 (64 %) ONLINE
[21:15:48][D][mesh_device:464]: Delayed publish online/offline for 8479 - online
[21:15:51][I][mesh_device:469]: Publish online/offline for 8479 - online
[21:16:34][D][sensor:094]: 'Uptime Sensor': Sending state 22710.34961 s with 0 decimals of accuracy
[21:17:34][D][sensor:094]: 'Uptime Sensor': Sending state 22770.34961 s with 0 decimals of accuracy

``` 

</details>

oxigen-dev, 2023-08-14:
Hi @fsaris ,

I'd like to attach another log from yesterday. It is a bit long but I didn't want to truncate it.
Issue: at 12:28 AM (I think in the log it is now 21:37 at line 17156) Home Assistant shows that 4 lights are Online: LIVINGROOM_SERVING, HALL, OFFICE, BEDROOM. Actually the first two were switched off for some hours and even the Office ceiling was shown Online, it was not possible to do anything with it (I was trying to adjust the brightness, HA showed as it was successful, but not in real life).

Could you please check it? Thanks!
[awox_202308013.log](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/12339904/awox_202308013.log)

fsaris, 2023-08-15:
Can you also add the logs (screenshot of events for instance) from HA of one of the lights that wasn't responsive?

oxigen-dev, 2023-08-15:
> Can you also add the logs (screenshot of events for instance) from HA of one of the lights that wasn't responsive?

This is what I've found in HA's Logbook for the Office ceiling:
[image]

This is for the Hall:
[image]

fsaris, 2023-08-16:
when there is no mesh connection with one of the lights all  lights are reported as `offline` and that seems to be going okay.

When a connection is re-established the lights are reported online again and get previous state.  And here something looks off. the mesh receives on the same second 1 message that the light is offline and 1 that it's online. For instance:

```
[17:52:30][D][mesh_device:350]: online status report: mesh: 31615, on: 1, color_mode: 0, transition_mode: 0, w_b: 127, temp: 25, c_b: 100, rgb: 9EC2FF 
[17:52:30][I][mesh_device:415]: 31615: ON temp: 25 (127 %) OFFLINE!!
[17:52:30][D][mesh_device:464]: Delayed publish online/offline for 31615 - offline
[17:52:30][D][mesh_device:350]: online status report: mesh: 31615, on: 1, color_mode: 0, transition_mode: 0, w_b: 127, temp: 25, c_b: 100, rgb: 9EC2FF 
[17:52:30][I][mesh_device:415]: 31615: ON temp: 25 (127 %) ONLINE
[17:52:30][D][mesh_device:464]: Delayed publish online/offline for 31615 - online
[17:52:32][D][mesh_device:350]: online status report: mesh: 22980, on: 1, color_mode: 0, transition_mode: 0, w_b: 5, temp: 127, c_b: 3, rgb: FF0033 
[17:52:32][I][mesh_device:415]: 22980: ON temp: 127 (5 %) OFFLINE!!
[17:52:32][D][mesh_device:464]: Delayed publish online/offline for 22980 - offline
[17:52:32][D][mesh_device:350]: online status report: mesh: 22980, on: 1, color_mode: 0, transition_mode: 0, w_b: 5, temp: 127, c_b: 3, rgb: FF0033 
[17:52:32][I][mesh_device:415]: 22980: ON temp: 127 (5 %) ONLINE
[17:52:32][D][mesh_device:464]: Delayed publish online/offline for 22980 - online
[17:52:33][D][mesh_device:118]: Skipped publishing availability for 31615 - Offline (is currently Online)
[17:52:33][I][mesh_device:469]: Publish online/offline for 31615 - online
```

I need to check if there is some kind of message number/timestamp that could be used to be sure the correct message is processed last. But I would need to have the raw decoded message for that. And that's currently not in the logs. You will need to set logging to verbose for that

oxigen-dev, 2023-08-19:
@fsaris Here is a part of the log for today. 4 of my ceilings were turned on and shown Online in HA. I've turned off two of them (Couch, Dining). In HA all of them became Offline (marked with blue on the pic), then all of them became Online (red) then the two which were turned off in real life went Offline again (yellow)
[20230819_github.log](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/12386833/20230819_github.log)
.
[image]

oxigen-dev, 2023-08-19:
@fsaris This is a log from 17.08.2023. For some reason two of my ceilings were "randomly" shown in HA as Online then Offline at the very same time(!). In real life they weren't even turned on in that time range. None of the MAC addresses in the logs represents those two lamps.
I hope I didn't make a mistake choosing the right timestamp in the logs as I've seen ESPHome + 2h = HA.
I hope those logs can help a bit. If you need additional ones, just let me know.
[image]
[20230817_github.log](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/12386848/20230817_github.log)

oxigen-dev, 2023-09-05:
Hi @fsaris ,

Did you have a chance to check it? 😀

fsaris, 2023-09-06:
Not into details yet, sorry.

I found in some tech sheet that each message should hold the number of hobs it needed to reach the hub. Maybe this can be used to make it more reliable. And I was thinking of asking the mesh again for its state when messages are received on more or less the same time. This is what happens for you, in my mesh this also happens but somehow the correct state is always the last that is processed

oxigen-dev, 2023-09-06:
> Not into details yet, sorry.
>
> I found in some tech sheet that each message should hold the number of hobs it needed to reach the hub. Maybe this can be used to make it more reliable. And I was [...]

Sure, no problem. I'll wait. Thanks.

Dillton, 2023-09-19:
Hi @fsaris.
Regarding my yesterdays question issue.
I have done reflash of the ESP device around 8PM yesterday and since then not a single light disconnected.
I don't want to spread some misinformations but could this behaviour be related to something inside arduino-esp32?
Because I had this set to v2.0.6 but yesterday I tried to use the latest v2.0.13.
[image]

oxigen-dev, 2023-10-11:
Hi @fsaris ,

Did you have some free time to work on this? This is the last missing piece in my smart home :)

Thanks!

fsaris, 2023-10-17:
Did some attempt's last days to get a sequence number or something similar from the messages but wasn't able (yet) to find something.

Are you using the latest version of the code? There are reports that using the current version fixed issues for some see #29

oxigen-dev, 2023-10-22:
> Did some attempt's last days to get a sequence number or something similar from the messages but wasn't able (yet) to find something.
> 
> Are you using the latest version of the code? There are reports that using the current version fixed issues for some see #29

I was testing for a few days with the newest version and esp-idf, but the situation is the same. I think we would still need the functionality what you are working on.
Do you have a buymeacoffee registration or something like that? Of course I would be happy to support you in some way to say thanks and keep you motivated.

fsaris, 2023-10-25:
Hi, supporting me by buying me a coffee or new stuff to thinker with would indeed help with keeping motivated 😃

- https://www.buymeacoffee.com/franssaris
- https://paypal.me/franssaris/25

fsaris, 2023-11-19:
@oxigen-dev I added an additional status call when the online status changes. Can you check if this helps you with the wrong online/offline status?

oxigen-dev, 2023-11-19:
> @oxigen-dev I added an additional status call when the online status changes. Can you check if this helps you with the wrong online/offline status?

@fsaris Thanks, I've already reflashed my ESP. I'll test it for a few days then I'll give feedback.

oxigen-dev, 2023-11-21:
@fsaris A strange thing happened today evening. I was turning off 4 ceilings at the same time, 3 of them was shown as offline, the 4th one is now online. After physically turning it off it came back online in HA for some reasan. Could you please check it? Thanks!
[image]

fsaris, 2023-11-21:
Do you also have the logs from the esp?

oxigen-dev, 2023-11-21:
> Do you also have the logs from the esp?

Unfortunately not. Maybe it would be a good idea to log everything again for a few days. I was just hoping that based on the symptom you'll have some idea were the problem is :)

oxigen-dev, 2023-11-21:
@fsaris I've restarted my ESP and turned on logging. After restarting 1 bulb and 2 ceilings went to online (1 of the 2 ceilings is the one which is turned off physically). I was not able to find anything in the log.

`INFO ESPHome 2023.11.1
INFO Reading configuration awox-ble-mesh-hub.yaml...
INFO Updating https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub.git@None
INFO Detected timezone 'Etc/UTC'
INFO Starting log output from awox-ble-mesh-hub/debug
INFO Connected to MQTT broker!
[22:53:57][D][sensor:094]: 'Uptime Sensor': Sending state 1498.64197 s with 0 decimals of accuracy
[22:54:04][D][mesh_device:353]: online status report: mesh: 22980, on: 1, color_mode: 0, transition_mode: 0, w_b: 5, temp: 127, c_b: 3, rgb: FF0033 
[22:54:04][I][mesh_device:418]: 22980: ON temp: 127 (5 %) ONLINE
[22:54:04][D][mesh_device:467]: Delayed publish online/offline for 22980 - online
[22:54:07][I][mesh_device:475]: Publish online/offline for 22980 - online
[22:54:09][D][mesh_device:353]: online status report: mesh: 22980, on: 1, color_mode: 0, transition_mode: 0, w_b: 5, temp: 127, c_b: 3, rgb: FF0033 
[22:54:09][I][mesh_device:418]: 22980: ON temp: 127 (5 %) OFFLINE!!
[22:54:09][D][mesh_device:467]: Delayed publish online/offline for 22980 - offline
[22:54:12][I][mesh_device:475]: Publish online/offline for 22980 - offline
[22:54:16][D][esp32_ble_tracker:246]: Starting scan...
[22:54:16][D][AwoxMesh:042]: scan end
[22:54:54][D][sensor:094]: 'Uptime Sensor': Sending state 1558.64197 s with 0 decimals of accuracy
[22:54:57][D][switch:012]: 'Awox BLE Mesh Hub Restart' Turning ON.
[22:54:57][D][switch:055]: 'Awox BLE Mesh Hub Restart': Sending state OFF
[22:54:57][I][restart:016]: Restarting device...
[22:54:57][I][app:134]: Rebooting safely...
[22:54:57][D][esp-idf:000]: I (1562648) MQTT_CLIENT: Client asked to disconnect

[22:55:04][I][mqtt:282]: MQTT Connected!
[22:55:04][I][app:062]: setup() finished successfully!
[22:55:04][I][app:102]: ESPHome version 2023.9.3 compiled on Nov 19 2023, 16:59:15
[22:55:04][C][wifi:546]: WiFi:
[22:55:04][C][wifi:382]:   Local MAC: LOCAL_MAC
[22:55:04][C][wifi:383]:   SSID: 'WIFI SSID'
[22:55:04][C][wifi:384]:   IP Address: 192.168.1.58
[22:55:04][C][wifi:386]:   BSSID: BSSID
[22:55:04][C][wifi:387]:   Hostname: 'awox-ble-mesh-hub'
[22:55:04][C][wifi:389]:   Signal strength: -52 dB ▂▄▆█
[22:55:04][C][wifi:393]:   Channel: 1
[22:55:04][C][wifi:394]:   Subnet: 255.255.255.0
[22:55:04][C][wifi:395]:   Gateway: 192.168.1.1
[22:55:04][C][wifi:396]:   DNS1: 0.0.0.0
[22:55:04][C][wifi:397]:   DNS2: 0.0.0.0
[22:55:04][C][logger:357]: Logger:
[22:55:04][C][logger:358]:   Level: DEBUG
[22:55:04][C][logger:359]:   Log Baud Rate: 115200
[22:55:04][C][logger:361]:   Hardware UART: UART0
[22:55:04][C][uptime.sensor:031]: Uptime Sensor 'Uptime Sensor'
[22:55:04][C][uptime.sensor:031]:   Device Class: 'duration'
[22:55:04][C][uptime.sensor:031]:   State Class: 'total_increasing'
[22:55:04][C][uptime.sensor:031]:   Unit of Measurement: 's'
[22:55:04][C][uptime.sensor:031]:   Accuracy Decimals: 0
[22:55:04][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[22:55:04][C][homeassistant.time:010]: Home Assistant Time:
[22:55:04][C][homeassistant.time:011]:   Timezone: 'UTC0'
[22:55:04][C][bluetooth_proxy:088]: Bluetooth Proxy:
[22:55:04][C][bluetooth_proxy:089]:   Active: YES
[22:55:04][C][restart:068]: Restart Switch 'Awox BLE Mesh Hub Restart'
[22:55:04][C][restart:070]:   Icon: 'mdi:restart'
[22:55:04][C][restart:091]:   Restore Mode: always OFF
[22:55:04][C][esp32_ble:238]: ESP32 BLE:
[22:55:04][C][esp32_ble:240]:   MAC address: MAC_ADDRESS
[22:55:04][C][esp32_ble:241]:   IO Capability: none
[22:55:04][C][esp32_ble_tracker:617]: BLE Tracker:
[22:55:04][C][esp32_ble_tracker:618]:   Scan Duration: 300 s
[22:55:04][C][esp32_ble_tracker:619]:   Scan Interval: 320.0 ms
[22:55:04][C][esp32_ble_tracker:620]:   Scan Window: 30.0 ms
[22:55:04][C][esp32_ble_tracker:621]:   Scan Type: ACTIVE
[22:55:04][C][esp32_ble_tracker:622]:   Continuous Scanning: True
[22:55:04][C][mdns:115]: mDNS:
[22:55:04][C][mdns:116]:   Hostname: awox-ble-mesh-hub
[22:55:04][C][ota:097]: Over-The-Air Updates:
[... 69 lines cut ...]
[22:55:20][W][mesh_device:488]: '8479': Can not yet send publish state, mac address not known...
[22:55:20][W][component:204]: Component esp32_ble took a long time for an operation (0.07 s).
[22:55:20][W][component:205]: Components should block for at most 20-30ms.
[22:55:20][D][mesh_device:386]: MAC report, dev [8479]: productID: A9 mac: OFFICE_CEILING_MAC => [removed]
[22:55:20][D][mesh_device:525]: '8479': Sending discovery...
[22:55:20][D][mesh_device:467]: Delayed publish online/offline for 8479 - online
[22:55:20][D][mesh_device:374]: status report: mesh: 8479, on: 1, color_mode: 0, transition_mode: 0, w_b: 59, temp: 25, c_b: 46, rgb: B8A1FF 
[22:55:20][I][mesh_device:418]: 8479: ON temp: 25 (59 %) ONLINE
[22:55:20][D][mesh_device:374]: status report: mesh: 8479, on: 1, color_mode: 0, transition_mode: 0, w_b: 59, temp: 25, c_b: 46, rgb: B8A1FF 
[22:55:20][I][mesh_device:418]: 8479: ON temp: 25 (59 %) ONLINE
[22:55:23][I][mesh_device:475]: Publish online/offline for 8479 - online
[22:55:23][I][mesh_device:475]: Publish online/offline for 8479 - online
[22:55:25][D][mesh_device:123]: Request info again for 22980
[22:55:30][D][mesh_device:123]: Request info again for 22980
[22:55:35][D][mesh_device:123]: Request info again for 22980
[22:55:40][D][mesh_device:123]: Request info again for 22980
[... cut ...]
[22:55:45][D][mesh_device:123]: Request info again for 22980
[22:55:50][D][mesh_device:123]: Request info again for 22980
[22:55:52][D][mesh_device:353]: online status report: mesh: 22980, on: 1, color_mode: 0, transition_mode: 0, w_b: 5, temp: 127, c_b: 3, rgb: FF0033 
[22:55:52][I][mesh_device:418]: 22980: ON temp: 127 (5 %) ONLINE
[22:55:52][W][mesh_device:488]: '22980': Can not yet send publish state, mac address not known...
[22:55:52][D][mesh_device:467]: Delayed publish online/offline for 22980 - online
`
[image]

fsaris, 2023-12-03:
Okay, so the light that was physical offline came online in HA after reboot of the esp device? 

If I read the logs correctly it only found 2 devices where one looks to not reply to a Mac address request so it's maybe offline. Can you check the mqtt messages and match the meshid with lights name? That makes understanding the logs easier 😃. I'm wondering if `22980` is your `Dining Ceiling`.

Are the 2 devices which you redacted the Mac addresses in the logs in reach of each other?

oxigen-dev, 2023-12-04:
> Okay, so the light that was physical offline came online in HA after reboot of the esp device?

Yes, correct.

> If I read the logs correctly it only found 2 devices where one looks to not reply to a Mac address request so it's maybe offline. Can you check the mqtt messages and match the meshid with lights name? That makes understanding the logs easier 😃. I'm wondering if `22980` is your `Dining Ceiling`.

22980: Bedroom Bulb
8479: Office Ceiling
3969: Dining Ceiling

> Are the 2 devices which you redacted the Mac addresses in the logs in reach of each other?

Those 2 devices don't have a stable connection. Between the Bedroom Bulb and the Office Ceiling there is another room (Living room). Approx. 70% of the time I can't control the Office Ceiling with HA if the ESP is connected to the Bedroom bulb.

oxigen-dev, 2023-12-22:
Hi @fsaris,

Did you have some time to work on this? 😁

fsaris, 2023-12-23:
Was a bussy time the last week's.

It's on my list to pick this up again first week of the new year.

Have several ideas that should be able to help us here

oxigen-dev, 2023-12-23:
> Was a bussy time the last week's.
> 
> It's on my list to pick this up again first week of the new year.
> 
> Have several ideas that should be able to help us here 

@fsaris Very good news! I wish you a Merry XMas and a Happy New Year!

fsaris, 2024-01-01:
@oxigen-dev could you try latest dev branch version? After restart Initial all devices should be marked offline until they really get online.

(Just another small tryout to improve)

oxigen-dev, 2024-01-01:
> @oxigen-dev could you try latest dev branch version? After restart Initial all devices should be marked offline until they really get online.
> 
> (Just another small tryout to improve)

@fsaris Dev branch was flashed successfully, I'm already testing it. I'll give you a feedback after some time.
Ps. Is it possible to tell (preferably from the HA GUI) which device is the ESP connected to  directly? Thanks!

oxigen-dev, 2024-01-03:
@fsaris Unfortunately I had this issue again yesterday.

- Every day from 8PM until 8AM next day the bulb in the bedroom is turned on. We can assume that the ESP is always connected to this bulb (in the night).
- Yesterday evening the ceilings in the bathroom and in the office were shown as Online although both of them were turned off.
- The connection between the bulb and the bathroom ceiling is very poor, 99.9% of the time they can't build up a MESH network. This means the connection is working once in a week (best case).
- The connection between the bulb and the office ceiling is also not so good, 50% of the time it is not working.
- The ESP is in the middle of my home, it can connect to every ceiling/bulb if only 1 lamp is turned on.

I've only tried to describe the situation in my home because it might help finding the root cause (although if the bulb can't reach the ceilings they should appear as unavailable, the situation is the same as they would have been turned off). What is you opinion? Thanks!

fsaris, 2024-01-04:
I'm currently looking if I can let the esp connect to multiple devices and let it work as access-point to multiple devices at the same time instead of only limiting to 1 device as access-point of the mesh

fsaris, 2024-01-07:
@oxigen-dev after some days of work and a big rewrite as a result 😄 the component is now able to connect to up to 3 devices.

On the mqtt topic `awox-ble-mesh-hub/connection_status` (first part is the name you gave the esp in the YAML) a report can be found of the current connections (will create proper sensors for this). 

When building the current dev branch it could be you need to clear the build files (click "Clear Build Files" on the dashboard) due to file name changes.

I noticed that when viewing the ESPHome log it uses a lot of more memory, so when running into reboots try to avoid having the log open for to long.

Happy testing

fsaris, 2024-01-07:
see the updated example YAML for the new optional config options. Set max connections default to 2 
https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/blob/dev/awox-ble-mesh-hub.yaml#L67

oxigen-dev, 2024-01-07:
> @oxigen-dev after some days of work and a big rewrite as a result 😄 the component is now able to connect to up to 3 devices.
>
> On the mqtt topic `awox-ble-mesh-hub/connection_status` (first part is [...]

@fsaris Big thanks for your work! Of course I've already started testing it :).

I tried to simulate the situation I was writing in my last comment, so I've turned on the bulb in the bedroom and after it was shown as Online in HA, I've turned on the ceiling in the bathroom. Good news is that after couple of seconds it was also shown Online in HA, yeah :).

Unfortunately it was not possible to modify the brightness as nothing happened. Then I turned off the ceiling but it was shown as Online, even after an ESP reboot.

Before reboot:
`[18:06:02][D][awox.mesh:195]: Available device BATHROOM_CEILING => rssi: -92
[18:06:02][D][awox.mesh:195]: Available device BEDROOM_BULB => rssi: -9999
[18:06:02][D][awox.mesh:207]: Currently 1 mesh devices reachable through active connections (2 currently known mesh devices)`

After reboot and some testing I had some strange log:

```
[18:15:05][E][esp32_ble_tracker:190]: Scan start failed: 1
[18:15:05][D][esp32_ble_tracker:266]: Starting scan...
[18:15:05][D][AwoxMesh:087]: scan end
[18:15:05][D][esp-idf:000]: E (44406) BT_APPL: bta_dm_ble_set_scan_fil_params(), fail to set scan params.

[18:15:05][D][esp-idf:000]: W (44419) BT_APPL:  bta_dm_ble_scan start scan failed. status=0x3

[18:15:05][D][esp-idf:000]: E (44426) BT_HCI: CC evt: op=0x200b, status=0xc

[18:15:05][D][esp-idf:000]: E (44431) BT_HCI: CC evt: op=0x200c, status=0xc

[18:15:05][D][esp-idf:000]: E (44435) BT_BTM: BTM_BleScan scan not active

[18:15:05][D][esp-idf:000]: W (44439) BT_APPL:  bta_dm_ble_scan stop scan failed, status=0x6

[18:15:05][D][esp32_ble_tracker:266]: Starting scan...
[18:15:05][D][AwoxMesh:087]: scan end
[18:15:05][D][esp-idf:000]: E (44469) BT_APPL: bta_dm_ble_set_scan_fil_params(), fail to set scan params.

[18:15:05][D][esp-idf:000]: W (44478) BT_APPL:  bta_dm_ble_scan start scan failed. status=0x3

[18:15:05][D][esp-idf:000]: E (44481) BT_HCI: CC evt: op=0x200b, status=0xc

[18:15:05][D][esp-idf:000]: E (44485) BT_HCI: CC evt: op=0x200c, status=0xc

[18:15:05][D][esp-idf:000]: E (44493) BT_BTM: BTM_BleScan scan not active

[18:15:05][D][esp-idf:000]: W (44498) BT_APPL:  bta_dm_ble_scan stop scan failed, status=0x6

[18:15:05][E][esp32_ble_tracker:190]: Scan start failed: 1
[18:15:05][D][esp32_ble_tracker:266]: Starting scan...
[18:15:05][D][AwoxMesh:087]: scan end
[18:15:05][D][esp-idf:000]: E (44536) BT_APPL: bta_dm_ble_set_scan_fil_params(), fail to set scan params.

[18:15:05][D][esp-idf:000]: W (44547) BT_APPL:  bta_dm_ble_scan start scan failed. status=0x3

[18:15:05][D][esp-idf:000]: E (44553) BT_HCI: CC evt: op=0x200b, status=0xc

[18:15:05][D][esp-idf:000]: E (44557) BT_HCI: CC evt: op=0x200c, status=0xc

[18:15:05][D][esp-idf:000]: E (44559) BT_BTM: BTM_BleScan scan not active

[18:15:05][D][esp-idf:000]: W (44562) BT_APPL:  bta_dm_ble_scan stop scan failed, status=0x6

[18:15:05][E][esp32_ble_tracker:190]: Scan start failed: 1
[18:15:05][D][esp32_ble_tracker:266]: Starting scan...
[18:15:05][D][AwoxMesh:087]: scan end
[18:15:05][D][esp-idf:000]: E (44603) BT_APPL: bta_dm_ble_set_scan_fil_params(), fail to set scan params.

[18:15:05][D][esp-idf:000]: W (44616) BT_APPL:  bta_dm_ble_scan start scan failed. status=0x3

[18:15:05][D][esp-idf:000]: E (44623) BT_HCI: CC evt: op=0x200b, status=0xc

[18:15:05][D][esp-idf:000]: E (44628) BT_HCI: CC evt: op=0x200c, status=0xc

[18:15:05][D][esp-idf:000]: E (44632) BT_BTM: BTM_BleScan scan not active

[18:15:05][D][esp-idf:000]: W (44640) BT_APPL:  bta_dm_ble_scan stop scan failed, status=0x6

[18:15:05][D][esp32_ble_tracker:266]: Starting scan...
[18:15:05][D][AwoxMesh:087]: scan end
[18:15:05][D][esp-idf:000]: E (44668) BT_APPL: bta_dm_ble_set_scan_fil_params(), fail to set scan params.

[18:15:05][D][esp-idf:000]: W (44677) BT_APPL:  bta_dm_ble_scan start scan failed. status=0x3

[18:15:05][D][esp-idf:000]: E (44681) BT_HCI: CC evt: op=0x200b, status=0xc

[18:15:05][D][esp-idf:000]: E (44684) BT_HCI: CC evt: op=0x200c, status=0xc

[18:15:05][D][esp32_ble_tracker:266]: Starting scan...
[18:15:05][D][AwoxMesh:087]: scan end
[18:15:05][D][esp-idf:000]: E (44712) BT_APPL: bta_dm_ble_set_scan_fil_params(), fail to set scan params.

[18:15:05][D][esp-idf:000]: W (44724) BT_APPL:  bta_dm_ble_scan start scan failed. status=0x3

[18:15:05][D][esp-idf:000]: E (44731) BT_HCI: CC evt: op=0x200b, status=0xc

[18:15:05][D][esp-idf:000]: E (44736) BT_HCI: CC evt: op=0x200c, status=0xc

[18:15:05][D][esp-idf:000]: E (44740) BT_BTM: BTM_BleScan scan not active

[18:15:05][D][esp-idf:000]: W (44743) BT_APPL:  bta_dm_ble_scan stop scan failed, status=0x6

[18:15:05][D][esp-idf:000]: E (44753) BT_APPL: bta_dm_ble_set_scan_fil_params(), fail to set scan params.

[18:15:05][E][esp32_ble_tracker:194]: Scan set param failed: 1
[18:15:05][D][esp-idf:000]: W (44765) BT_APPL:  bta_dm_ble_scan start scan failed. status=0x3

[18:15:05][D][esp-idf:000]: E (44770) BT_HCI: CC evt: op=0x200b, status=0xc

[18:15:05][D][esp-idf:000]: E (44774) BT_HCI: CC evt: op=0x200c, status=0xc

[18:15:05][E][esp32_ble_tracker:180]: ESP-IDF BLE scan could not restart after 255 attempts, rebooting to restore BLE stack...
[18:15:05][I][app:127]: Forcing a reboot...
[18:15:05][D][esp-idf:000]: I (44800) MQTT_CLIENT: Client asked to disconnect

INFO Processing expected disconnect from ESPHome API for awox-ble-mesh-hub @ 192.168.1.58
WARNING Disconnected from API
```

fsaris, 2024-01-07:
```
[18:15:05][D][AwoxMesh:087]: scan end
[18:15:05][D][esp-idf:000]: E (44406) BT_APPL: bta_dm_ble_set_scan_fil_params(), fail to set scan params.
[18:15:05][D][esp-idf:000]: W (44419) BT_APPL:  bta_dm_ble_scan start scan failed. status=0x3
[18:15:05][D][esp-idf:000]: E (44426) BT_HCI: CC evt: op=0x200b, status=0xc
[18:15:05][D][esp-idf:000]: E (44431) BT_HCI: CC evt: op=0x200c, status=0xc
[18:15:05][D][esp-idf:000]: E (44435) BT_BTM: BTM_BleScan scan not active
[18:15:05][D][esp-idf:000]: W (44439) BT_APPL:  bta_dm_ble_scan stop scan failed, status=0x6
....
```

I had the same strange logs, that's why I for now set the max connections to 2. Not sure yet what's causing this. Letting it currently run for a longer time to see how it behaves. 

```
INFO Processing expected disconnect from ESPHome API for awox-ble-mesh-hub @ 192.168.1.58
WARNING Disconnected from API
```

Do you have the `api:` enabled? I had to disable is to prevent memory issues.

Can you check latest dev version to also switch to max 2 connections? I also improved the logging a bit.

oxigen-dev, 2024-01-07:
> ```
> [18:15:05][D][AwoxMesh:087]: scan end
> [18:15:05][D][esp-idf:000]: E (44406) BT_APPL: bta_dm_ble_set_scan_fil_params(), fail to set scan params.
> [18:15:05][D][esp-idf:000]: W (44419) BT_APPL: [...]

Yes, `api:` is enabled. Now I've set max connections to 2. Unfortunately I've got this strange thing again.
The other thing what happened again is that the bathroom ceiling was turned off for 20 minutes and it was still shown as Online. After that I've restarted the ESP which fixed the status of the ceiling but then came this strange thing.
I'll continue testing it. Should I increase the log level to verbose?

oxigen-dev, 2024-01-07:
@fsaris Would it be possible to set friendly names for the lamps in the configuration or get them from HA? As I have around 10 lamps it is not so easy to debug the situation using the logs only by the MAC addresses :). Thanks!

fsaris, 2024-01-08:
> I'll continue testing it. Should I increase the log level to verbose?

No, normal debug logging should give enough.

> The other thing what happened again is that the bathroom ceiling was turned off for 20 minutes and it was still shown as Online. 

Mmm, looks that there is still a "is online" message going around that report a false positive. (Logs would help te verify).

Can also introduce a periodic re-connect of all connections maybe that helps keeping the state correct.

> After that I've restarted the ESP which fixed the status of the ceiling but then came this strange thing.

Looks that the Bluetooth controller is locked somehow but didn't find the origin of this yet.

> Would it be possible to set friendly names for the lamps in the configuration or get them from HA?

Will check if I can add a config to match a name with Mac address

fsaris, 2024-01-09:
@oxigen-dev I disabled some Bluetooth Cache setting (that also isn't used in the main branch) maybe that helps to prevent the strange errors

oxigen-dev, 2024-01-09:
> @oxigen-dev I disabled some Bluetooth Cache setting (that also isn't used in the main branch) maybe that helps to prevent the strange errors

I've flashed my ESP right after your commit. I would not call "testing" what I did today, I can only speak about my experiences.

First, I've tried with `max_connections=3` and in some minutes I've got the same error. With `max_connections=2` I've had better luck, but in some hours the same thing happened.

My overall experience was that device detection is working most of the time for lamps which are far from each other (in [my case](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/issues/12#issuecomment-1875256069) the bedroom bulb and bathroom ceiling) although it also happened that 1 out of 4 lamps which are 2-3 meters from the ESP was unavailable.
Detecting ceilings which are switched off was most of the time not working, only after a reboot.

I'll try to provide later some test scenarios with logs.

fsaris, 2024-01-10:
@oxigen-dev updated version.

1. when connection disconnects all linked devices are marked as offline
2. every 20 seconds the connections are checks if there are connections that have the same mesh_ids linked. If true, 1 connection is dropped

oxigen-dev, 2024-01-10:
> @oxigen-dev updated version.
> 
>     1. when connection disconnects all linked devices are marked as offline
> 
>     2. every 20 seconds the connections are checks if there are connections that have the same mesh_ids linked. If true, 1 connection is dropped

Hi @fsaris ,

I've did some tests and detecting the turned off lamps seems to be way more reliable!

- I was first testing with the bedroom bulb and bathroom ceiling.
- Both of them were controllable (actually I've had some Automations in HA for the bathroom ceiling but they never worked as the ESP was never able to connect to it in the night so I was quite surprised when the color changed to red :D).
- As I was turning off the bathroom ceiling it was shown in HA as Offline in a few seconds. I did this test several times.
- However after turning off the bathroom ceiling and switching on the office ceiling (bedroom bulb was still on) I had some problems. It was shown as Online but it was not controllable then it went to Offline-Online-Offline etc. I was never able to change the brightness. This ceiling is more close to the ESP than the bathroom ceiling and normally if only this lamp is turned on it works well.

I was creating a log, maybe it can help.

[awox_20240110.log](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/13895212/awox_20240110.log)

fsaris, 2024-01-12:
@oxigen-dev ok so we are getting closer to a solution :)

I pushed a new version with a small fix

oxigen-dev, 2024-01-13:
> @oxigen-dev ok so we are getting closer to a solution :)
> 
> I pushed a new version with a small fix

Hi @fsaris ,

I'm testing the new code since yesterday. Switched off lamp detection still works like a charm :)

In connecting to and controlling the ceilings I still experience some issues (eg. sometimes the ceiling iterates between Online->Unavailable->Online->Unavailable in every few minutes then works fine for some hours OR the ceiling is Online but is not controllable).

I think one reason for this status iteration is the constant reboots of the ESP. Sometimes it can run >10 hours, but today from 00:00 - 03:00 the following happened:
[image]
Now I've reflashed ESPHome with `api:` `bluetooth_proxy:` and `time:` disabled. I still have `ap:` (but not `captive_portal:`), `text_sensor:`, `switch:`,  `logger:`, `debug:`, `ota:` and the mandatory things.

Why the ceilings are sometimes not controllable.. I don't have an idea yet. However, I was creating a log all day and in that timeframe the bathroom ceiling was not controllable:
[awox_20240113.log](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/13928272/awox_20240113.log)

Thanks!

fsaris, 2024-01-13:
I found a small thing that could result in a memory issue.
For my the heap size also decreased when the it couldn't correctly create a Bluetooth connection.

But after moving the esp a little it could connect and it was stable for since last evening until just yet.

Your logs tell me also that the device isn't able to connect to the `BATHROOM_CEILING`. Maybe there is more interference due to other devices that are enabled that makes the connection unstable?

Added some sensors telling more about the active connection etc.

oxigen-dev, 2024-01-13:
> I found a small thing that could result in a memory issue.
> For my the heap size also decreased when the it couldn't correctly create a Bluetooth connection.
>
> But after moving the esp a little it [...]

@fsaris Thanks for the fix, I'm already testing it. I think in the last few weeks we were able to get very close to a reliable and robust integration.

About the unstable connection: it is very likely, that you are right. On HW side I think I tried to do the maximum (I have an external antenna), but maybe it is still not enough. 

It would be nice to develop a concept to be able to use multiple ESPs. I would buy another one and test it.

[16 more replies]

closed 2024-01-18 by fsaris