fsaris/EspHome-AwoX-BLE-mesh-hub #15: New device request

michalpulda, 2023-05-04:
Hello,

Could you please add a new device? Please see the MQTT payload bellow. In the original integration it went by name "ESMLm-w9w", so I guess it would mean to uncomment row 207 of the `device_info.h` file.

Thanks!

```
schema: json
name: Unknown device type
unique_id: awox-A4:C1:38:56:C1:CD-light
icon: mdi:lightbulb-help-outline
state_topic: esp32u-ble-awox/16845/state
command_topic: esp32u-ble-awox/16845/command
availability:
  - topic: esp32u-ble-awox/16845/availability
  - topic: esp32u-ble-awox/status
availability_mode: all
color_mode: true
brightness: true
brightness_scale: 255
supported_color_modes:
  - brightness
device:
  identifiers:
    - esp-awox-mesh-16845
    - A4:C1:38:56:C1:CD
  name: Unknown device type
  model: t
  manufacturer: AwoX
  via_device: 083af26f0f90
platform: mqtt
```

fsaris, 2023-05-07:
Can you check the log in ESPHome en look for productID

michalpulda, 2023-05-08:
Here it is:

```
[15:04:11][D][mesh_device:379]: MAC report, dev [16845]: productID: 49 mac: A4:C1:38:56:C1:CD => 1010000100101011000000111100110101000001110000001100110011011000011000000000000100000000000100000100100111001101110000010101011000111000000000000000000000000000
[15:04:11][D][mesh_device:499]: '16845': Sending discovery...

[15:06:43][D][mesh_device:631]: [16845] Process command state
[15:06:44][D][mesh_device:346]: online status report: mesh: 16845, on: 1, color_mode: 0, transition_mode: 0, w_b: 64, temp: 127, c_b: 0, rgb: 000000 
[15:06:44][I][mesh_device:411]: 16845: ON temp: 127 (64 %) ONLINE

[15:06:47][D][mesh_device:631]: [16845] Process command state
[15:06:48][D][mesh_device:346]: online status report: mesh: 16845, on: 0, color_mode: 0, transition_mode: 0, w_b: 64, temp: 127, c_b: 0, rgb: 000000 
[15:06:48][I][mesh_device:411]: 16845: OFF temp: 127 (64 %) ONLINE
```

Thanks!

fsaris, 2023-05-11:
@michalpulda please use the latest version of this component and add the next lines to your yaml config:

```yaml
device_info:
-  product_id: 0x49
   device_type: DIM
   name: EGLOBulb A60 Warm
   model: ESMLm_w9w
   manufacturer: EGLO
```

closed 2023-05-11 by fsaris

michalpulda, 2023-05-12:
> @michalpulda please use the latest version of this component and add the next lines to your yaml config:

I have added it and it works good, **thank you**!

Should I migrate to this configuration my other devices (3x GLOSpot 170/w, EGLOSpot 120/w, ETriSpot_85, EGLO PLUG PLUS)? Do you plan to drop original automatic setup? Thanks!

fsaris, 2023-05-12:
> I have added it and it works good, **thank you**!
>
> Should I migrate to this configuration my other devices (3x GLOSpot 170/w, EGLOSpot 120/w, ETriSpot_85, EGLO PLUG PLUS)? Do you plan to drop [...]

The existing config is dropped so yes you need to add your other device types