fsaris/EspHome-AwoX-BLE-mesh-hub #20: [device_info] is an invalid option for [awox_mesh]. Please check the indentation.

danieldur, 2023-05-18:
Hello Frans, 

I've been struggling with this ESP32 for the past 5-6 hours and I'm not sure what am I doing wrong. Blow are my config and then the error I'm getting when I tried to flash the config. I tryed playing around with the indentation of device_info, looked at other examples in the settings, but I can't seem to be able to figure it out on my own. 

Please help me! 

Thanks! 
Daniel

```
esphome:
  name: "esp1"
  friendly_name: "esp1"

esp32:
  board: esp32dev
  framework:
    type: arduino
    version: 2.0.6 #https://github.com/espressif/arduino-esp32/releases

status_led:
  pin: GPIO2

# Enable logging
logger:

debug:

text_sensor:
  - platform: debug
    reset_reason:
      name: "Reset Reason"

# Enable Home Assistant API
#api:
#  encryption:
#    key: "xxx"

ota:
  password: " xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp1 Fallback Hotspot"
    password: "xxx"

captive_portal:
    
external_components:
  source: github://fsaris/EspHome-AwoX-BLE-mesh-hub

mqtt:
  broker: !secret mqtt_host
  username: mqtt_user
  password: !secret mqtt_password

esp32_ble_tracker:
  scan_parameters:
    active: false

awox_mesh:
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password
  device_info:
    - product_id: 0x30
      device_type: RGB
      name: "kid"
      model: EPanel_120
      manufacturer: EGLO

``` 

```
INFO Reading configuration /config/esphome/esp1.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

awox_mesh: [source /config/esphome/esp1.yaml:59]
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password
  
  [device_info] is an invalid option for [awox_mesh]. Please check the indentation.
  device_info: 
    - product_id: 48
      device_type: RGB
      name: kid
      model: EPanel_120
      manufacturer: EGLO
```

fsaris, 2023-05-18:
Looks like you don't have the latest version of the component, maybe you need to force a refresh of the code (https://esphome.io/components/external_components.html#refresh)

danieldur, 2023-05-19:
Thanks! That was indeed the issue!
But I still cannot make it work with my mesh. So, if it is ok, I will continue posting here without creating a new issue.
Here are some logs: https://pastebin.com/EzfqtwQ4 (I hope it is enough detail) and here is my config: https://pastebin.com/pPLvjwy5 .

Thank you for all the help!

fsaris, 2023-05-20:
Your device looks to be out of reach of your lights.
A RSSI of -80 till -90 is low to setup a proper connection.

closed 2023-08-08 by fsaris