fsaris/EspHome-AwoX-BLE-mesh-hub #22: Light panel changes colour by itself after a while

Marcel1202, 2023-06-30:
Whenever I turn off my light panel electrically and turn it back on, it is always red at first (which is fine, I guess).
I can change the colour just fine using HA using your component, but after a few minutes the light changes colour by itself.
The changes don't seem arbitrary, each colour is displayed for a few seconds, and it cycles in patterns.

`[02:19:32][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 0, transition_mode: 0, w_b: 88, temp: 25, c_b: 69, rgb: FF0033 
[02:19:32][I][mesh_device:415]: 15: ON temp: 25 (88 %) ONLINE
[02:19:41][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 0, transition_mode: 0, w_b: 114, temp: 0, c_b: 89, rgb: FF0033 
[02:19:41][I][mesh_device:415]: 15: ON temp: 0 (114 %) ONLINE
[02:19:52][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 1, transition_mode: 0, w_b: 127, temp: 0, c_b: 100, rgb: FF5E05 
[02:19:52][I][mesh_device:415]: 15: ON #FF5E05 (100 %) ONLINE
[02:20:01][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 1, transition_mode: 0, w_b: 127, temp: 0, c_b: 100, rgb: FFDB05 
[02:20:01][I][mesh_device:415]: 15: ON #FFDB05 (100 %) ONLINE
[02:20:12][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 1, transition_mode: 0, w_b: 127, temp: 0, c_b: 100, rgb: 82FF2E 
[02:20:12][I][mesh_device:415]: 15: ON #82FF2E (100 %) ONLINE
[02:20:21][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 1, transition_mode: 0, w_b: 127, temp: 0, c_b: 100, rgb: 47FFE7 
[02:20:21][I][mesh_device:415]: 15: ON #47FFE7 (100 %) ONLINE
[02:20:32][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 1, transition_mode: 0, w_b: 127, temp: 0, c_b: 100, rgb: 0F0FFF 
[02:20:32][I][mesh_device:415]: 15: ON #0F0FFF (100 %) ONLINE
[02:20:41][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 1, transition_mode: 0, w_b: 127, temp: 0, c_b: 100, rgb: FF2E96 
[02:20:41][I][mesh_device:415]: 15: ON #FF2E96 (100 %) ONLINE
[02:20:52][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 1, transition_mode: 0, w_b: 127, temp: 0, c_b: 100, rgb: 000000 
[02:20:52][I][mesh_device:415]: 15: ON #000000 (100 %) ONLINE
[02:21:01][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 0, transition_mode: 0, w_b: 114, temp: 127, c_b: 89, rgb: 000000 
[02:21:01][I][mesh_device:415]: 15: ON temp: 127 (114 %) ONLINE
[02:21:12][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 1, transition_mode: 0, w_b: 127, temp: 127, c_b: 100, rgb: FF0033 
[02:21:12][I][mesh_device:415]: 15: ON #FF0033 (100 %) ONLINE
[02:21:21][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 0, transition_mode: 0, w_b: 88, temp: 101, c_b: 69, rgb: FF0033 
[02:21:21][I][mesh_device:415]: 15: ON temp: 101 (88 %) ONLINE
[02:21:32][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 0, transition_mode: 0, w_b: 88, temp: 25, c_b: 69, rgb: FF0033 
[02:21:32][I][mesh_device:415]: 15: ON temp: 25 (88 %) ONLINE
[02:21:42][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 0, transition_mode: 0, w_b: 114, temp: 0, c_b: 89, rgb: FF0033 
[02:21:42][I][mesh_device:415]: 15: ON temp: 0 (114 %) ONLINE
[02:21:51][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 1, transition_mode: 0, w_b: 127, temp: 0, c_b: 100, rgb: FF5E05 
[02:21:51][I][mesh_device:415]: 15: ON #FF5E05 (100 %) ONLINE
[02:22:02][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 1, transition_mode: 0, w_b: 127, temp: 0, c_b: 100, rgb: FFDB05 
[02:22:02][I][mesh_device:415]: 15: ON #FFDB05 (100 %) ONLINE
[02:22:11][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 1, transition_mode: 0, w_b: 127, temp: 0, c_b: 100, rgb: 82FF2E 
[02:22:11][I][mesh_device:415]: 15: ON #82FF2E (100 %) ONLINE
[02:22:22][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 1, transition_mode: 0, w_b: 127, temp: 0, c_b: 100, rgb: 47FFE7 
[02:22:22][I][mesh_device:415]: 15: ON #47FFE7 (100 %) ONLINE
[02:22:31][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 1, transition_mode: 0, w_b: 127, temp: 0, c_b: 100, rgb: 0F0FFF 
[02:22:31][I][mesh_device:415]: 15: ON #0F0FFF (100 %) ONLINE
[02:22:42][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 1, transition_mode: 0, w_b: 127, temp: 0, c_b: 100, rgb: FF2E96 
[02:22:42][I][mesh_device:415]: 15: ON #FF2E96 (100 %) ONLINE
[02:22:51][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 1, transition_mode: 0, w_b: 127, temp: 0, c_b: 100, rgb: 000000 
[02:22:51][I][mesh_device:415]: 15: ON #000000 (100 %) ONLINE
[02:23:02][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 0, transition_mode: 0, w_b: 114, temp: 127, c_b: 89, rgb: 000000 
[02:23:02][I][mesh_device:415]: 15: ON temp: 127 (114 %) ONLINE
[02:23:11][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 1, transition_mode: 0, w_b: 127, temp: 127, c_b: 100, rgb: FF0033 
[02:23:11][I][mesh_device:415]: 15: ON #FF0033 (100 %) ONLINE
[02:23:22][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 0, transition_mode: 0, w_b: 88, temp: 101, c_b: 69, rgb: FF0033 
[02:23:22][I][mesh_device:415]: 15: ON temp: 101 (88 %) ONLINE
[02:23:31][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 0, transition_mode: 0, w_b: 88, temp: 25, c_b: 69, rgb: FF0033 
[02:23:31][I][mesh_device:415]: 15: ON temp: 25 (88 %) ONLINE
[02:23:42][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 0, transition_mode: 0, w_b: 114, temp: 0, c_b: 89, rgb: FF0033 
[02:23:42][I][mesh_device:415]: 15: ON temp: 0 (114 %) ONLINE
[02:23:52][D][mesh_device:350]: online status report: mesh: 15, on: 1, color_mode: 1, transition_mode: 0, w_b: 127, temp: 0, c_b: 100, rgb: FF5E05 `

Marcel1202, 2023-06-30:
Fixed this by adding code into your Awox Integration that logs mesh name and password. Still, an easier way to do this and better instructions would be appreciated. Anyway, thanks for providing this component.

fsaris, 2023-06-30:
It's on my list to add pairing of a new light to feature set. But that more of a winter project 😉

closed 2023-08-08 by fsaris