fsaris/EspHome-AwoX-BLE-mesh-hub #26: Trying to connect to lights but fails the loops

gallaid, 2023-08-09:
[01:30:38][I][app:102]: ESPHome version 2023.7.1 compiled on Aug 10 2023, 01:25:25
[01:30:38][C][status_led:019]: Status LED:
[01:30:38][C][status_led:020]:   Pin: GPIO2
[01:30:38][C][wifi:543]: WiFi:
[01:30:38][C][wifi:379]:   Local MAC: 40:91:51:FD:08:88
[01:30:38][C][wifi:380]:   SSID: [redacted]
[01:30:38][C][wifi:381]:   IP Address: 192.168.50.30
[01:30:38][C][wifi:383]:   BSSID: [redacted]
[01:30:38][C][wifi:384]:   Hostname: 'awox-ble-mesh-hub'
[01:30:38][C][wifi:386]:   Signal strength: -54 dB ▂▄▆█
[01:30:38][C][wifi:390]:   Channel: 8
[01:30:38][C][wifi:391]:   Subnet: 255.255.255.0
[01:30:38][C][wifi:392]:   Gateway: 192.168.50.1
[01:30:38][C][wifi:393]:   DNS1: 0.0.0.0
[01:30:38][C][wifi:394]:   DNS2: 0.0.0.0
[01:30:38][C][logger:301]: Logger:
[01:30:38][C][logger:302]:   Level: DEBUG
[01:30:38][C][logger:303]:   Log Baud Rate: 115200
[01:30:38][C][logger:305]:   Hardware UART: UART0
[01:30:38][C][esp32_ble:238]: ESP32 BLE:
[01:30:38][C][esp32_ble:240]:   MAC address: 40:91:51:FD:08:8A
[01:30:38][C][esp32_ble:241]:   IO Capability: none
[01:30:38][C][esp32_ble_tracker:616]: BLE Tracker:
[01:30:38][C][esp32_ble_tracker:617]:   Scan Duration: 300 s
[01:30:38][C][esp32_ble_tracker:618]:   Scan Interval: 320.0 ms
[01:30:38][C][esp32_ble_tracker:619]:   Scan Window: 30.0 ms
[01:30:38][C][esp32_ble_tracker:620]:   Scan Type: PASSIVE
[01:30:38][C][esp32_ble_tracker:621]:   Continuous Scanning: True
[01:30:39][C][mdns:112]: mDNS:
[01:30:39][C][mdns:113]:   Hostname: awox-ble-mesh-hub
[01:30:39][C][ota:093]: Over-The-Air Updates:
[01:30:39][C][ota:094]:   Address: 192.168.50.30:3232
[01:30:39][C][ota:097]:   Using Password.
[01:30:39][W][ota:103]: Last Boot was an unhandled reset, will proceed to safe mode in 6 restarts
[01:30:39][C][mqtt:127]: MQTT:
[01:30:39][C][mqtt:129]:   Server Address: 192.168.50.20:1883 (192.168.50.20)
[01:30:39][C][mqtt:130]:   Username: [redacted]
[01:30:39][C][mqtt:131]:   Client ID: [redacted]
[01:30:39][C][mqtt:133]:   Discovery prefix: 'homeassistant'
[01:30:39][C][mqtt:134]:   Discovery retain: YES
[01:30:39][C][mqtt:136]:   Topic Prefix: 'awox-ble-mesh-hub'
[01:30:39][C][mqtt:138]:   Log Topic: 'awox-ble-mesh-hub/debug'
[01:30:39][C][mqtt:141]:   Availability: 'awox-ble-mesh-hub/status'
[01:30:58][I][AwoxMesh:071]: Failed to connect A4:C1:38:99:B0:CB => rssi: -65
[01:30:58][I][esp32_ble_client:081]: [0] [A4:C1:38:99:B0:CB] Disconnecting.
[01:30:58][I][MeshDevice:155]: DISCONNECTING
[01:30:58][I][mesh_device:475]: Publish connected to mesh device - offline
[01:30:58][D][AwoxMesh:057]: Total devices: 4
[01:30:58][D][AwoxMesh:059]: Available device A4:C1:38:99:B0:CB => rssi: -66
[01:30:58][D][AwoxMesh:059]: Available device A4:C1:38:D9:8C:92 => rssi: -70
[01:30:58][D][AwoxMesh:059]: Available device A4:C1:38:FD:4C:DF => rssi: -82
[01:30:58][D][AwoxMesh:059]: Available device A4:C1:38:BE:DE:45 => rssi: -88
[01:30:58][I][AwoxMesh:063]: Try to connect A4:C1:38:99:B0:CB => rssi: -66
[01:30:58][I][mesh_device:475]: Publish connected to mesh device - offline
[01:30:58][I][esp32_ble_client:064]: [0] [A4:C1:38:99:B0:CB] 0x00 Attempting BLE connection
[01:30:58][I][MeshDevice:175]: CONNECTING
[01:30:58][W][component:204]: Component awox_mesh took a long time for an operation (0.06 s).
[01:30:58][W][component:205]: Components should block for at most 20-30ms.
[01:31:08][I][MeshDevice:159]: IDLE
[01:31:08][D][mesh_device:144]: [0] [A4:C1:38:99:B0:CB] ESP_GATTC_DISCONNECT_EVT, reason 0
[01:31:08][W][esp32_ble_client:134]: [0] [A4:C1:38:99:B0:CB] Connection failed, status=133
[01:31:08][I][MeshDevice:159]: IDLE
[01:31:08][D][esp32_ble_client:048]: [0] [A4:C1:38:99:B0:CB] Found device
[01:31:08][I][MeshDevice:167]: DISCOVERED
[01:31:08][D][esp32_ble_tracker:213]: Pausing scan to make connection...
[01:31:08][I][MeshDevice:171]: READY_TO_CONNECT
[01:31:08][I][esp32_ble_client:064]: [0] [A4:C1:38:99:B0:CB] 0x00 Attempting BLE connection
[01:31:08][I][MeshDevice:175]: CONNECTING
[01:31:31][I][mqtt:286]: MQTT Connected!
[01:31:31][I][app:062]: setup() finished successfully!

superRandomGuy941, 2023-08-10:
What is the problem here?

gallaid, 2023-08-10:
I edited my original post to explain more.
Here is the .yaml I am trying to install on my ESP32.

esphome:
  name: awox-ble-mesh-hub
  friendly_name: "AwoX BLE mesh hub"

esp32:
  board: esp32dev
  framework:
    type: arduino
    # Set to fixed version due to build issues https://community.home-assistant.io/t/esphome-could-not-find-package-json/499026/8
    version: 2.0.6 #https://github.com/espressif/arduino-esp32/releases

status_led:
  pin: GPIO2

# Enable logging
logger:
  #level: VERY_VERBOSE

# debug:
#   update_interval: 5s

# text_sensor:
#   - platform: debug
#     reset_reason:
#      name: "Reset Reason"

# sensor:
#   - platform: debug
#     free:
#       name: "Heap Free"
#     loop_time:
#       name: "Loop Time"

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_user
  password: !secret mqtt_password

# Disable Home Assistant API to prevent frequent MQTT disconnections
#api:

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.50.30
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.50.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

external_components:
  source: github://fsaris/EspHome-AwoX-BLE-mesh-hub
  # - source:
  #     type: local
  #     path: components

esp32_ble_tracker:
  scan_parameters:
    active: false

awox_mesh:
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password
  address_prefix: A4:C1
  device_info:
    # Example device type, see device info in HomeAssistant or MQTT message to find the 'product_id'
    - product_id: 0x32
      device_type: RGB
      manufacturer: EGLO
      name: Spot 120
      model: EGLOSpot 120/w
      icon: mdi:wall-sconce-flat

superRandomGuy941, 2023-08-10:
Maybe the manual ip is the problem, try it without that

fsaris, 2023-08-10:
@gallaid is there no activity after the last log you posted? Currently it can take a few connect attempts before the BLE connection is established

gallaid, 2023-08-10:
@fsaris 
It just loops over and over again until the countdown reaches 0.
[01:30:39][W][ota:103]: Last Boot was an unhandled reset, will proceed to safe mode in 6 restarts

gallaid, 2023-08-10:
> Maybe the manual ip is the problem, try it without that

Tried and no change.

fsaris, 2023-08-11:
Can you let is run with logging set to verbose and post the logs here?

What version of esphome are you using?

gallaid, 2023-08-11:
@fsaris 

> Can you let is run with logging set to verbose and post the logs here?
> 
> What version of esphome are you using?
ESPHome
Current version: 2023.7.1
I attached the Logs
[logs_awox_Bluetooth_run.txt](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/12321301/logs_awox_Bluetooth_run.txt)

NeonPimpZ, 2023-08-15:
I am facing the same problem. 

I'll post a part of my log
`INFO ESPHome 2023.7.1
INFO Reading configuration /config/esphome/awox-mesh-hub.yaml...
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from awox-mesh-hub.local using esphome API
INFO Successfully connected to awox-mesh-hub.local
[11:34:26][I][app:102]: ESPHome version 2023.7.1 compiled on Aug 15 2023, 11:29:46
[11:34:26][C][status_led:019]: Status LED:
[11:34:26][C][status_led:020]:   Pin: GPIO2
[11:34:26][C][wifi:543]: WiFi:
[11:34:26][C][wifi:379]:   Local MAC: [redacted]
[11:34:26][C][wifi:380]:   SSID: [redacted]
[11:34:26][C][wifi:381]:   IP Address: 192.168.1.250
[11:34:26][C][wifi:383]:   BSSID: [redacted]
[11:34:26][C][wifi:384]:   Hostname: 'awox-mesh-hub'
[11:34:26][C][wifi:386]:   Signal strength: -87 dB ▂▄▆█
[11:34:26][C][wifi:390]:   Channel: 11
[11:34:26][C][wifi:391]:   Subnet: 255.255.255.0
[11:34:26][C][wifi:392]:   Gateway: 192.168.1.1
[11:34:26][C][wifi:393]:   DNS1: 192.168.1.1
[11:34:26][C][wifi:394]:   DNS2: 0.0.0.0
[11:34:26][C][esp32_ble:238]: ESP32 BLE:
[11:34:26][C][esp32_ble:240]:   MAC address: [redacted]
[11:34:26][C][esp32_ble:241]:   IO Capability: none
[11:34:26][C][esp32_ble_tracker:616]: BLE Tracker:
[11:34:26][C][esp32_ble_tracker:617]:   Scan Duration: 300 s
[11:34:26][C][esp32_ble_tracker:618]:   Scan Interval: 320.0 ms
[11:34:26][C][esp32_ble_tracker:619]:   Scan Window: 30.0 ms
[11:34:26][C][esp32_ble_tracker:620]:   Scan Type: PASSIVE
[11:34:26][C][esp32_ble_tracker:621]:   Continuous Scanning: True
[11:34:26][C][mdns:112]: mDNS:
[11:34:26][C][mdns:113]:   Hostname: awox-mesh-hub
[11:34:26][C][ota:093]: Over-The-Air Updates:
[11:34:26][C][ota:094]:   Address: awox-mesh-hub.local:3232
[11:34:26][C][ota:097]:   Using Password.
[11:34:26][W][ota:103]: Last Boot was an unhandled reset, will proceed to safe mode in 7 restarts
[11:34:26][C][api:138]: API Server:
[11:34:26][C][api:139]:   Address: awox-mesh-hub.local:6053
[11:34:26][C][api:141]:   Using noise encryption: YES
[11:34:26][C][mqtt:127]: MQTT:
[11:34:27][C][mqtt:129]:   Server Address: 192.168.1.180:1883 (192.168.1.180)
[11:34:27][C][mqtt:130]:   Username: [redacted]
[11:34:27][C][mqtt:131]:   Client ID: [redacted]
[11:34:27][C][mqtt:133]:   Discovery prefix: 'homeassistant'
[11:34:27][C][mqtt:134]:   Discovery retain: YES
[11:34:27][C][mqtt:136]:   Topic Prefix: 'awox-mesh-hub'
[11:34:27][C][mqtt:138]:   Log Topic: 'awox-mesh-hub/debug'
[11:34:27][C][mqtt:141]:   Availability: 'awox-mesh-hub/status'
[11:34:28][D][AwoxMesh:057]: Total devices: 2
[11:34:28][D][AwoxMesh:059]: Available device A4:C1:38:7B:2C:91 => rssi: -44
[11:34:28][D][AwoxMesh:059]: Available device A4:C1:38:35:52:1F => rssi: -89
[11:34:28][I][AwoxMesh:063]: Try to connect A4:C1:38:7B:2C:91 => rssi: -44
[11:34:28][I][mesh_device:475]: Publish connected to mesh device - offline
[11:34:28][I][esp32_ble_client:064]: [0] [A4:C1:38:7B:2C:91] 0x00 Attempting BLE connection
[11:34:28][I][MeshDevice:175]: CONNECTING
[11:34:29][W][component:204]: Component awox_mesh took a long time for an operation (0.06 s).
[11:34:29][W][component:205]: Components should block for at most 20-30ms.
[11:34:48][I][AwoxMesh:071]: Failed to connect A4:C1:38:7B:2C:91 => rssi: -44
[11:34:48][I][esp32_ble_client:081]: [0] [A4:C1:38:7B:2C:91] Disconnecting.
[11:34:48][I][MeshDevice:155]: DISCONNECTING
[11:34:48][I][mesh_device:475]: Publish connected to mesh device - offline
[11:34:48][D][AwoxMesh:057]: Total devices: 2
[11:34:48][D][AwoxMesh:059]: Available device A4:C1:38:7B:2C:91 => rssi: -44
[11:34:49][D][AwoxMesh:059]: Available device A4:C1:38:35:52:1F => rssi: -86
[11:34:49][I][AwoxMesh:063]: Try to connect A4:C1:38:7B:2C:91 => rssi: -44
[11:34:49][I][mesh_device:475]: Publish connected to mesh device - offline
[11:34:49][I][esp32_ble_client:064]: [0] [A4:C1:38:7B:2C:91] 0x00 Attempting BLE connection
[11:34:49][I][MeshDevice:175]: CONNECTING
[11:34:58][I][MeshDevice:159]: IDLE
[11:34:58][D][mesh_device:144]: [0] [A4:C1:38:7B:2C:91] ESP_GATTC_DISCONNECT_EVT, reason 0
[11:34:59][W][esp32_ble_client:134]: [0] [A4:C1:38:7B:2C:91] Connection failed, status=133
[11:34:59][I][MeshDevice:159]: IDLE
[11:34:59][D][esp32_ble_client:048]: [0] [A4:C1:38:7B:2C:91] Found device
[11:34:59][I][MeshDevice:167]: DISCOVERED
[11:34:59][D][esp32_ble_tracker:213]: Pausing scan to make connection...
[11:34:59][I][MeshDevice:171]: READY_TO_CONNECT
[11:34:59][I][esp32_ble_client:064]: [0] [A4:C1:38:7B:2C:91] 0x00 Attempting BLE connection
[11:34:59][I][MeshDevice:175]: CONNECTING
WARNING awox-mesh-hub.local: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for awox-mesh-hub.local
WARNING Disconnected from API
`

This continues after rebooting. Sometimes it will connect to the light (never seen it in HA though, but the log says connected) but it will also fail with the same 'Connection reset by peer' error.

fsaris, 2023-08-16:
@gallaid not sure yet what's happening. It could be that the device is out of reach of the lights to setup a connection. Even if it gets a RSSI reading it doesn't  have to be enough to connect.

@NeonPimpZ the `Connection reset by peer` could be the result from a reboot of the esp due to a lack of resources. What is the message in the logs after that  line?

gallaid, 2023-08-17:
@fsaris The ESP32 is right under one of the lights that I am trying to connect to.
Could it be a faulty board? Is there a test I can do to test it?

fsaris, 2023-08-20:
@gallaid can you maybe try current dev branch? I updated the dependencies and this should fix issues with some (newer) ESP boards. Also update ESPHome to 2023.8.1.

You can use this in the YAML to install the dev version:

```
external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub
      ref: dev
```

fsaris, 2023-09-28:
Closing this for now, please let me know if the issue is still present after updating to the latest version

closed 2023-09-28 by fsaris