fsaris/EspHome-AwoX-BLE-mesh-hub #28: Awox service can have multiple credentials

mrPjer, 2023-09-12:
Hey, thanks a lot for publishing this!

I got stuck on `Device authentication error: known mesh credentials are not excepted by the device. Did you re-pair them to your Awox app with a different account?`, even though I used the credentials from the JSFiddle. Then I decided to see if there are any other credentials in the Awox account, and it turns out that you have credentials per-service. I have a bluetooth light and a zigbee light on my account, and these don't share tokens. I wanted to control the Bluetooth light, which I added before the Zigbee one, but the JSFiddle printed out the access token for the zigbee one, which didn't work. When I tried the `mesh` access token, everything started working.

Here's an example response JSON with multiple services. I guess the Fiddle should be updated to get the `mesh` access token, and print all of them, if there are more.

```
  "results": [
    {
      "objectId": "$OBJECT_ID1",
      "service": "zigbee",
      "client_id": "$CLIENT_ID",
      "access_token": "$ACCESS_TOKEN_1",
      "refresh_token": "$REFRESH_TOKEN_1",
      "expires": 0,
      "owner": {
        "__type": "Pointer",
        "className": "_User",
        "objectId": "$CLIENT_IDIL"
      },
      "createdAt": "2023-01-01T00:00:00.000Z",
      "updatedAt": "2023-01-02T00:00:00.000Z",
      "ACL": {
        "$CLIENT_IDIL": {
          "read": true,
          "write": true
        }
      }
    },
    {
      "objectId": "$OBJECT_ID2",
      "owner": {
        "__type": "Pointer",
        "className": "_User",
        "objectId": "$CLIENT_IDIL"
      },
      "service": "mesh",
      "client_id": "$CLIENT_ID",
      "access_token": "$ACCESS_TOKEN_2",
      "refresh_token": "$REFRESH_TOKEN_2",
      "expires": 0,
      "createdAt": "2022-01-01T00:00:00.000Z",
      "updatedAt": "2022-01-02T00:00:00.000Z",
      "ACL": {
        "$CLIENT_IDIL": {
          "read": true,
          "write": true
        }
      }
    }
  ]
}
```

fsaris, 2023-09-12:
TNX for the feedback. Need to check for the correct service value. Will update the jsfiddle

4D4M-Github, 2023-11-02:
@fsaris 
In this context, my experience might be interesting. I've tried this version of MQTT from the beginning, but always ran into the authentication error above (with the correct user / pass combination). Now I tried again, maybe your improvements solved this too. Same result, so I tried a profile delete and recreate so I got a new mesh password and now it works straight away.

Stefan3005, 2024-01-30:
I think this is still an issue.
I get the ZigBee access_token if I use the awoxh-mesh-credentials-tool.
Furthermore, no multiple services are shown.
I get my access_token manually, but for users which don't know how to do that, that will be a reason to don't use your really cool component.

I think an easy and fast solution should be:
Change in the index.html
```js
    const credentials = response.results[0];

    element.innerHTML = '<pre>' + JSON.stringify({
    'mesh_name': credentials.client_id,
    'mesh_password': credentials.access_token,
    //  'mesh key': credentials.refresh_token,
    }) + '</pre>';
```

to

```js
    let credentials = null;
    for (let i = 0; i < response.results.length; i++) {
    if (response.results[i].service === "mesh") {
        credentials = response.results[i];
        break;
    }
    }

    if (credentials === null) {
    element.innerText = "No mesh credentials found!";
    return;
    } else {
    element.innerHTML =
        "<pre>" +
        JSON.stringify({
        mesh_name: credentials.client_id,
        mesh_password: credentials.access_token,
        //  'mesh key': credentials.refresh_token,
        }) +
        "</pre>";
    }
```

fsaris, 2024-02-18:
all found services are shown now

commit 246d5c4b (2023-11-19): Add tool to get AwoX mesh credentials 

Resolves #28
closed 2024-02-18 by fsaris