fsaris/EspHome-AwoX-BLE-mesh-hub #29: ESP thinks it's connected to the mesh even though it is not

michalpulda, 2023-09-13:
Hello!

I ran into a strange behavior today. The lights were not available in the HA, so I investigated and it seemed to me that the ESP seemed to think it is connected to the mesh, but it was not. After that I restarted the ESP and it connected to the mesh flawlessly. So, I tried a few more restarts to catch the behavior in logs and it finally did it again, so here is the log:

[logs_esp32u-ble-awox_run_not_available.txt](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/12595583/logs_esp32u-ble-awox_run_not_available.txt)

I have created an automation in HA to restart the ESP if the bulb closest (centimeters far from) the ESP is unavailable for a few minutes.

Thanks,
Michal

Jonesbonew, 2023-09-22:
Same problem, ecxept it doesnt always say unavailable so i cant make the automation to restart ESP

fsaris, 2023-09-28:
So it fails to connect to one of the light but thinks it did?

Could you maybe try latest version and check if you still run into this issue?

Jonesbonew, 2023-10-01:
I tried the latest version for a day and so far it works a lot better! Awesome!

michalpulda, 2023-10-02:
I have tried the last version and even though I rebooted the ESP32 multiple times I haven't run into these problems.

What was new in the logs (at least to me) is that when the ESP32 was trying to connect to the mesh, password and user name was not accepted (once even twice in a row). After that it tried to connect again and everything went smoothly.

So, I think it works now, thank you!

closed 2023-10-17 by fsaris