fsaris/EspHome-AwoX-BLE-mesh-hub #31: Compiling of ESPHome fails with 3 errors

dabepp, 2023-09-25:
This is my ESPHome File:
```
 esp32:
  board: wemos_d1_mini32
substitutions:
  name: esp32-bluetooth-proxy-35ba64
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  compile_process_limit: 2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

bluetooth_proxy:
  active: true

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_user
  password: !secret mqtt_password
  
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

api:
  encryption:
    key: "<redacted>"

logger:
  level: debug
  

external_components:
  source: github://fsaris/EspHome-AwoX-BLE-mesh-hub

awox_mesh:
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password
  address_prefix: A4:C1:38:74:7C:A7
  device_info:
    # Example device type, see device info in HomeAssistant or MQTT message to find the 'product_id'
    - product_id: 0x30
      device_type: RGB
      manufacturer: EGLO
      name: EGLOPanel 30X120
      model: EPanel_120
      icon: mdi:wall-sconce-flat
```

This is the error:

```
HARDWARE:` ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.3
Compiling /data/esp32-bluetooth-proxy-35ba64/.pioenvs/esp32-bluetooth-proxy-35ba64/src/esphome/components/awox_mesh/awox_mesh.o
Compiling /data/esp32-bluetooth-proxy-35ba64/.pioenvs/esp32-bluetooth-proxy-35ba64/src/esphome/components/awox_mesh/mesh_device.o
src/esphome/components/awox_mesh/awox_mesh.cpp:1:9: warning: #pragma once in main file
 #pragma once
         ^~~~
In file included from src/esphome/components/awox_mesh/awox_mesh.cpp:7:
src/esphome/components/awox_mesh/awox_mesh.h:8:10: fatal error: AES.h: No such file or directory

*************************************************************
* Looking for AES.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:AES.h"
* Web  > https://registry.platformio.org/search?q=header:AES.h
*
*************************************************************

 #include <AES.h>
          ^~~~~~~
compilation terminated.
*** [/data/esp32-bluetooth-proxy-35ba64/.pioenvs/esp32-bluetooth-proxy-35ba64/src/esphome/components/awox_mesh/awox_mesh.o] Error 1
src/esphome/components/awox_mesh/mesh_device.cpp:8:10: fatal error: AES.h: No such file or directory

*************************************************************
* Looking for AES.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:AES.h"
* Web  > https://registry.platformio.org/search?q=header:AES.h
*
*************************************************************

 #include <AES.h>
          ^~~~~~~
compilation terminated.
*** [/data/esp32-bluetooth-proxy-35ba64/.pioenvs/esp32-bluetooth-proxy-35ba64/src/esphome/components/awox_mesh/mesh_device.o] Error 1
========================= [FAILED] Took 22.44 seconds =========================
```

What can I do?

dabepp, 2023-09-26:
Tried it with another D1 mini, same result. Anybody that can help on that issue?

dabepp, 2023-09-27:
> Let me know if it helps.

Thanks for you reply. Yes it did help, basically what I did is flash another ESP and use the example yaml.
It just got compiled (My R4 with 1 GB memory needs most addons to be shut down in order to compile sucessfully).

So the integration seems to work in general, I only don't have a switch to control the lights lights... :-)

I'll see what the issue is there. Log looks like this:
'''
[11:20:30][C][api:138]: API Server:
[11:20:30][C][api:139]:   Address: 192.168.178.99:6053
[11:20:30][C][api:141]:   Using noise encryption: YES
[11:20:30][C][mqtt:133]: MQTT:
[11:20:30][C][mqtt:135]:   Server Address: 192.168.178.32:1883 (192.168.178.32)
[11:20:30][C][mqtt:136]:   Username: [redacted]
[11:20:30][C][mqtt:137]:   Client ID: [redacted]
[11:20:30][C][mqtt:139]:   Discovery prefix: 'homeassistant'
[11:20:30][C][mqtt:140]:   Discovery retain: YES
[11:20:30][C][mqtt:142]:   Topic Prefix: 'awox-ble-mesh-hub'
[11:20:30][C][mqtt:144]:   Log Topic: 'awox-ble-mesh-hub/debug'
[11:20:30][C][mqtt:147]:   Availability: 'awox-ble-mesh-hub/status'
[11:23:45][I][ota:117]: Boot seems successful, resetting boot loop counter.
[11:23:45][D][esp32.preferences:114]: Saving 1 preferences to flash...
[11:23:45][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[11:23:46][D][esp32_ble_tracker:246]: Starting scan...
[11:23:46][D][AwoxMesh:045]: scan end
[11:28:46][D][esp32_ble_tracker:246]: Starting scan...
[11:28:46][D][AwoxMesh:045]: scan end
[11:33:46][D][esp32_ble_tracker:246]: Starting scan...
[11:33:46][D][AwoxMesh:045]: scan end
[11:38:46][D][esp32_ble_tracker:246]: Starting scan...
[11:38:46][D][AwoxMesh:045]: scan end
[11:43:46][D][esp32_ble_tracker:246]: Starting scan...
[11:43:46][D][AwoxMesh:045]: scan end
[11:48:46][D][esp32_ble_tracker:246]: Starting scan...
[11:48:46][D][AwoxMesh:045]: scan end
[11:53:46][D][esp32_ble_tracker:246]: Starting scan...
[11:53:46][D][AwoxMesh:045]: scan end
[11:58:46][D][esp32_ble_tracker:246]: Starting scan...
[11:58:46][D][AwoxMesh:045]: scan end
'''

michalpulda, 2023-09-27:
It seems that it has not yet connected to the awox mesh.

dabepp, 2023-09-27:
Not sure what did the trick but it is finally working. Gonna parse the yaml later!
[image]

is it normal that the Log in esphome shows all the mqtt traffic?
[image]

fsaris, 2023-09-28:
> I'll see what the issue is there. Log looks like this:

Is/was the device in range of the light? The scan should normally report all found BLE devices

fsaris, 2023-09-28:
@dabepp looks like your first attempt was for framework esp-idf. The code in main wasn't compatible with that yet, it needed the Arduino framework. In de dev branch the code was prepared to support the framework esp-idf and this is now merged to the main branch. 

As you got it running now I will close this issue

closed 2023-09-28 by fsaris