fsaris/EspHome-AwoX-BLE-mesh-hub #33: Eglo Fueva-Z devices not published on MQTT

mtpaasto, 2023-09-30:
I have many Fueva-Z lights ( https://www.eglo.com/uk/recessed-light-fueva-z-900102.html )
Not sure if this is a bug or just me but I can't list lights on MQTT/home assistant. I obtained my mesh name and password with https://jsfiddle.net/fsaris/uncm54vb/37/ only difference being that I chaged URL to https://l4hparse-prod.awox.cloud/parse/login as I am using Awox Home Conrol app instead of Awox smart conrol. Some log attached below. For some reason it shows my mesh network name  instead of proper name.

V][AwoxMesh:044]: Found Awox device <MESH_NAME> - A4:C1:38:0A:AC:EF. RSSI: -88 dB (total devices: 16)
[VV][scheduler:226]: Running timeout 'connecting' with interval=20000 last_execution=20292 (now=40303)
[I][AwoxMesh:071]: Failed to connect A4:C1:38:95:D5:82 => rssi: -55
[I][esp32_ble_client:081]: [0] [A4:C1:38:95:D5:82] Disconnecting.
[I][MeshDevice:155]: DISCONNECTING
[I][mesh_device:475]: Publish connected to mesh device - offline
[D][AwoxMesh:057]: Total devices: 16
[D][AwoxMesh:059]: Available device A4:C1:38:B1:9A:DF => rssi: -59
[D][AwoxMesh:059]: Available device A4:C1:38:95:D5:82 => rssi: -59
[D][AwoxMesh:059]: Available device A4:C1:38:0E:6C:75 => rssi: -59
[D][AwoxMesh:059]: Available device A4:C1:38:FF:E1:D2 => rssi: -59
[D][AwoxMesh:059]: Available device A4:C1:38:56:C1:55 => rssi: -60
[D][AwoxMesh:059]: Available device A4:C1:38:A8:81:49 => rssi: -61
[D][AwoxMesh:059]: Available device A4:C1:38:CC:2F:5F => rssi: -61
[D][AwoxMesh:059]: Available device A4:C1:38:F7:C3:75 => rssi: -63
[D][AwoxMesh:059]: Available device A4:C1:38:B9:23:95 => rssi: -67
[D][AwoxMesh:059]: Available device A4:C1:38:4A:8A:DD => rssi: -68
[D][AwoxMesh:059]: Available device A4:C1:38:97:7B:2E => rssi: -68
[D][AwoxMesh:059]: Available device A4:C1:38:92:4D:15 => rssi: -69
[D][AwoxMesh:059]: Available device A4:C1:38:A1:D8:A0 => rssi: -71
[D][AwoxMesh:059]: Available device A4:C1:38:29:69:37 => rssi: -77
[D][AwoxMesh:059]: Available device A4:C1:38:4A:4F:7B => rssi: -79
[D][AwoxMesh:059]: Available device A4:C1:38:0A:AC:EF => rssi: -88
[I][AwoxMesh:063]: Try to connect A4:C1:38:B1:9A:DF => rssi: -59
[I][mesh_device:475]: Publish connected to mesh device - offline
[I][esp32_ble_client:064]: [0] [A4:C1:38:B1:9A:DF] 0x00 Attempting BLE connection
[I][MeshDevice:175]: CONNECTING
[VV][scheduler:032]: set_timeout(name='connecting', timeout=20000)
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
[VV][esp32_ble_tracker:395]: Parse Result:
[VV][esp32_ble_tracker:412]:   Address: A4:C1:38:4A:4F:7B (PUBLIC)
[VV][esp32_ble_tracker:414]:   RSSI: -87
[VV][esp32_ble_tracker:415]:   Name: '<MESH_NAME>'
[VV][esp32_ble_tracker:423]:   Ad Flag: 5
[VV][esp32_ble_tracker:426]:   Service UUID: 0x1800
[VV][esp32_ble_tracker:429]:   Manufacturer data: 35.10.7B.4F.4A.38 (6)
[VV][esp32_ble_tracker:445]: Adv data: 02.01.05.03.02.00.18.09.09.69.52.74.55.42.58.4C.41.09.FF.60.01.35.10.7B.4F.4A.38 (27)
[V][AwoxMesh:044]: Found Awox device <MESH_NAME> - A4:C1:38:4A:4F:7B. RSSI: -87 dB (total devices: 16)
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
[VV][esp32_ble_tracker:395]: Parse Result:
[VV][esp32_ble_tracker:412]:   Address: 54:15:89:A8:84:60 (PUBLIC)
[VV][esp32_ble_tracker:414]:   RSSI: -70
[VV][esp32_ble_tracker:415]:   Name: ''
[VV][esp32_ble_tracker:423]:   Ad Flag: 24
[VV][esp32_ble_tracker:429]:   Manufacturer data: 62.20.01.D3.E4.00.00.28.00.00.00.54.15.89.A8.84.60.00.00.00.00.00.00.00 (24)
[VV][esp32_ble_tracker:445]: Adv data: 02.01.18.1B.FF.CB.0E.62.20.01.D3.E4.00.00.28.00.00.00.54.15.89.A8.84.60.00.00.00.00.00.00.00 (31)
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
[VV][esp32_ble_tracker:395]: Parse Result:
[VV][esp32_ble_tracker:412]:   Address: A4:C1:38:4A:4F:7B (PUBLIC)
[VV][esp32_ble_tracker:414]:   RSSI: -81
[VV][esp32_ble_tracker:415]:   Name: '<MESH_NAME>'
[VV][esp32_ble_tracker:423]:   Ad Flag: 5
[VV][esp32_ble_tracker:426]:   Service UUID: 0x1800
[VV][esp32_ble_tracker:429]:   Manufacturer data: 35.10.7B.4F.4A.38 (6)
[VV][esp32_ble_tracker:445]: Adv data: 02.01.05.03.02.00.18.09.09.69.52.74.55.42.58.4C.41.09.FF.60.01.35.10.7B.4F.4A.38 (27)
[V][AwoxMesh:044]: Found Awox device <MESH_NAME> - A4:C1:38:4A:4F:7B. RSSI: -81 dB (total devices: 16)
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
[VV][esp32_ble_tracker:395]: Parse Result:
[VV][esp32_ble_tracker:412]:   Address: 58:2B:DB:6B:B8:A6 (PUBLIC)
[VV][esp32_ble_tracker:414]:   RSSI: -92
[VV][esp32_ble_tracker:415]:   Name: 'PAX Calima'
[VV][esp32_ble_tracker:417]:   TX Power: 2
[VV][esp32_ble_tracker:423]:   Ad Flag: 6
[VV][esp32_ble_tracker:426]:   Service UUID: 0x180A
[VV][esp32_ble_tracker:445]: Adv data: 02.01.06.03.03.0A.18.0B.09.50.41.58.20.43.61.6C.69.6D.61.02.0A.03 (22)
[VV][esp32_ble_tracker:395]: Parse Result:
[... 43 lines cut ...]
[VV][esp32_ble_tracker:423]:   Ad Flag: 5
[VV][esp32_ble_tracker:426]:   Service UUID: 0x1800
[VV][esp32_ble_tracker:429]:   Manufacturer data: 97.20.37.69.29.38 (6)
[VV][esp32_ble_tracker:445]: Adv data: 02.01.05.03.02.00.18.09.09.69.52.74.55.42.58.4C.41.09.FF.60.01.97.20.37.69.29.38 (27)
[V][AwoxMesh:044]: Found Awox device <MESH_NAME> - A4:C1:38:29:69:37. RSSI: -72 dB (total devices: 16)
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
[VV][esp32_ble_tracker:395]: Parse Result:
[VV][esp32_ble_tracker:412]:   Address: 80:64:6F:E4:6A:4E (PUBLIC)
[VV][esp32_ble_tracker:414]:   RSSI: -84
[VV][esp32_ble_tracker:415]:   Name: ''
[VV][esp32_ble_tracker:423]:   Ad Flag: 6
[VV][esp32_ble_tracker:429]:   Manufacturer data: 01.05.00.0B.01.10.0A.80.64.6F.E4.6A.4C (13)
[VV][esp32_ble_tracker:445]: Adv data: 02.01.06.10.FF.A9.0B.01.05.00.0B.01.10.0A.80.64.6F.E4.6A.4C (20)
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
[VV][esp32_ble_tracker:395]: Parse Result:
[VV][esp32_ble_tracker:412]:   Address: A4:C1:38:29:69:37 (PUBLIC)
[VV][esp32_ble_tracker:414]:   RSSI: -73
[VV][esp32_ble_tracker:415]:   Name: '<MESH_NAME>'
[VV][esp32_ble_tracker:423]:   Ad Flag: 5
[VV][esp32_ble_tracker:426]:   Service UUID: 0x1800
[VV][esp32_ble_tracker:429]:   Manufacturer data: 97.20.37.69.29.38 (6)
[VV][esp32_ble_tracker:445]: Adv data: 02.01.05.03.02.00.18.09.09.69.52.74.55.42.58.4C.41.09.FF.60.01.97.20.37.69.29.38 (27)
[V][AwoxMesh:044]: Found Awox device <MESH_NAME> - A4:C1:38:29:69:37. RSSI: -73 dB (total devices: 16)
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
[V][esp32_ble:178]: (BLE) gap_event_handler - 3

fsaris, 2023-09-30:
The Z product range are Zigbee lights you could use https://www.zigbee2mqtt.io/supported-devices/#v=EGLO,Eglo,AwoX with a Zigbee controller

mtpaasto, 2023-09-30:
These support both BT and Zigbee. I have tried to control them with home assistant and it works but with that it disables usage of BT mesh and I loose control over them with my Eglo  BT remote. So I would like to control them from Home assistant with Bt.

fsaris, 2023-09-30:
Are you sure they support the same ble mesh? If I remember correctly from previous research it's a different protocol

mtpaasto, 2023-09-30:
My understanding is that it supports both zigbee and bt mesh. I can control it with my phone and in the application it mentiones mesh. 
Esp32 seems to find devices but I am not sure why it fails to parse name correctly?
[D][AwoxMesh:057]: Total devices: 11
[D][AwoxMesh:059]: Available device A4:C1:38:B9:23:95 => rssi: -78
[D][AwoxMesh:059]: Available device A4:C1:38:95:D5:82 => rssi: -79
[D][AwoxMesh:059]: Available device A4:C1:38:F7:C3:75 => rssi: -81
[D][AwoxMesh:059]: Available d[D][AwoxMesh:057]: Total devices: 11
[D][AwoxMesh:059]: Available device A4:C1:38:B9:23:95 => rssi: -78
[D][AwoxMesh:059]: Available device A4:C1:38:95:D5:82 => rssi: -79
[D][AwoxMesh:059]: Available device A4:C1:38:F7:C3:75 => rssi: -81
[D][AwoxMesh:059]: Available device A4:C1:38:A8:81:49 => rssi: -83
[D][AwoxMesh:059]: Available device A4:C1:38:4A:4F:7B => rssi: -83
[D][AwoxMesh:059]: Available device A4:C1:38:A1:D8:A0 => rssi: -85
[D][AwoxMesh:059]: Available device A4:C1:38:0E:6C:75 => rssi: -86
[D][AwoxMesh:059]: Available device A4:C1:38:0A:AC:EF => rssi: -89
[D][AwoxMesh:059]: Available device A4:C1:38:56:C1:55 => rssi: -90
[D][AwoxMesh:059]: Available device A4:C1:38:FF:E1:D2 => rssi: -92
[D][AwoxMesh:059]: Available device A4:C1:38:29:69:37 => rssi: -93
[I][AwoxMesh:063]: Try to connect A4:C1:38:B9:23:95 => rssi: -78
[I][mesh_device:475]: Publish connected to mesh device - offline
[I][esp32_ble_client:064]: [0] [A4:C1:38:B9:23:95] 0x00 Attempting BLE connection
[I][MeshDevice:175]: CONNECTING
[VV][scheduler:032]: set_timeout(name='connecting', timeout=20000)
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
evice A4:C1:38:A8:81:49 => rssi: -83
[D][AwoxMesh:059]: Available device A4:C1:38:4A:4F:7B => rssi: -83
[D][AwoxMesh:059]: Available device A4:C1:38:A1:D8:A0 => rssi: -85
[D][AwoxMesh:059]: Available device A4:C1:38:0E:6C:75 => rssi: -86
[D][AwoxMesh:059]: Available device A4:C1:38:0A:AC:EF => rssi: -89
[D][AwoxMesh:059]: Available device A4:C1:38:56:C1:55 => rssi: -90
[D][AwoxMesh:059]: Available device A4:C1:38:FF:E1:D2 => rssi: -92
[D][AwoxMesh:059]: Available device A4:C1:38:29:69:37 => rssi: -93
[I][AwoxMesh:063]: Try to connect A4:C1:38:B9:23:95 => rssi: -78
[I][mesh_device:475]: Publish connected to mesh device - offline
[I][esp32_ble_client:064]: [0] [A4:C1:38:B9:23:95] 0x00 Attempting BLE connection
[I][MeshDevice:175]: CONNECTING
[VV][scheduler:032]: set_timeout(name='connecting', timeout=20000)
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
[V][esp32_ble:178]: (BLE) gap_event_handler - 3

mtpaasto, 2023-10-01:
Ok, not sure if this is related but I can see following GATT related traces which might be related.
^[[0;37m[V][esp32_ble_client:069]: [0] [A4:C1:38:B9:23:95]  characteristic 00010203-0405-0607-0809-0A0B0C0D1911, handle 0x12, properties 0x1a^[[0m
^[[0;37m[V][esp32_ble_client:069]: [0] [A4:C1:38:B9:23:95]  characteristic 00010203-0405-0607-0809-0A0B0C0D1912, handle 0x15, properties 0xe^[[0m
^[[0;37m[V][esp32_ble_client:069]: [0] [A4:C1:38:B9:23:95]  characteristic 00010203-0405-0607-0809-0A0B0C0D1913, handle 0x18, properties 0xa^[[0m
^[[0;37m[V][esp32_ble_client:069]: [0] [A4:C1:38:B9:23:95]  characteristic 00010203-0405-0607-0809-0A0B0C0D1915, handle 0x1b, properties 0x6^[[0m
^[[0;37m[V][esp32_ble_client:069]: [0] [A4:C1:38:B9:23:95]  characteristic 00010203-0405-0607-0809-0A0B0C0D1914, handle 0x1e, properties 0xa^[[0m

^[[0;33m[W][esp32_ble_client:215]: [0] [A4:C1:38:B9:23:95] esp_ble_gattc_get_descr_by_char_handle error, status=10^[[0m

I think 00010203-0405-0607-0809-0A0B0C0D1915 is not defined in mesh_device.h

fsaris, 2023-10-01:
Okay, did some more reading looks like the Z products use Zigbee direct (https://csa-iot.org/wp-content/uploads/2022/12/20-27688-037-zigbee_direct_spec.pdf). That's a different protocol as telink that's used in this esphome component.

Looks like more vendors use this like MOES and IKEA (see for instance https://github.com/Koenkk/zigbee2mqtt/issues/18335). 

So I think it should be possible to use the devices over Zigbee and link your remote. Or are you using a BLE mesh remote?

mtpaasto, 2023-10-02:
Ok, you might be right. I can add lights directly to home assistant with ZHA but I have found it more unreliable and slower than controlling them with Eglo remote (zigbee) or Phone (BLE). 
Perhaps one possibility would be to upgrade my Zigbee coordinator HW from CC2531 to something more powerful or try if Zigbee2mqtt would work better.