fsaris/EspHome-AwoX-BLE-mesh-hub #34: [Feature request] MAC address whitelist

oxigen-dev, 2023-10-25:
Hi @fsaris ,

I'm facing with yet another strange issue. As you might know, I have a lot of Eglo ceilings and 1 Eglo bulb.
I'm using an ESP32 with external antenna to control the lamps. If only one lamp is turned on, then no matter which is that, it will work fine. The problem comes if multiple ceilings are turned on. The ESP connects to one of them and it tries to control all of the lamps using BT MESH. The issue is that the ESP has a wider range compared to the ceilings (or is in a better position in the house), and some ceilings which the ESP is connected to can't reach other ceilings in the house.

I've created a sketch for better understanding.
[image]

For example if only C1 or C7 is turned on, they are working fine. If both are turned on then the ESP connects to eg. C1 and C7 will be shown in HA as Unavailable. If I turn off C1 then C7 gets Online.

I was thinking to install another ESP close to C7 but I think some small change in the code would be still needed (otherwise if I would turn on C7 first then maybe the already available ESP would connect to that, and the second one would not be able to do so).

Now we have in the yaml file "address_prefix". I'm wondering whether this could be modified to handle a list of MAC addresses (aka whitelist) where we can define which lamps the ESP is allowed to connect to.

I'm also open for other good ideas, if you have one :)

fsaris, 2023-11-04:
A Mac list will be possible. 

Not sure if this will work if both groups of lights are part of the same mesh, maybe you need to setup 2 different mesh networks to let this work properly. Because if a device in the mesh knows a different device exits it will keep reporting its state. Online/offline etc. So the 2 groups of lights shouldn't be in reach of each other or be 2 different mesh networks

oxigen-dev, 2023-11-04:
@fsaris So if I have to create 2 mesh networks anyway, then no modification in the code is necessary, right? 😁

fsaris, 2023-11-19:
@oxigen-dev I think you still will need the list. Or at least it will speed up the connection process. Else it could try to connect to a device of the other mesh and that will fail and take some time to again try to connect to a different device.

oxigen-dev, 2023-11-20:
@fsaris Oh, I see. I would be happy to test a version if one would be available.
Still, how can I create multiple MESH networks is not completely clear for me. If I'd need to reset some ceilings, create new accounts etc. then this wouldn't be very user friendly (especially the multiple accounts part). I would appreciate hearing your thoughts.

fsaris, 2024-01-14:
@oxigen-dev the allowed list is now part of dev branch

closed 2024-01-14 by fsaris