fsaris/EspHome-AwoX-BLE-mesh-hub #35: Can't connect to lights/connect them to home assistant

ogitmb, 2023-11-10:
Hi I am trying and trying but can't get it to work...
No new devices are showing up in Home Assistant.
Any clue on what i could try or check based on my log-files?

Any ideas welcome :-)

I should have two lights within range:
    - product_id: 0x41
      device_type: RGB
      name: EGLO Surface round
      model: EFueva_300r
      manufacturer: EGLO
    - product_id: 0x3F
      device_type: RGB
      name: EGLO Surface round
      model: EFueva_225r
      manufacturer: EGLO

**LOGFILE ESP32-Device**
[logs_awox_logs (4).txt](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/13323718/logs_awox_logs.4.txt)

I

> NFO ESPHome 2023.10.6
> INFO Reading configuration /config/esphome/awox.yaml...
> INFO Starting log output from awox-ble-mesh-hub/debug
> INFO Connected to MQTT broker!
> [00:04:09][C][mqtt:147]:   Availability: 'awox-ble-mesh-hub/status'
> [00:04:59][I][mqtt:274]: MQTT Connected!
> [00:04:59][I][app:062]: setup() finished successfully!
> [00:04:59][I][app:102]: ESPHome version 2023.10.6 compiled on Nov 10 2023, 23:26:03
> [00:04:59][C][status_led:019]: Status LED:
> [00:04:59][C][status_led:020]:   Pin: GPIO2
> [00:04:59][C][wifi:546]: WiFi:
> [00:04:59][C][wifi:382]:   Local MAC: 08:D1:F9:CE:E6:98
> [00:04:59][C][wifi:383]:   SSID: 'New_Paradise'[redacted]
> [00:04:59][C][wifi:384]:   IP Address: 192.168.178.170
> [00:04:59][C][wifi:386]:   BSSID: F0:B0:14:62:47:7F[redacted]
> [00:04:59][C][wifi:387]:   Hostname: 'awox-ble-mesh-hub'
> [00:04:59][C][wifi:389]:   Signal strength: -64 dB ▂▄▆█
> [00:04:59][C][wifi:393]:   Channel: 6
> [00:04:59][C][wifi:394]:   Subnet: 255.255.255.0
> [00:04:59][C][wifi:395]:   Gateway: 192.168.178.1
> [00:04:59][C][wifi:396]:   DNS1: 192.168.178.1
> [00:04:59][C][wifi:397]:   DNS2: 0.0.0.0
> [00:04:59][C][logger:416]: Logger:
> [00:04:59][C][logger:417]:   Level: DEBUG
> [00:04:59][C][logger:418]:   Log Baud Rate: 115200
> [00:04:59][C][logger:420]:   Hardware UART: UART0
> [00:04:59][C][esp32_ble:238]: ESP32 BLE:
> [00:04:59][C][esp32_ble:240]:   MAC address: 08:D1:F9:CE:E6:9A
> [00:04:59][C][esp32_ble:241]:   IO Capability: none
> [00:04:59][C][esp32_ble_tracker:617]: BLE Tracker:
> [00:04:59][C][esp32_ble_tracker:618]:   Scan Duration: 300 s
> [00:04:59][C][esp32_ble_tracker:619]:   Scan Interval: 320.0 ms
> [00:04:59][C][esp32_ble_tracker:620]:   Scan Window: 30.0 ms
> [00:04:59][C][esp32_ble_tracker:621]:   Scan Type: PASSIVE
> [00:04:59][C][esp32_ble_tracker:622]:   Continuous Scanning: True
> [00:05:00][C][mdns:115]: mDNS:
> [00:05:00][C][mdns:116]:   Hostname: awox-ble-mesh-hub
> [00:05:00][C][ota:097]: Over-The-Air Updates:
> [00:05:00][C][ota:098]:   Address: awox-ble-mesh-hub.local:3232
> [00:05:00][C][ota:101]:   Using Password.
> [00:05:00][W][ota:107]: Last Boot was an unhandled reset, will proceed to safe mode in 3 restarts
> [00:05:00][C][mqtt:133]: MQTT:
> [00:05:00][C][mqtt:135]:   Server Address: 192.168.178.171:1883 (192.168.178.171)
> [00:05:00][C][mqtt:136]:   Username: 'mqttAwox'[redacted]
> [00:05:00][C][mqtt:137]:   Client ID: 'awox-ble-mesh-hub-08d1f9cee698'[redacted]
> [00:05:00][C][mqtt:139]:   Discovery prefix: 'homeassistant'
> [00:05:00][C][mqtt:140]:   Discovery retain: YES
[... 160 lines cut ...]
> [00:05:33][C][mqtt:135]:   Server Address: 192.168.178.171:1883 (192.168.178.171)
> [00:05:33][C][mqtt:136]:   Username: 'mqttAwox'[redacted]
> [00:05:33][C][mqtt:137]:   Client ID: 'awox-ble-mesh-hub-08d1f9cee698'[redacted]
> [00:05:33][C][mqtt:139]:   Discovery prefix: 'homeassistant'
> [00:05:33][C][mqtt:140]:   Discovery retain: YES
> [00:05:33][C][mqtt:142]:   Topic Prefix: 'awox-ble-mesh-hub'
> [00:05:33][C][mqtt:144]:   Log Topic: 'awox-ble-mesh-hub/debug'
> [00:05:33][C][mqtt:147]:   Availability: 'awox-ble-mesh-hub/status'
> [00:05:44][D][AwoxMesh:056]: Total devices: 2
> [00:05:55][D][AwoxMesh:058]: Available device A4:C1:38:79:28:63 => rssi: -77
> [00:05:55][D][AwoxMesh:058]: Available device A4:C1:38:A1:19:17 => rssi: -93
> [00:05:55][I][AwoxMesh:062]: Try to connect A4:C1:38:79:28:63 => rssi: -77
> [00:05:55][I][mesh_device:478]: Publish connected to mesh device - offline
> [00:05:55][I][MeshDevice:159]: IDLE
> [00:05:55][D][esp32_ble_client:048]: [0] [A4:C1:38:79:28:63] Found device
> [00:05:55][I][MeshDevice:167]: DISCOVERED
> [00:05:55][D][esp32_ble_tracker:214]: Pausing scan to make connection...
> [00:05:55][I][MeshDevice:171]: READY_TO_CONNECT
> [00:05:55][I][esp32_ble_client:064]: [0] [A4:C1:38:79:28:63] 0x00 Attempting BLE connection
> [00:05:55][I][MeshDevice:175]: CONNECTING
> [00:05:55][D][esp-idf:000]: W (21608) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
> 
> [00:05:55][D][esp-idf:000]: W (24581) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
> 
> [00:05:55][D][esp-idf:000]: W (27308) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

ogitmb, 2023-11-10:
This is what my yaml looks like:

esphome:
  name: awox-ble-mesh-hub
  friendly_name: "AwoX BLE mesh hub"

esp32:
  # Match this with the board your using
  board: nodemcu-32s
  framework:
    # Be sure to "Clean Build Files" when switching framework
    type: esp-idf

status_led:
  # Match this with the board your using
  pin: GPIO2

# Enable logging
logger:
  #level: VERY_VERBOSE

# debug:
#   update_interval: 5s

# text_sensor:
#   - platform: debug
#     reset_reason:
#      name: "Reset Reason"

# sensor:
#   - platform: debug
#     free:
#       name: "Heap Free"
#     loop_time:
#       name: "Loop Time"

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_user_Awox
  password: !secret mqtt_password

# Disable Home Assistant API to prevent frequent MQTT disconnections
#api:

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub
  #  ref: dev
  # - source:
  #     type: local
  #     path: components

esp32_ble_tracker:
  scan_parameters:
    active: false

awox_mesh:
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password
  address_prefix: A4:C1
  device_info:
    # Example device type, see device info in HomeAssistant or MQTT message to find the 'product_id'
    - product_id: 0x41
      device_type: RGB
      name: EGLO Surface round
      model: EFueva_300r
      manufacturer: EGLO
      icon: mdi:ceiling-light
    - product_id: 0x3F
      device_type: RGB
      name: EGLO Surface round
      model: EFueva_225r
      manufacturer: EGLO
      icon: mdi:ceiling-light

fsaris, 2023-11-11:
Do you have the correct board type configured for the device your using?

```
esp32:
  # Match this with the board your using
  board: nodemcu-32s
```

The devices are recognized, but connection fails

fsaris, 2023-11-11:
Also try to enable verbose logging maybe it tells you more about why it fails to connect

ogitmb, 2023-11-11:
thank you so much for taking the time to look over it - I am happy to have someone working on the awox integration 

The board was actually one of the issues, after i changed it (and also created a new awox account, did a factory reset on the lights and added them to the new awox account) it did kind of work for a moment.
It connected to one of the lights and I was able to see it in the mqtt integration and added it to home assistant and could turn on/off. After a few minutes the connection to that light was lost but in that moment the other one connected and i could add and control it as well.

Log: 
[logs_awox_logs_2200_connected with One,then dropped it and connected to the other one.txt](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/13326706/logs_awox_logs_2200_connected.with.One.then.dropped.it.and.connected.to.the.other.one.txt)

But after that i had the lights turned off and also had the esp32 restarted (as i unplugged it from the pc to connect it directly to a power plug). i could not get it to connect again at all.
I recompiled the yaml with very_verboose logging to see the details but i'm not really identifying the underlying issue :-/
It does always find both lights but the conenctions keeps failing 
[logs_awox_logs_2328_verboosenotworkinganymore.txt](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/13326710/logs_awox_logs_2328_verboosenotworkinganymore.txt)

Any sugggestions?

fsaris, 2023-11-12:
It looks to be out of range to make a connection. The esp sees the BLE devices (you gat a rssi value for each device) but it's not enough to connect. Most esphome antennes aren't that great 😬

Try moving it nearer to one of the devices