fsaris/EspHome-AwoX-BLE-mesh-hub #43: Eglo Panel 10x120 is recognized as dimmable bulb

Raphcmx, 2023-12-27:
device_info:
    # Example device type, see device info in HomeAssistant or MQTT message to find the 'product_id'
    product_id: 0x7E
    device_type: RGB
    name: EGLOPanel 10X120
    model: EPanel_120_10
    manufacturer: EGLO

even though the fitting device configuration is used the device gets recognized as "unknown" so it functions as a dimmable light

fsaris, 2023-12-27:
Can you post some debug logging and your yaml config?

Raphcmx, 2023-12-27:
Yes of course I forgot. I'm new to home assistant so i dont know if this is enough I found these warnings in the logger. Or do you need the whole installation process log? For better understanding, it recognizes the light just fine but as a dimmable so it is not possible to set the color of the light in any form. Also took out the secrets in the txt file in my configuration there are the right secrets.

[yaml.txt](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/13780204/yaml.txt)

[logger.txt](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/13780226/logger.txt)

fsaris, 2023-12-28:
The logs look like HA logs. I'm interested in the logs from esphome

Raphcmx, 2023-12-28:
What settings do I need to change so I can see older logs?

Now I am only able to see this: 

INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome/bluetooth-proxy-buro.yaml...
INFO Updating https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub@None
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from /dev/ttyUSB1 with baud rate 115200
[22:42:31][D][mesh_device:353]: online status report: mesh: 13379, on: 1, color_mode: 0, transition_mode: 0, w_b: 127, temp: 102, c_b: 100, rgb: 3EDFCF 
[22:42:31][I][mesh_device:418]: 13379: ON temp: 102 (127 %) ONLINE

I think thats not what you need :D

fsaris, 2023-12-29:
Older logs are not kept by esphome.

I'm interested in the logs until it connects to your light and reports the device info over mqtt.

Open the log view and restart the esp device. Wait until it only logs the `online status report`. That should give enough info.

Did you reinstall after updating the yaml?

You should disable the `status_led` in your yaml because the used pin doesn't match your board.