fsaris/EspHome-AwoX-BLE-mesh-hub #48: MQTT doesn't find my devices...

davidmcgh, 2024-01-02:
Hi! I'm totally new to this topic. In the yaml file of the esp32 module I added the custom component via git link. After that the components were downloaded. I modified the components source to local components in the nodemcu.yaml and I typed in the path of the downloaded files. Then I modified the awox-ble-mesh-hub.yaml which is in the folder of the components. Nothing is happening. I don't know what I'm doing wrong.

Here are my yaml files:
nodemcu.yaml = yaml file of the esp32 device (I use nodemcu esp-32s)
awox-ble-mesh-hub = file in the components folder

[nodemcu_yaml.txt](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/13807751/nodemcu_yaml.txt)
[awox-ble-mesh-hub_yaml.txt](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/13807754/awox-ble-mesh-hub_yaml.txt)

fsaris, 2024-01-02:
1. No need to change the source to local.
2. You are missing parts of the config. The tracker and AwoX config is needed https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/blob/9fc9c5873c53f712fa4d92f9f6f516f68acda204/awox-ble-mesh-hub.yaml#L59

fsaris, 2024-01-02:
Ah, it's part of your second yaml

Can you provide some logs?

davidmcgh, 2024-01-02:
How to view the log?

davidmcgh, 2024-01-02:
Success! Thanks!

closed 2024-01-02 by davidmcgh

fsaris, 2024-01-02:
Nice that it works now. What was the issue/solution?