fsaris/EspHome-AwoX-BLE-mesh-hub #56: Error: The program size is greater than maximum allowed

mykalanui, 2024-01-17:
[image]

I'm getting this Error when trying to install the following yaml:
```yaml
esphome:
  name: esphome-awox-mesh
  friendly_name: AwoX Mesh ESP32

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  #level: VERY_VERBOSE

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_user
  password: !secret mqtt_password

# Disable Home Assistant API to prevent frequent MQTT disconnections
#api:

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub

esp32_ble_tracker:
  scan_parameters:
    active: false

awox_mesh:
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password

  device_info:
    # Example device type, see device info in HomeAssistant or MQTT message to find the 'product_id'
    -  product_id: 0x3F
       device_type: RGB
       name: EGLO Surface round
       model: EFueva_225r
       manufacturer: EGLO
``` 

Can anyone please help me with this?

fsaris, 2024-01-18:
Don't use framework arduino

```
esp32:
  ....
  framework:
    # Be sure to "Clean Build Files" when switching framework
    type: esp-idf
```

That will decrease the size and improve functionality.

What board are you exactly using? Maybe it's not 100% correct and is it limited in size now by this setting.

mykalanui, 2024-01-18:
This has done the trick. Thanks for your Hel

> Don't use framework arduino
>
> ```
> esp32:
>   ....
>   framework:
>     # Be sure to "Clean Build Files" when switching framework
>     type: esp-idf
> ```
>
> That will decrease the size and improve [...]

This has done the trick. Thank you for your help and for the quick reply.
Would you suggest to always use framework-typ "esp-idf" for HA-related things? "arduino" comes as stock from ESPHome.

However, now that everything is set up (ESP32, MQTT Broker, MQTT Integration) I don't receive any mqtt messages. 
Have you got any hints for me?

mykalanui, 2024-01-18:
I have just tried updating the yaml with "very verbose logs" and install, and was hit with quite a few errors. Can you make any sense of it?

```
INFO ESPHome 2023.12.7
INFO Reading configuration /config/esphome/awox-mesh-esp32.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-awox-mesh (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Creating a new virtual environment for IDF Python dependencies
Installing ESP-IDF's Python dependencies
Collecting urllib3<2
  Downloading urllib3-1.26.18-py2.py3-none-any.whl (143 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 143.8/143.8 kB 1.7 MB/s eta 0:00:00
Collecting cryptography<35.0.0,>=2.1.4
  Downloading cryptography-3.4.8-cp36-abi3-manylinux_2_24_x86_64.whl (3.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.0/3.0 MB 30.0 MB/s eta 0:00:00
Collecting future>=0.18.3
  Downloading future-0.18.3.tar.gz (840 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 840.9/840.9 kB 60.9 MB/s eta 0:00:00
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting pyparsing<2.4.0,>=2.0.3
  Downloading pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.8/61.8 kB 44.7 MB/s eta 0:00:00
Collecting kconfiglib~=13.7.1
  Downloading kconfiglib-13.7.1-py2.py3-none-any.whl (145 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 145.7/145.7 kB 36.9 MB/s eta 0:00:00
Collecting idf-component-manager~=1.0
  Downloading idf_component_manager-1.4.1-py2.py3-none-any.whl (140 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 140.7/140.7 kB 39.2 MB/s eta 0:00:00
Collecting cffi>=1.12
  Downloading cffi-1.16.0-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (464 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 464.8/464.8 kB 18.6 MB/s eta 0:00:00
Collecting packaging
  Downloading packaging-23.2-py3-none-any.whl (53 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 53.0/53.0 kB 34.0 MB/s eta 0:00:00
Collecting requests<3
  Downloading requests-2.31.0-py3-none-any.whl (62 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 62.6/62.6 kB 42.7 MB/s eta 0:00:00
Collecting requests-file
  Downloading requests_file-1.5.1-py2.py3-none-any.whl (3.7 kB)
Collecting requests-toolbelt
  Downloading requests_toolbelt-1.0.0-py2.py3-none-any.whl (54 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 54.5/54.5 kB 31.8 MB/s eta 0:00:00
Collecting schema
  Downloading schema-0.7.5-py2.py3-none-any.whl (17 kB)
Collecting six
  Downloading six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting tqdm<5
  Downloading tqdm-4.66.1-py3-none-any.whl (78 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 78.3/78.3 kB 28.2 MB/s eta 0:00:00
Collecting colorama
  Downloading colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Collecting pyyaml>5.2
  Downloading PyYAML-6.0.1-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (757 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 757.7/757.7 kB 45.9 MB/s eta 0:00:00
Collecting cachecontrol[filecache]>0.12.6
  Downloading cachecontrol-0.13.1-py3-none-any.whl (22 kB)
Collecting contextlib2>0.6.0
  Downloading contextlib2-21.6.0-py2.py3-none-any.whl (13 kB)
Collecting click
  Downloading click-8.1.7-py3-none-any.whl (97 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 97.9/97.9 kB 31.0 MB/s eta 0:00:00
[... 73 lines cut ...]
src/esphome/components/awox_mesh/helpers.h:9:20: warning: 'std::__cxx11::string esphome::awox_mesh::string_as_binary_string(std::__cxx11::string)' defined but not used [-Wunused-function]
 static std::string string_as_binary_string(std::string words) {
                    ^~~~~~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/esphome-awox-mesh/src/esphome/components/awox_mesh/mesh_connection.o
In file included from src/esphome/components/awox_mesh/device.h:8,
                 from src/esphome/components/awox_mesh/mesh_connection.h:13,
                 from src/esphome/components/awox_mesh/awox_mesh.h:15,
                 from src/esphome/components/awox_mesh/awox_mesh.cpp:7:
src/esphome/components/awox_mesh/helpers.h: At global scope:
src/esphome/components/awox_mesh/helpers.h:17:20: warning: 'std::__cxx11::string esphome::awox_mesh::int_as_hex_string(unsigned char, unsigned char, unsigned char)' defined but not used [-Wunused-function]
 static std::string int_as_hex_string(unsigned char hex1, unsigned char hex2, unsigned char hex3) {
                    ^~~~~~~~~~~~~~~~~
src/esphome/components/awox_mesh/helpers.h:9:20: warning: 'std::__cxx11::string esphome::awox_mesh::string_as_binary_string(std::__cxx11::string)' defined but not used [-Wunused-function]
 static std::string string_as_binary_string(std::string words) {
                    ^~~~~~~~~~~~~~~~~~~~~~~
cc1plus: some warnings being treated as errors
*** [.pioenvs/esphome-awox-mesh/src/esphome/components/awox_mesh/awox_mesh.o] Error 1
In file included from src/esphome/components/awox_mesh/device.h:8,
                 from src/esphome/components/awox_mesh/mesh_connection.h:13,
                 from src/esphome/components/awox_mesh/mesh_connection.cpp:5:
src/esphome/components/awox_mesh/helpers.h:17:20: warning: 'std::__cxx11::string esphome::awox_mesh::int_as_hex_string(unsigned char, unsigned char, unsigned char)' defined but not used [-Wunused-function]
 static std::string int_as_hex_string(unsigned char hex1, unsigned char hex2, unsigned char hex3) {
                    ^~~~~~~~~~~~~~~~~
========================= [FAILED] Took 119.39 seconds =========================
```

fsaris, 2024-01-19:
Try cleaning the build files and try again with wire connected. If normal debug logging build that would be enough for getting first idea of what's going wrong.

With wire connected check the logs. Are you sure the mqtt credentials are correct? Logging over wifi uses mqtt as transport.

You should have seen at least the wifi config in the log output.

mykalanui, 2024-01-20:
When I mentioned wired connection I meant it's wired to my laptop, not the home server. I will try hooking it up to the server later. In the meantime I have rebuilt the files, did all Updates on HA, reinstalled the yaml via web.esphome.
Now Updating over Wifi compiles and uploads fine, as opposed to my last post. So far so good. 

I am still not getting any Signals from my EGLO-Lamps however. The log ist still the same as 2 posts ago.
Credentials check out.

The original issue is solved, would you like to close this or is it OK to continue here?
Your help is very much appreciated!

What would you suggest as the next step for debugging?

mykalanui, 2024-01-20:
I will briefly describe my setup:

There's two EGLO Fueva C 225mm Lamps here.
They've been running for two years now.
They're connected to the AwoX app on my phone, no updates pending, working fine from there.
I have also got a remote which I don't use.

For ESPHome I'm running an ESP32 dev board (ESP-32-WROOM-32 with CP2102), another board from the same batch is working fine on a different project (BT-Proxy, not MQTT-related).

fsaris, 2024-01-20:
Are you sure the mqtt settings are correct?

The esp needs to be able to connect to the mqtt server else debug over wifi doesn't work.

If still doesn't work connect over cable (should also be possible from any laptop that can connect with the esphome dashboard) and check the log output 

[image]

mykalanui, 2024-01-21:
You were right (of course): I had to change the mqtt host (the default "core-mosquitto" did not work). I'm now using HA's static IP address. 
Reading the log through ESP Dashboard - as proposed by you - helped figuring it out.

Now I'm one step closer, but the ESP seems to be in a boot loop. See current log (excerpts):
```
[I][awox.mesh:441]: Publish connected to mesh device - offline
[D][awox.mesh:266]: Total devices: 2
[D][awox.mesh:269]: Available device A4:C1:38:A3:29:9E [0] => rssi: -68
[D][awox.mesh:269]: Available device A4:C1:38:14:C3:B7 [0] => rssi: -71
[D][awox.mesh:294]: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
[D][awox.mesh:300]: Try to connecty to device A4:C1:38:A3:29:9E no mesh id known yet
[I][awox.mesh:167]: Try to connect A4:C1:38:A3:29:9E => rssi: -68
[I][awox.connection:112]: IDLE
[D][esp32_ble_client:053]: [0] [A4:C1:38:A3:29:9E] Found device
[I][awox.connection:120]: DISCOVERED
[D][esp32_ble_tracker:215]: Pausing scan to make connection...
[I][awox.connection:124]: READY_TO_CONNECT
[I][esp32_ble_client:069]: [0] [A4:C1:38:A3:29:9E] 0x00 Attempting BLE connection
[I][awox.connection:128]: CONNECTING
[I][esp32_ble_client:201]: [0] [A4:C1:38:A3:29:9E] Connected
[I][awox.connection:132]: CONNECTED
[I][awox.connection:135]: Connected....

abort() was called at PC 0x401bd26a on core 1

Backtrace: 0x40082ace:0x3ffd03c0 0x40091611:0x3ffd03e0 0x4009754e:0x3ffd0400 0x401bd26a:0x3ffd0470 0x401a26e3:0x3ffd0490 0x401b2e97:0x3ffd04b0 0x401a3671:0x3ffd04d0 0x401a39aa:0x3ffd04f0 0x401a3a19:0x3ffd0510 0x400de747:0x3ffd0530 0x400deb34:0x3ffd0580 0x400df4be:0x3ffd0620 0x400dfbd7:0x3ffd0740 0x401d7e1a:0x3ffd07a0 0x400e1991:0x3ffd07c0 0x401d7dc5:0x3ffd07e0 0x400e08e1:0x3ffd0800 0x401d8399:0x3ffd0820 0x401d844d:0x3ffd0840 0x400ea3b8:0x3ffd0860 0x400ec282:0x3ffd0890 0x400dfda2:0x3ffd08b0

ELF file SHA256: bd47845bcf720a6f

Rebooting...
```

mykalanui, 2024-01-22:
In the meantime I have tried ref:dev and factory resetting the lamps. To no avail...

dabepp, 2024-01-23:
I have the same issue with my ESP Dev Board. I used the YAML pasted below.
I can compile and install it until I put in the actual `awox_mesh:` Command. If I have this block in I get the error
(Same via OTA and manual download)

````
Linking .pioenvs/esphome-web-ee267c/firmware.elf
RAM:   [==        ]  18.3% (used 60076 bytes from 327680 bytes)
Flash: [==========]  112.8% (used 2070693 bytes from 1835008 bytes)
Error: The program size (2070693 bytes) is greater than maximum allowed (1835008 bytes)
*** [checkprogsize] Explicit exit, status 1
========================= [FAILED] Took 113.14 seconds =========================
````
YAML:
```
esphome:
  name: esphome-web-ee267c
  friendly_name: Awox

esp32:
  board: mhetesp32devkit
  framework:
     type: arduino
# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "<snip>"

ota:

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_user
  password: !secret mqtt_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.178.99
    gateway: 192.168.0.1
    subnet: 255.255.255.0
external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub

bluetooth_proxy:
  active: true

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

captive_portal:

awox_mesh:
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password
  address_prefix: A4:C1:##:##:##:##
  device_info:
    # Example device type, see device info in HomeAssistant or MQTT message to find the 'product_id'
    - product_id: 0x30
      device_type: RGB
      manufacturer: EGLO
      name: EGLOPanel 30X120
      model: EPanel_120
      icon: mdi:wall-sconce-flat 

# Example configuration entry
spi:
  miso_pin: GPIO19
  clk_pin: GPIO18

sensor:
  - platform: max6675
    name: "Ofen"
    id: ofen_2024
    cs_pin: GPIO5
#    cs_pin: GPIO12
    update_interval: 60s
```

dabepp, 2024-01-23:
Sorry forgot to mention that I already did that and got this error:

```
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.2") 
-- Component directory /data/cache/platformio/packages/framework-espidf/components/wpa_supplicant does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/data/build/esphome-web-ee267c/.pioenvs/esphome-web-ee267c/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'cxx'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:236 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:518 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:393 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 8.20 seconds ==========================
```

mykalanui, 2024-01-25:
I'm running an ESP32 dev board (DyiMore ESP-32-WROOM-32 with CP2102 from Amazon), another board from the same batch is working fine on a different project (BT-Proxy, not MQTT-related).

I did try a different model in yaml but I still get the same error. Can you recommend a board that definitely works so I can buy that?

fsaris, 2024-01-25:
I'm using https://nl.aliexpress.com/item/1005002410521023.html 

And I use the following config:

```
esp32:
  board: nodemcu-32s
  framework:
    type: esp-idf
 ```

mykalanui, 2024-01-26:
Here's my log (only the last few lines before reboot) in verbose mode:

```
[V][mqtt.idf:145]: MQTT_EVENT_DATA esphome-awox-mesh/debug
[V][mqtt.idf:109]: Event dispatched from event loop event_id=6
[V][mqtt.idf:145]: MQTT_EVENT_DATA esphome-awox-mesh/debug
[V][mqtt.idf:109]: Event dispatched from event loop event_id=6
[V][mqtt.idf:145]: MQTT_EVENT_DATA esphome-awox-mesh/debug
[V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 46
[V][esp32_ble_client:119]: [0] [A4:C1:38:14:C3:B7] gattc_event_handler: event=46 gattc_if=3
[V][mqtt.idf:109]: Event dispatched from event loop event_id=6
[V][mqtt.idf:145]: MQTT_EVENT_DATA esphome-awox-mesh/debug
[V][mqtt.idf:109]: Event dispatched from event loop event_id=6
[V][mqtt.idf:145]: MQTT_EVENT_DATA esphome-awox-mesh/debug
[V][mqtt.idf:109]: Event dispatched from event loop event_id=6
[V][mqtt.idf:145]: MQTT_EVENT_DATA esphome-awox-mesh/debug
[V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 7
[V][esp32_ble_client:119]: [0] [A4:C1:38:14:C3:B7] gattc_event_handler: event=7 gattc_if=3
[V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 7
[V][esp32_ble_client:119]: [0] [A4:C1:38:14:C3:B7] gattc_event_handler: event=7 gattc_if=3
[V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 7
[V][esp32_ble_client:119]: [0] [A4:C1:38:14:C3:B7] gattc_event_handler: event=7 gattc_if=3
[V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 6
[V][esp32_ble_client:119]: [0] [A4:C1:38:14:C3:B7] gattc_event_handler: event=6 gattc_if=3
[V][esp32_ble_client:194]: [0] [A4:C1:38:14:C3:B7] ESP_GATTC_SEARCH_CMPL_EVT
[V][esp32_ble_client:197]: [0] [A4:C1:38:14:C3:B7] Service UUID: 0x1800
[V][esp32_ble_client:199]: [0] [A4:C1:38:14:C3:B7]  start_handle: 0x1  end_handle: 0x6
[V][esp32_ble_client:197]: [0] [A4:C1:38:14:C3:B7] Service UUID: 0x180A
[V][esp32_ble_client:199]: [0] [A4:C1:38:14:C3:B7]  start_handle: 0x7  end_handle: 0xf
[V][esp32_ble_client:197]: [0] [A4:C1:38:14:C3:B7] Service UUID: 00010203-0405-0607-0809-0A0B0C0D1910
[V][esp32_ble_client:199]: [0] [A4:C1:38:14:C3:B7]  start_handle: 0x10  end_handle: 0x1c
[I][esp32_ble_client:201]: [0] [A4:C1:38:14:C3:B7] Connected
[I][awox.connection:132]: CONNECTED
[I][awox.connection:135]: Connected....
[V][esp32_ble_client:069]: [0] [A4:C1:38:14:C3:B7]  characteristic 00010203-0405-0607-0809-0A0B0C0D1911, handle 0x12, properties 0x1a
[V][esp32_ble_client:069]: [0] [A4:C1:38:14:C3:B7]  characteristic 00010203-0405-0607-0809-0A0B0C0D1912, handle 0x15, properties 0xe
[V][esp32_ble_client:069]: [0] [A4:C1:38:14:C3:B7]  characteristic 00010203-0405-0607-0809-0A0B0C0D1913, handle 0x18, properties 0x6
[0
abort() was called at PC 0x401be1a6 on core 1

Backtrace: 0x40082ace:0x3ffd0390 0x40091611:0x3ffd03b0 0x4009754e:0x3ffd03d0 0x401be1a6:0x3ffd0440 0x401a361f:0x3ffd0460 0x401b3dd3:0x3ffd0480 0x401a45ad:0x3ffd04a0 0x401a48e6:0x3ffd04c0 0x401a4955:0x3ffd04e0 0x400de957:0x3ffd0500 0x400ded44:0x3ffd0550 0x400df70a:0x3ffd05f0 0x400dfe63:0x3ffd0710 0x401d8cee:0x3ffd07a0 0x400e201d:0x3ffd07c0 0x400e0be7:0x3ffd07e0 0x400e0d31:0x3ffd0800 0x401d926d:0x3ffd0820 0x401d9321:0x3ffd0840 0x400eb164:0x3ffd0860 0x400ed042:0x3ffd0890 0x400e0076:0x3ffd08b0

ELF file SHA256: b66b5404bf2b5f04

Rebooting...
```

I assume there's nothing new in there for you to find a solution for me?

Thanks for sharing your board model, I have ordered a few and will try with these in a couple of weeks.

fsaris, 2024-01-27:
closing this as the initial problem  is resolved now

closed 2024-01-27 by fsaris