fsaris/EspHome-AwoX-BLE-mesh-hub #57: [Feature request] Color transition

oxigen-dev, 2024-01-18:
Hi @fsaris ,

Would it be possible to add the Color transition mode as an effect to the device in HA? 
I think it should be this: https://github.com/oxigen-dev/python-awox-mesh-light/blob/491bf8414f493ee6b1221dec20d03c0984bc0048/awoxmeshlight/__init__.py#L272

The lamps also have some candle flicker mode.. For me it is not a must, I was mentioning is just for completeness :)

Thanks!

fsaris, 2024-01-27:
@oxigen-dev in the dev branch there is now support for 2 effects: 

- candle (starts a flickering effect)
- color loop (transitions through a color pallet)

closed 2024-01-27 by fsaris

oxigen-dev, 2024-01-27:
Thanks, I'll test them!

oxigen-dev, 2024-01-27:
@fsaris I was doing some tests. Color loop seems to be working fine.
The candle effect is most of the time not starting. I was randomly pressing everything for 1 minute then it started working.
If an effect is activated and eg. I'm manually activating a Scene then the effect stays on "color loop" and doesn't go back automatically to "stop".

fsaris, 2024-01-28:
> If an effect is activated and eg. I'm manually activating a Scene then the effect stays on "color loop" and doesn't go back automatically to "stop".

That seems to be an issue with HA. Had that too but the reported state was updated by the esp.
Will do some extra testing but for now it looks to be a HA frontend issue.

> The candle effect is most of the time not starting. I was randomly pressing everything for 1 minute then it started working

Probably related to the issue above. Did you check the mqtt Transmitted messages for that light? Was the `effect` value correctly send?

oxigen-dev, 2024-01-28:
Starting with the effect `stop`. After selecting `candle`, nothing happens:
[image]
If selecting `color_loop`:
[image]
If selecting `candle` after `color_loop`:
[image]

It is there for only a msec :)

fsaris, 2024-01-28:
These are the state messages, but I'm interested in the command messages.

These can be viewed in HA.

Click on `MQTT INFO` on the device page 
[image]

And look for `Transmitted messages`

oxigen-dev, 2024-01-28:
1. Effect selected: `stop`
2. Activate scene (100% brightness, white) -> effect in HA changed to `candle`. Maybe `stop` or `off` (as I've seen in other integrations) should be the first in the list.
[image]
3. Effect: `color_loop`
[image]
4. Effect: `candle`
[image]
5. Effect: `stop`
[image]
6. Effect: `candle` -> nothing happens
[image]

fsaris, 2024-01-28:
So after `stop` it brakes?

When `effect: stop` is send as command it's not part of the `status` response. Maybe that's needed. Not sure how other integrations do this.
I see that Zigbee2mqtt doesn't add the `effect` to the `state` response.

> Maybe stop or off (as I've seen in other integrations) should be the first in the list.

I only found examples where it's the last in the list (Zigbee2mqtt, Tasmota).

oxigen-dev, 2024-01-28:
> So after `stop` it brakes?

Just stop->candle is not working.

> When `effect: stop` is send as command it's not part of the `status` response. Maybe that's needed. Not sure how other integrations do this. I see that Zigbee2mqtt doesn't add the `effect` to the [...]

This is a Philips Hue bulb with Zigbee2MQTT:
1. Effect was set to `blink`
2. Effect was set to `stop_effect`
[image]

List of effects (I think it follows the same logic as you):
[image]

State topic doesn't contain any information about the selected effect:
[image]