fsaris/EspHome-AwoX-BLE-mesh-hub #58: New group support strange behaviour

Dillton, 2024-01-20:
Hi @fsaris.

I have some points from observation since group support was added.

1) In Awox app I have 4 zones defined so this should be the number of created groups however 3 of 4 groups are duplicated.
Group 0 and 1 is Living Room
Group 2 and 3 is Bedroom
Group 4 and 5 is Office
Group 6 is Server Room

I think the the reason behind this is from Menu -> Settings -> Manage rooms which are the zones 
[image]

and at the same time from Menu -> My groups where I have 3 EGLO remote controllers and each have assigned one of the zones.
[image]

Could you take a look at ths if this can be somehow filtered? It might be confusing for someone in this state.

2) Group 6 (Server Room) in this case has only EGLO PLUG inside so I think this group should be created as switch or shouldn't be created at all.

3) Last thing which I believe would be nice and you can consider this as feature request is to have a config option in yaml to enable/disable group support. Something like use_awox_groups: true/false or so.
Personally I have custom light groups inside HA with other non EGLO lights for some usecases and currently I don't see any benefit or scenario where I would use native groups from Awox.

Anyway thank you for great job with Awox lights it's very appriciated.

fsaris, 2024-01-20:
> so is there any reason for moving to AwoxHomeControl? 

I needed to switch to update my AwoX Zigbee lights :)
Further no real reason

The app isn't used by the esp. You only need the credentials to get the mesh credentials. And the source depends on the app you use.

> and create group only from one source? 

The source is currently used to get the group info. Each device is asked for there group info. That is also the reason that only a group number is known. 

> group should be created as switch or shouldn't be created at all.

Switch groups are currently not implemented.

> feature request is to have a config option in yaml to enable/disable group support.

Was thinking about this, but not really sure what the benefits would be. You could also easily disable the discovered groups in HA.

> currently I don't see any benefit or scenario where I would use native groups from Awox.

Speed, when controlling as group the response is al lot faster. 

When controlling multiple devices at the same time the commands have a few miliseconds between each other else the mesh could drop some of them.

Using groups makes the lights respond instantly.

Dillton, 2024-01-21:
Thank you very much for explanation.

I will try to play a little bit more. So far I have tried to compare HA light group vs native group and yes it's much faster in responding.
For example I have 3 lights in group, when controlling via HA group they are toggled one by one in total this takes like 2 seconds but via native group all at the same time like 100ms in total so I see potential here now.

Regarding the group with Eglo plug I missed how the group is created and I can tell now it's created as switch group but control is not working as you said not yet implemented. However the group state changes when I toggle the plug directly.
So this is just info for future.
[image]

closed 2024-01-21 by fsaris