fsaris/EspHome-AwoX-BLE-mesh-hub #60: [Feature request] Time duration of transitions between color and effects

oxigen-dev, 2024-01-23:
Hi @fsaris ,

I was just wondering whether it would be possible to add the time transition functionality.
[image]

Now it doesn't take into consideration the time set.

Thanks!

fsaris, 2024-01-27:
The lights don't support this feature. You can set `fade_duration` and a `color_duration` value but that's only for `sequence` (the `color loop` effect.

closed 2024-01-27 by fsaris

oxigen-dev, 2024-01-27:
@fsaris What if that would be implemented on ESP side? Eg. Brightness is now 0%, I'd like to set it to 100% in 1000 ms and it would then increase the brightness with +10% in every 100 ms?

fsaris, 2024-01-28:
As this is a hub and not a real light controller I don't think that will work properly. 

To much to consider:
- lag of the mesh network
- other commands comming in during transition for same device 
- handling transition intervals and also handling other lights