fsaris/EspHome-AwoX-BLE-mesh-hub #61: [BUG] Single light triggers whole group

Dillton, 2024-01-24:
Hello @fsaris.

Sorry to bother again but I have noticed today some strange behaviour.
My problem is when I turn on a single light it will after a second turn on the rest of the lights in the group defined in Awox app.
No trigger from HA.
For example I have 3 lights in office, I turn on light above my desk and second later the other two lights are turned on.

Found out that this is happening only when the group for that zone/area is enabled. But there is more detail.

In #58 I mentioned the groups are duplicated for me and with this issue the duplication is kind of related I would say.
This behaviour is only happening when groups created from zones/areas are enabled. 
When I disabled them and enabled the groups created from remote controllers it's working as usual.

Below is a log when one light was turned on and other 2 followed:
[image]

fsaris, 2024-01-24:
If you check the logs you see that on the same second a command for a specific mesh_id (1115) and for a specific group_id (4).

Are you sure you didn't combine/group these "lights" in HA?

Dillton, 2024-01-24:
As I also mentioned in #58 I have groups in HA like this for office, living room and bedroom:
[image]

But these groups are not connected to the new groups from ESP. You can see there are single lights attached in that group.
So I expect when I turn on single light which is part of the HA group then the group state will be ON and other lights in group stays OFF. This was working normally before group feature and this is standard group behaviour in HA.

The same behaviour I would expect from the groups created from ESP. But all lights in group are turned on instead.
So I think there must be something that replicates the turn on command to all light in that group.
Something that is not coming from HA.

fsaris, 2024-01-24:
Can you check the mqtt info (discovery info and linked topics with last x messages are visible there) for device "group 4" in HA? And also the logged events for that device? Curious what is shown there.

Dillton, 2024-01-25:
Sure.

[image]

`Awox Office Group (light.awox_office_group)
MQTT discovery data:
Topic: homeassistant/light/group-4/config
Payload
schema: json
name: null
unique_id: group-4
icon: mdi:lightbulb-group
state_topic: awox-ble-mesh-hub/group-4/state
command_topic: awox-ble-mesh-hub/group-4/command
availability:
  - topic: awox-ble-mesh-hub/group-4/availability
  - topic: awox-ble-mesh-hub/status
  - topic: awox-ble-mesh-hub/connected
availability_mode: all
color_mode: true
brightness: true
brightness_scale: 255
supported_color_modes:
  - rgb
  - color_temp
min_mireds: 153
max_mireds: 370
device:
  identifiers:
    - esp-awox-mesh-group-4
  model: Group - 4
  manufacturer: ESPHome AwoX BLE mesh
  name: Group 4
  via_device: 0cb815c3aed0
platform: mqtt
Subscribed topics:
awox-ble-mesh-hub/group-4/availability
1 most recently received message(s)
Received 14:05:36
QoS: 0, Retained
Payload: online
awox-ble-mesh-hub/status
1 most recently received message(s)
Received 14:05:36
QoS: 0, Retained
Payload: online
awox-ble-mesh-hub/connected
10 most recently received message(s)
Received 14:13:37
QoS: 0
Payload: online
Received 14:13:42
QoS: 0
Payload: online
Received 14:13:47
QoS: 0
Payload: online
Received 14:13:52
QoS: 0
Payload: online
Received 14:13:57
QoS: 0
Payload: online
Received 14:14:02
QoS: 0
Payload: online
Received 14:14:07
QoS: 0
Payload: online
Received 14:14:12
QoS: 0
Payload: online
Received 14:14:17
QoS: 0
Payload: online
Received 14:14:22
QoS: 0
Payload: online
awox-ble-mesh-hub/group-4/state
10 most recently received message(s)
Received 14:12:41
QoS: 0
Payload: state: 'ON'
color_mode: color_temp
brightness: 255
color_temp: 172
color:
  r: 117
  g: 44
  b: 3
Received 14:12:42
QoS: 0
Payload: state: 'ON'
color_mode: color_temp
brightness: 255
color_temp: 172
color:
  r: 117
  g: 44
  b: 3
Received 14:12:42
QoS: 0
Payload: state: 'ON'
color_mode: color_temp
brightness: 255
color_temp: 172
color:
  r: 117
  g: 44
  b: 3
Received 14:13:12
QoS: 0
Payload: state: 'ON'
color_mode: color_temp
brightness: 255
color_temp: 174
color:
  r: 117
  g: 44
  b: 3
Received 14:13:12
QoS: 0
Payload: state: 'ON'
color_mode: color_temp
brightness: 255
color_temp: 174
color:
  r: 117
  g: 44
  b: 3
Received 14:13:12
QoS: 0
Payload: state: 'ON'
color_mode: color_temp
brightness: 255
color_temp: 174
color:
  r: 117
  g: 44
  b: 3
Received 14:13:12
QoS: 0
Payload: state: 'ON'
color_mode: color_temp
brightness: 255
color_temp: 174
color:
  r: 117
  g: 44
  b: 3
Received 14:13:14
QoS: 0
Payload: state: 'ON'
color_mode: color_temp
brightness: 255
color_temp: 174
color:
  r: 117
  g: 44
  b: 3
Received 14:13:14
QoS: 0
Payload: state: 'ON'
color_mode: color_temp
brightness: 255
color_temp: 174
color:
  r: 117
  g: 44
  b: 3
Received 14:13:14
QoS: 0
Payload: state: 'ON'
color_mode: color_temp
brightness: 255
color_temp: 174
color:
  r: 117
  g: 44
  b: 3`

fsaris, 2024-01-25:
Nothing under `Transmitted messages:`?

Dillton, 2024-01-25:
Oh sorry. I have updated above comment now.

fsaris, 2024-01-25:
I can't see what exactly triggers it. But you see in the transmitted messages that the group is turned on by HA 

```
Transmitted 20:23:43
QoS: 0
Payload: state: 'ON'
transition: 1
brightness: 68
Transmitted 20:24:15
```

The `off` and `on` after that one are triggered by you. At least that's what the log from your screenshot states.

Dillton, 2024-01-25:
Well yeah the group is turned on by HA thats correct.
But why also all lights in this group are turned ON instead of just one light which I want.
At the time only one single light was turned on in HA not a group of lights.

It would be best to show you as I am not sure if you know what I mean from the description.
Maybe if you will have some time and you have discord or some other app where I could share my screen so you can see directly what is going on for me?

Dillton, 2024-01-25:
Ok so I found the root cause.
It's not a bug here in code also nothing wrong in Awox itself.

This all happens only when I use Awox made group in Adaptive lighting integration (AL).
So AL takes the group as single device so when I turn on any light then the AL comes to action changes brightness and temperature automatically if the light or group with that light is a part of it's config.
That means when the AL changes the brightness/temperature its done for the group not the single light which in the end results in turning on the rest of the lights inside such a group.

Solution for this is to configure all light entities I want and don't try to make it simple with groups.
[image]

Anyway thank you very much @fsaris for your time I stolen from you for no reason in the end.
Now I can finally think about how to implement Awox groups inside dashboards etc. to have smoother group controls.

closed 2024-01-25 by Dillton