fsaris/EspHome-AwoX-BLE-mesh-hub #62: Logger: Compile Warnings and Errors

mykalanui, 2024-01-25:
I have noticed several different warnings and errors when compiling with different log levels:

DEBUG (compiles with warnings):
```
...

|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/esphome-awox-mesh/src/esphome/components/awox_mesh/awox_mesh.o
Compiling .pioenvs/esphome-awox-mesh/src/esphome/components/awox_mesh/device.o
In file included from src/esphome/components/awox_mesh/device.h:8,
                 from src/esphome/components/awox_mesh/device.cpp:6:
src/esphome/components/awox_mesh/helpers.h:9:20: warning: 'std::__cxx11::string esphome::awox_mesh::string_as_binary_string(std::__cxx11::string)' defined but not used [-Wunused-function]
 static std::string string_as_binary_string(std::string words) {
                    ^~~~~~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/esphome-awox-mesh/src/esphome/components/awox_mesh/group.o
In file included from src/esphome/components/awox_mesh/device.h:8,
                 from src/esphome/components/awox_mesh/group.h:6,
                 from src/esphome/components/awox_mesh/group.cpp:3:
src/esphome/components/awox_mesh/helpers.h:9:20: warning: 'std::__cxx11::string esphome::awox_mesh::string_as_binary_string(std::__cxx11::string)' defined but not used [-Wunused-function]
 static std::string string_as_binary_string(std::string words) {
                    ^~~~~~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/esphome-awox-mesh/src/esphome/components/awox_mesh/mesh_connection.o
In file included from src/esphome/components/awox_mesh/device.h:8,
                 from src/esphome/components/awox_mesh/mesh_connection.h:13,
                 from src/esphome/components/awox_mesh/awox_mesh.h:15,
                 from src/esphome/components/awox_mesh/awox_mesh.cpp:7:
src/esphome/components/awox_mesh/helpers.h:17:20: warning: 'std::__cxx11::string esphome::awox_mesh::int_as_hex_string(unsigned char, unsigned char, unsigned char)' defined but not used [-Wunused-function]
 static std::string int_as_hex_string(unsigned char hex1, unsigned char hex2, unsigned char hex3) {
                    ^~~~~~~~~~~~~~~~~
src/esphome/components/awox_mesh/helpers.h:9:20: warning: 'std::__cxx11::string esphome::awox_mesh::string_as_binary_string(std::__cxx11::string)' defined but not used [-Wunused-function]
 static std::string string_as_binary_string(std::string words) {
                    ^~~~~~~~~~~~~~~~~~~~~~~
In file included from src/esphome/components/awox_mesh/device.h:8,
                 from src/esphome/components/awox_mesh/mesh_connection.h:13,
                 from src/esphome/components/awox_mesh/mesh_connection.cpp:5:
src/esphome/components/awox_mesh/helpers.h:17:20: warning: 'std::__cxx11::string esphome::awox_mesh::int_as_hex_string(unsigned char, unsigned char, unsigned char)' defined but not used [-Wunused-function]
 static std::string int_as_hex_string(unsigned char hex1, unsigned char hex2, unsigned char hex3) {
                    ^~~~~~~~~~~~~~~~~
Compiling .pioenvs/esphome-awox-mesh/src/esphome/components/esp32/core.o
Compiling .pioenvs/esphome-awox-mesh/src/esphome/components/esp32/gpio.o
Compiling .pioenvs/esphome-awox-mesh/src/esphome/components/esp32/preferences.o

...

Compiling .pioenvs/esphome-awox-mesh/app_trace/app_trace_util.o
Compiling .pioenvs/esphome-awox-mesh/app_trace/host_file_io.o
Compiling .pioenvs/esphome-awox-mesh/app_trace/gcov/gcov_rtio.o
In file included from src/esphome/components/awox_mesh/device.h:8,
                 from src/esphome/components/awox_mesh/mesh_connection.h:13,
                 from src/esphome/components/awox_mesh/awox_mesh.h:15,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/components/awox_mesh/helpers.h:17:20: warning: 'std::__cxx11::string esphome::awox_mesh::int_as_hex_string(unsigned char, unsigned char, unsigned char)' defined but not used [-Wunused-function]
 static std::string int_as_hex_string(unsigned char hex1, unsigned char hex2, unsigned char hex3) {
                    ^~~~~~~~~~~~~~~~~
src/esphome/components/awox_mesh/helpers.h:9:20: warning: 'std::__cxx11::string esphome::awox_mesh::string_as_binary_string(std::__cxx11::string)' defined but not used [-Wunused-function]
 static std::string string_as_binary_string(std::string words) {
                    ^~~~~~~~~~~~~~~~~~~~~~~
Archiving .pioenvs/esphome-awox-mesh/esp-idf/app_trace/libapp_trace.a
Compiling .pioenvs/esphome-awox-mesh/app_update/esp_ota_ops.o
Compiling .pioenvs/esphome-awox-mesh/app_update/esp_app_desc.o

...

``` 

VERBOSE (compile fails):

```
INFO ESPHome 2023.12.9
[... 48 lines cut ...]
                 from src/esphome/components/awox_mesh/group.cpp:3:
src/esphome/components/awox_mesh/helpers.h:9:20: warning: 'std::__cxx11::string esphome::awox_mesh::string_as_binary_string(std::__cxx11::string)' defined but not used [-Wunused-function]
 static std::string string_as_binary_string(std::string words) {
                    ^~~~~~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/esphome-awox-mesh/src/esphome/components/awox_mesh/mesh_connection.o
In file included from src/esphome/components/awox_mesh/device.h:8,
                 from src/esphome/components/awox_mesh/mesh_connection.h:13,
                 from src/esphome/components/awox_mesh/awox_mesh.h:15,
                 from src/esphome/components/awox_mesh/awox_mesh.cpp:7:
src/esphome/components/awox_mesh/helpers.h: At global scope:
src/esphome/components/awox_mesh/helpers.h:17:20: warning: 'std::__cxx11::string esphome::awox_mesh::int_as_hex_string(unsigned char, unsigned char, unsigned char)' defined but not used [-Wunused-function]
 static std::string int_as_hex_string(unsigned char hex1, unsigned char hex2, unsigned char hex3) {
                    ^~~~~~~~~~~~~~~~~
src/esphome/components/awox_mesh/helpers.h:9:20: warning: 'std::__cxx11::string esphome::awox_mesh::string_as_binary_string(std::__cxx11::string)' defined but not used [-Wunused-function]
 static std::string string_as_binary_string(std::string words) {
                    ^~~~~~~~~~~~~~~~~~~~~~~
*** [.pioenvs/esphome-awox-mesh/src/esphome/components/awox_mesh/awox_mesh.o] Error 1
In file included from src/esphome/components/awox_mesh/device.h:8,
                 from src/esphome/components/awox_mesh/mesh_connection.h:13,
                 from src/esphome/components/awox_mesh/mesh_connection.cpp:5:
src/esphome/components/awox_mesh/helpers.h:17:20: warning: 'std::__cxx11::string esphome::awox_mesh::int_as_hex_string(unsigned char, unsigned char, unsigned char)' defined but not used [-Wunused-function]
 static std::string int_as_hex_string(unsigned char hex1, unsigned char hex2, unsigned char hex3) {
                    ^~~~~~~~~~~~~~~~~
========================= [FAILED] Took 64.37 seconds =========================
```

fsaris, 2024-01-25:
The warnings aren't a issue, it's only complaining about unused code.

Are you using the `main` or `dev` branch?

fsaris, 2024-01-25:
ah, in the verbose version there is indeed a compiler  error.

fsaris, 2024-01-25:
@mykalanui tnx for pointing this out, fix in new version. Regarding #56 this will only have a impact there if you used `VERBOSE` logging.

closed 2024-01-25 by fsaris