fsaris/EspHome-AwoX-BLE-mesh-hub #63: Error: The program size (2021481 bytes) is greater than maximum allowed (1835008 bytes)

dabepp, 2024-01-28:
Referring to the solved Issue #56 I still face troubles.
Did everything that was mentioned (Change to ESP-IDF, Clean Buildfiles etc.)

Here is my config (Removed everything but the AWOX Stuff)
```
esphome:
  name: esphome-web-ee267c
  friendly_name: Awox

esp32:
#  board: esp32dev
  board: mhetesp32devkit
  framework:
#    type: esp-idf
     type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "<SNIP>"

ota:

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_user
  password: !secret mqtt_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.178.99
    gateway: 192.168.0.1
    subnet: 255.255.255.0
external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub

captive_portal:

awox_mesh:
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password
  address_prefix: A4:C1:<SNIP>
  device_info:
    - product_id: 0x30
      device_type: RGB
      manufacturer: EGLO
      name: EGLOPanel 30X120
      model: EPanel_120
      icon: mdi:wall-sconce-flat
```

This is the error I get when including Arduino Framework:

```
Linking .pioenvs/esphome-web-ee267c/firmware.elf
RAM:   [==        ]  18.3% (used 59916 bytes from 327680 bytes)
Flash: [==========]  110.2% (used 2021481 bytes from 1835008 bytes)
Error: The program size (2021481 bytes) is greater than maximum allowed (1835008 bytes)
*** [checkprogsize] Explicit exit, status 1
========================= [FAILED] Took 117.44 seconds =========================
```

This is the error I get when I use esp-idf as Framework:
```
Processing esphome-web-ee267c (board: mhetesp32devkit; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Component directory /data/cache/platformio/packages/framework-espidf/components/wpa_supplicant does not contain a CMakeLists.txt file. No component will be added
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/data/build/esphome-web-ee267c/.pioenvs/esphome-web-ee267c/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'cxx'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:236 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:518 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:393 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 2.36 seconds ==========================
```

fsaris, 2024-01-28:
The `api` and `captive_portal` is still part of your config. So if you want to use the `arduino*` build try again without those.

But that is probably not the reason for the failing build `esp-idf`.

Can you try the following and without `api` and `captive_portal`.

```
external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub
      ref: main
```

Be sure to click "Clean build files" so all is build again from scratch.

What version of esphome are you using?

dabepp, 2024-01-28:
Did not work...

Here with framework esp-idf:
```
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.2") 
-- Component directory /data/cache/platformio/packages/framework-espidf/components/wpa_supplicant does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/data/build/esphome-web-ee267c/.pioenvs/esphome-web-ee267c/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'cxx'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:236 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:518 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:393 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 2.92 seconds ==========================

``` 

And with Framework Arduino:

```
Linking .pioenvs/esphome-web-ee267c/firmware.elf
RAM:   [==        ]  18.1% (used 59372 bytes from 327680 bytes)
Flash: [==========]  104.2% (used 1912649 bytes from 1835008 bytes)
Error: The program size (1912649 bytes) is greater than maximum allowed (1835008 bytes)
*** [checkprogsize] Explicit exit, status 1
========================= [FAILED] Took 82.33 seconds =========================
```

fsaris, 2024-01-28:
Do you have the `esp32_ble_tracker` in your config? That is required for `awox` to work.

```
esp32_ble_tracker:
  scan_parameters:
    active: false
```

dabepp, 2024-01-28:
> Do you have the `esp32_ble_tracker` in your config? That is required for `awox` to work.
> 
> ```
> esp32_ble_tracker:
>   scan_parameters:
>     active: false
> ```

I added it again, but still no success...

```
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true
```

Error (Whole output this time):

```
INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/esphome-web-ee267c.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-ee267c (board: mhetesp32devkit; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/awox_mesh/awox_mesh.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/awox_mesh/device.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/awox_mesh/group.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/awox_mesh/mesh_connection.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/esp32/gpio.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/esp32/preferences.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/esp32_ble/ble.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/esp32_ble/ble_advertising.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/esp32_ble/ble_uuid.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/esp32_ble_client/ble_characteristic.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/esp32_ble_client/ble_client_base.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/esp32_ble_client/ble_service.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/json/json_util.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/mdns/mdns_host.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/mdns/mdns_libretiny.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/mqtt/custom_mqtt_device.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/mqtt/mqtt_backend_esp32.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/mqtt/mqtt_binary_sensor.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/mqtt/mqtt_button.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/mqtt/mqtt_client.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/mqtt/mqtt_climate.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/mqtt/mqtt_component.cpp.o
Compiling .pioenvs/esphome-web-ee267c/src/esphome/components/mqtt/mqtt_cover.cpp.o
[... 88 lines cut ...]
Compiling .pioenvs/esphome-web-ee267c/FrameworkArduino/esp32-hal-psram.c.o
Compiling .pioenvs/esphome-web-ee267c/FrameworkArduino/esp32-hal-rgb-led.c.o
Compiling .pioenvs/esphome-web-ee267c/FrameworkArduino/esp32-hal-rmt.c.o
Compiling .pioenvs/esphome-web-ee267c/FrameworkArduino/esp32-hal-sigmadelta.c.o
Compiling .pioenvs/esphome-web-ee267c/FrameworkArduino/esp32-hal-spi.c.o
Compiling .pioenvs/esphome-web-ee267c/FrameworkArduino/esp32-hal-time.c.o
Compiling .pioenvs/esphome-web-ee267c/FrameworkArduino/esp32-hal-timer.c.o
Compiling .pioenvs/esphome-web-ee267c/FrameworkArduino/esp32-hal-tinyusb.c.o
Compiling .pioenvs/esphome-web-ee267c/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pioenvs/esphome-web-ee267c/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pioenvs/esphome-web-ee267c/FrameworkArduino/firmware_msc_fat.c.o
Compiling .pioenvs/esphome-web-ee267c/FrameworkArduino/libb64/cdecode.c.o
Compiling .pioenvs/esphome-web-ee267c/FrameworkArduino/libb64/cencode.c.o
Compiling .pioenvs/esphome-web-ee267c/FrameworkArduino/main.cpp.o
Compiling .pioenvs/esphome-web-ee267c/FrameworkArduino/stdlib_noniso.c.o
Compiling .pioenvs/esphome-web-ee267c/FrameworkArduino/wiring_pulse.c.o
Compiling .pioenvs/esphome-web-ee267c/FrameworkArduino/wiring_shift.c.o
Archiving .pioenvs/esphome-web-ee267c/libFrameworkArduino.a
Linking .pioenvs/esphome-web-ee267c/firmware.elf
RAM:   [==        ]  18.1% (used 59372 bytes from 327680 bytes)
Flash: [==========]  104.2% (used 1912673 bytes from 1835008 bytes)
Error: The program size (1912673 bytes) is greater than maximum allowed (1835008 bytes)
*** [checkprogsize] Explicit exit, status 1
========================= [FAILED] Took 81.00 seconds =========================
```

fsaris, 2024-01-28:
Okay, build succeeds but is to large for your device. But it looks that you used framework arduino this time.

Can you post your yaml config to. It's hard to see what your exactly testing.

Does it build now when using `esp-idf`?

fsaris, 2024-01-28:
Are you using a wired or wireless connection to flash the device?

dabepp, 2024-01-28:
> Are you using a wired or wireless connection to flash the device?

Wireless. But I also tried the manual download and get the same result:

```
Linking .pioenvs/esphome-web-ee267c/firmware.elf
RAM:   [==        ]  18.3% (used 59836 bytes from 327680 bytes)
Flash: [==========]  105.3% (used 1932401 bytes from 1835008 bytes)
Error: The program size (1932401 bytes) is greater than maximum allowed (1835008 bytes)
*** [checkprogsize] Explicit exit, status 1
========================= [FAILED] Took 86.78 seconds =========================
```

dabepp, 2024-01-28:
> Does it build now when using `esp-idf`?

No:

```
INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/esphome-web-ee267c.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-ee267c (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.2") 
-- Component directory /data/cache/platformio/packages/framework-espidf/components/wpa_supplicant does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/data/build/esphome-web-ee267c/.pioenvs/esphome-web-ee267c/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'cxx'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:236 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:518 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:393 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 2.72 seconds ==========================
```

fsaris, 2024-01-28:
Your build issue seems to be related to your esphome environment see https://github.com/esphome/issues/issues/5133 for a similar issue.

For wireless and manual download it needs the same free room on the device. Try the `plug into...` install method.

dabepp, 2024-01-28:
> Try the `plug into...` install method.

I did. This is what I got:

```
INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/esphome-web-ee267c.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-ee267c (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Component directory /data/cache/platformio/packages/framework-espidf/components/wpa_supplicant does not contain a CMakeLists.txt file. No component will be added
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/data/build/esphome-web-ee267c/.pioenvs/esphome-web-ee267c/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'cxx'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:236 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:518 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:393 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 2.26 seconds ==========================

```

fsaris, 2024-01-28:
Try your last attempt again only then with `framework arduino`

mykalanui, 2024-02-06:
the yaml in your first post shows that you are still using arduino framework.

esp-idf is commented out:

```
esp32:
#  board: esp32dev
  board: mhetesp32devkit
  framework:
#    type: esp-idf
     type: arduino
```

dabepp, 2024-02-06:
> the yaml in your first post shows that you are still using arduino framework.

I tried both. Both fail.

Arduino:
```RAM:   [==        ]  18.3% (used 59836 bytes from 327680 bytes)
Error: The program size (1932401 bytes) is greater than maximum allowed (1835008 bytes)
*** [checkprogsize] Explicit exit, status 1
Flash: [==========]  105.3% (used 1932401 bytes from 1835008 bytes)
========================= [FAILED] Took 91.26 seconds =========================
```

esp-idf:
```
Installing collected packages: pyparsing, urllib3, tqdm, six, pyyaml, pycparser, packaging, msgpack, kconfiglib, idna, future, filelock, contextlib2, colorama, click, charset-normalizer, certifi, schema, requests, cffi, requests-toolbelt, requests-file, cryptography, cachecontrol, idf-component-manager
  DEPRECATION: future is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559
  Running setup.py install for future: started
  Running setup.py install for future: finished with status 'done'
Successfully installed cachecontrol-0.14.0 certifi-2024.2.2 cffi-1.16.0 charset-normalizer-3.3.2 click-8.1.7 colorama-0.4.6 contextlib2-21.6.0 cryptography-3.4.8 filelock-3.13.1 future-0.18.3 idf-component-manager-1.4.2 idna-3.6 kconfiglib-13.7.1 msgpack-1.0.7 packaging-23.2 pycparser-2.21 pyparsing-2.3.1 pyyaml-6.0.1 requests-2.31.0 requests-file-1.5.1 requests-toolbelt-1.0.0 schema-0.7.5 six-1.16.0 tqdm-4.66.1 urllib3-1.26.18
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.2") 
-- Component directory /data/cache/platformio/packages/framework-espidf/components/wpa_supplicant does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/data/build/esphome-web-ee267c/.pioenvs/esphome-web-ee267c/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'cxx'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:236 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:518 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:393 (idf_build_process)
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 19.95 seconds =========================
```

TotalPanther317, 2024-02-10:
I have exactly the same issue, I use the arduino framework. If I try using esp-idf the 'freertos' module can not be found.

fsaris, 2024-02-10:
That is strange as that is just core esphome.

If you try to build a Bluetooth Proxy (https://esphome.io/components/bluetooth_proxy.html) does it build?

TotalPanther317, 2024-02-11:
I tried it again without specifying a framework, which did not work. I also used esp-idf and arduino with this code:
```
esphome:
  name: awox-ble-mesh-hub
  friendly_name: "AwoX BLE mesh hub"

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    # Set to fixed version due to build issues https://community.home-assistant.io/t/esphome-could-not-find-package-json/499026/8
    #version: 2.0.6 #https://github.com/espressif/arduino-esp32/releases

# Enable logging
#logger:
  
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_user
  password: !secret mqtt_password

# Disable Home Assistant API to prevent frequent MQTT disconnections
api:

ota:
  password: 988af1bdbc81aa0760c360430673b560

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub

esp32_ble_tracker:
  scan_parameters:
    active: false

bluetooth_proxy:
  active: true
```
It fails with the program size being greater than maxiumum allowed.

Plugging it into my homeassistant pi does not change the maximum allowed size.

TotalPanther317, 2024-02-11:
It will fail every time with the program size error. This is my code:

```
esphome:
  name: awox-ble-mesh-hub
  friendly_name: "AwoX BLE mesh hub"

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    # Set to fixed version due to build issues https://community.home-assistant.io/t/esphome-could-not-find-package-json/499026/8
    #version: 2.0.6 #https://github.com/espressif/arduino-esp32/releases

# Enable logging
#logger:
  
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_user
  password: !secret mqtt_password

# Disable Home Assistant API to prevent frequent MQTT disconnections
api:

ota:
  password: 

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub

esp32_ble_tracker:
  scan_parameters:
    active: false

bluetooth_proxy:
  active: true
```

I have included the bluetooth_proxy, removing it changes nothing. I tried arduino and esp-idf, both fail.

fsaris, 2024-02-11:
- **"Error: The program size (1932401 bytes) is greater than maximum allowed (1835008 bytes)"**
   Use the wired option to flash the device of try to decrease the image size by leaving stuff out and/of use framework esp-idf
- **"build fails with error about some dependency"**
  Try to build only the minimum (so without the awox component) and check if that works and keep adding until you find the part that breaks.

closed 2024-02-11 by fsaris

TotalPanther317, 2024-02-11:
It does not build with the minimum, this is the code:
```
esphome:
  name: hub
  friendly_name: "hub"

esp32:
  board: esp32dev
  framework:
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:

ota:
  password:  
```

the freertos thing does not exist in my installation somehow.

fsaris, 2024-02-11:
Then maybe they can assist you on the esphome channels

TotalPanther317, 2024-02-11:
I also tried removing as much as possible, it will go down to 100.8%. However I can not delete more.

TotalPanther317, 2024-02-11:
I found the big problem with this. Esphome will only let you use 2MB (my board has 4), I dont know how I can increase that.

fsaris, 2024-02-12:
Did you use the flash over USB method? That should allow you to use all the room. If not then you're maybe using the wrong board config.

Using the OTA method requires additional free room on the device (that's why it's limited to only have of the room).

Fixing the esp-idf build should also give you a smaller result.

TotalPanther317, 2024-02-12:
Well, I tried connecting it to my rpi and then uploading via usb. It did not really change a thing. But I am not relly sure if I did it right.

fsaris, 2024-02-13:
Connect the board USB to the pc where you opened the web dashboard of esphome and choose "Install" -> "plug into this computer"

See https://esphome.github.io/esp-web-tools/ for more details

TotalPanther317, 2024-02-14:
To everyone experiencing this problem, first uninstall all esphome versions, then reinstall.

michalpulda, 2024-02-21:
I'm wondering if it's not connected to a partition table difference between `arduino` and `esp-idf`. The ESPHome bluetooth proxy [documentation](https://esphome.io/components/bluetooth_proxy.html) says:

> When switching from arduino to esp-idf, make sure to update the device with a serial cable as the partition table is different between the two frameworks as [OTA Update Component](https://esphome.io/components/ota) updates will not change the partition table.

If so, maybe the https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/discussions/71 steps shall be changed?

fsaris, 2024-03-07:
@michalpulda thnx for the addition, updated #71

Silvr5, 2024-05-04:
Hi everyone, I'm currently experiencing the issue described above. Today I updated my esphome environment running in docker from 2023.11.6 to 2024.4.2 and the exact same config that built before perfectly now fails to build with the program size error. I tested some different configurations with the yaml and found out that basic config with the added external component and the ble tracker compiles correctly, however as soon as I add te awox_mesh component, the program size gets to big. The error appears independent of the flashing method. I tested the OTA, the wired connection as well as the manual download. Currently I use the arduino framework as the esp-idf will not compile correctly inside the docker environment (I get a compilation error with some cryptography module).
Is there another way to reduce the program size? 

Currently I get the following error:
```
Linking .pioenvs/buero-esp32/firmware.elf
RAM:   [=         ]  11.1% (used 59212 bytes from 532480 bytes)
Flash: [==========]  101.7% (used 1866073 bytes from 1835008 bytes)
Error: The program size (1866073 bytes) is greater than maximum allowed (1835008 bytes)
*** [checkprogsize] Explicit exit, status 1
========================= [FAILED] Took 228.23 seconds =========================
```
This is the configuration I use:
```yaml
esphome:
  name: "buero-esp32"
  friendly_name: "ESP32 BLE AWOX"

esp32:
  board: az-delivery-devkit-v4
  framework:
    #type: esp-idf
    type: arduino

ota:
  password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_user_esp32
  password: !secret mqtt_password_esp32

external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub

esp32_ble_tracker:
  scan_parameters:
    active: false

awox_mesh:
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password
  address_prefix: A4:C1
  device_info:
    -  product_id: 0x96
       device_type: RGB
       name: EGLO RGB+TW
       model: EGLO-RGB-TW
       manufacturer: EGLO
       
    -  product_id: 0x25
       device_type: RGB
       name: EGLOPanel 30X30
       model: EPanel_300
       manufacturer: EGLO
```

fsaris, 2024-05-04:
> Currently I use the arduino framework as the esp-idf will not compile correctly inside the docker environment (I get a compilation error with some cryptography module).

When switching framework make sure to clear the build cache and use the wired method to upload/flash.

Silvr5, 2024-05-05:
I tried most of the already mentioned solutions above but none of them helped me. I always cleared the build cache before compiling with another framework, however as I mentioned the build itself fails due to some building error in the cryptography module. I did some research today and tried several hints to get this fixed but the esp-idf framework is unfortunately not building correctly. This is the shortened error message:
````
Building wheels for collected packages: cryptography
  Building wheel for cryptography (pyproject.toml): started
  Building wheel for cryptography (pyproject.toml): still running...
  Building wheel for cryptography (pyproject.toml): still running...
  Building wheel for cryptography (pyproject.toml): still running...
  Building wheel for cryptography (pyproject.toml): finished with status 'error'
  error: subprocess-exited-with-error
````
```
running build_rust
          Updating crates.io index
      warning: spurious network error (2 tries remaining): http parser error: stream ended at an unexpected time; class=Http (34)
      warning: spurious network error (1 tries remaining): http parser error: stream ended at an unexpected time; class=Http (34)
      error: Unable to update registry `crates-io`
      
      Caused by:
        failed to fetch `https://github.com/rust-lang/crates.io-index`
      
      Caused by:
        network failure seems to have happened
        if a proxy or similar is necessary `net.git-fetch-with-cli` may help here
        https://doc.rust-lang.org/cargo/reference/config.html#netgit-fetch-with-cli
      
      Caused by:
        SSL error: 0xffff8780 - SSL - The peer notified us that the connection is going to be closed; class=Ssl (16)
      
          =============================DEBUG ASSISTANCE=============================
          If you are seeing a compilation error please try the following steps to
          successfully install cryptography:
          1) Upgrade to the latest pip and try again. This will fix errors for most
             users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
          2) Read https://cryptography.io/en/latest/installation.html for specific
             instructions for your platform.
          3) Check our frequently asked questions for more information:
             https://cryptography.io/en/latest/faq.html
          4) Ensure you have a recent Rust toolchain installed:
             https://cryptography.io/en/latest/installation.html#rust
          5) If you are experiencing issues with Rust for *this release only* you may
             set the environment variable `CRYPTOGRAPHY_DONT_BUILD_RUST=1`.
          =============================DEBUG ASSISTANCE=============================
      
      error: `cargo metadata --manifest-path src/rust/Cargo.toml --format-version 1` failed with code 101
      -- Output captured from stdout:
      
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography, which is required to install pyproject.toml-based projects
*** Error 1
Reading CMake configuration...
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/config/.esphome/build/buero-esp32/.pioenvs/buero-esp32/CMakeFiles/CMakeOutput.log".
See also "/config/.esphome/build/buero-esp32/.pioenvs/buero-esp32/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /config/.esphome/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /config/.esphome/build/buero-esp32/.pioenvs/buero-esp32/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/config/.esphome/platformio/packages/tool-ninja/ninja cmTC_754a4 && [1/2] Building C object CMakeFiles/cmTC_754a4.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_754a4
    FAILED: cmTC_754a4 
    : && /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_754a4.dir/testCCompiler.c.obj  -o cmTC_754a4   && :
    /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: cannot open shared object file: No such file or directory
    collect2: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.
```