fsaris/EspHome-AwoX-BLE-mesh-hub #75: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)

p-h-a-i-l, 2024-02-28:
I have a very hard time getting my system up and running:

```
[02:02:19][D][awox.mesh:238]: Total devices: 1
[02:02:19][D][awox.mesh:241]: Available device A4:C1:38:41:E1:28 [0] => rssi: -39
[02:02:19][D][awox.mesh:266]: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
[02:02:19][D][awox.mesh:272]: Try to connecty to device A4:C1:38:41:E1:28 no mesh id known yet
[02:02:19][I][awox.mesh:139]: Try to connect A4:C1:38:41:E1:28 => rssi: -39
[02:02:19][I][awox.connection:116]: IDLE
[02:02:19][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[02:02:19][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[02:02:19][V][awox.mesh:058]: Found existing device: A4:C1:38:41:E1:28
[02:02:19][V][awox.mesh:088]: Found Awox device 523yp9DW - A4:C1:38:41:E1:28. RSSI: -40 dB (total devices: 1)
[02:02:19][V][awox.mesh:058]: Found existing device: A4:C1:38:41:E1:28
[02:02:19][V][awox.mesh:088]: Found Awox device  - A4:C1:38:41:E1:28. RSSI: -39 dB (total devices: 1)
```

Here is what I tried:
- tried to get get it running with https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub - devices were added to home assistant, but no cotnrol possible
- Two different ESP32 Boards
- Moved Lamp from AwoX Smart Control to AwoX Home Control

The ESP32 is very close (1m) to the Lamp.
MQTT is working, I can see messages on the Broker.

This is my config:
```
esphome:
  name: awox-ble-mesh-hub2
  friendly_name: "AwoX BLE mesh hub2"

esp32:
  # Match this with the board your using
  board: esp32dev
  framework:
    # Be sure to "Clean Build Files" when switching framework
    type: esp-idf

# Enable logging
logger:
#  level: VERY_VERBOSE
  level: VERBOSE

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_user
  password: !secret mqtt_password

# Disable Home Assistant API to prevent frequent MQTT disconnections
#api:

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub

esp32_ble_tracker:
  scan_parameters:
    active: false

awox_mesh:
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password
```

Log with VERBOSE: 
[logs_esphome-web-1d4804_run.txt](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/14427291/logs_esphome-web-1d4804_run.txt)

Here is a Log with VERY_VERBOSE [logs_esphome-web-1d4804_logs.txt](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/14427267/logs_esphome-web-1d4804_logs.txt)

I have no clue where to look further. Can someone give me assistance?

fsaris, 2024-02-28:
You are probably running into this issue https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/issues/74

Fix was just merged to main branch

closed 2024-02-28 by fsaris

p-h-a-i-l, 2024-02-28:
You are right, thanks!