fsaris/EspHome-AwoX-BLE-mesh-hub #79: Connection problems

oxigen-dev, 2024-03-01:
Hi @fsaris ,

I've upgraded to the newest release from [6b36e809a448166384880a34319149c7d69776bb](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/tree/6b36e809a448166384880a34319149c7d69776bb)

Unfortunately now connecting to the device is mainly only working after reboot.

`[20:29:56][I][awox.mesh.mqtt:076]: Publish connected to mesh device - offline
[20:29:58][D][awox.mesh:238]: Total devices: 1
[20:29:58][D][awox.mesh:241]: Available device [redacted] [3590] => rssi: -9999
[20:29:58][D][awox.mesh:266]: Currently 0 mesh devices reachable through active connections (1 currently known and 1 fully recognized)
[20:29:58][D][awox.mesh:129]: No devices found to connect to
[20:29:58][I][awox.mesh.mqtt:076]: Publish connected to mesh device - offline
[20:30:03][D][awox.mesh:238]: Total devices: 1
[20:30:03][D][awox.mesh:241]: Available device [redacted] [3590] => rssi: -9999
[20:30:03][D][awox.mesh:266]: Currently 0 mesh devices reachable through active connections (1 currently known and 1 fully recognized)
[20:30:03][D][awox.mesh:129]: No devices found to connect to
[20:30:03][I][awox.mesh.mqtt:076]: Publish connected to mesh device - offline
[20:30:07][D][sensor:094]: 'Awox BLE Mesh Hub 2 Uptime Sensor': Sending state 7597.82812 s with 0 decimals of accuracy
[20:30:08][D][awox.mesh:238]: Total devices: 1
[20:30:08][D][awox.mesh:241]: Available device [redacted] [3590] => rssi: -9999
[20:30:08][D][awox.mesh:266]: Currently 0 mesh devices reachable through active connections (1 currently known and 1 fully recognized)
[20:30:08][D][awox.mesh:129]: No devices found to connect to
[20:30:08][I][awox.mesh.mqtt:076]: Publish connected to mesh device - offline`

After rebooting the RSSI is back to ~60.

Can you please check it? Thanks!

fsaris, 2024-03-02:
Looks that the `esp32_ble_tracker` isn't reporting online devices for you.

What version of ESPHome are you using? And what `esp32_ble_tracker` do you have?

oxigen-dev, 2024-03-02:
> Looks that the `esp32_ble_tracker` isn't reporting online devices for you.
> 
> What version of ESPHome are you using? And what `esp32_ble_tracker` do you have?

I was using 2024.2.0. Now I'm back to ESPHome 2023.11.6 and AwoX BLE Mesh Hub 2024.1.1 and everything is working fine.

fsaris, 2024-03-02:
ok, got the same. the scanner doesn't restart automatically anymore. Currently investigating

fsaris, 2024-03-02:
Can you test current `dev`?

fsaris, 2024-03-02:
ok, needed to revert my fix. looks to be an issue in `2024.2.1`  switching back to `2024.2.0`.

@oxigen-dev you are sure you used `2024.2.0` and not `2024.2.0`?

fsaris, 2024-03-02:
😔 after a while scanner also breaks for 2024.2.0

oxigen-dev, 2024-03-02:
> ok, needed to revert my fix. looks to be an issue in `2024.2.1` switching back to `2024.2.0`.
> 
> @oxigen-dev you are sure you used `2024.2.0` and not `2024.2.0`?

I was using ghcr.io/esphome/esphome and 2024.2.1 is tagged as latest:
[image]

fsaris, 2024-03-03:
@oxigen-dev can you give current `dev` a try?

oxigen-dev, 2024-03-03:
> @oxigen-dev can you give current `dev` a try?

No issues until now although I would need a few days of use before a final feedback.

oxigen-dev, 2024-03-05:
> > @oxigen-dev can you give current `dev` a try?
> 
> No issues until now although I would need a few days of use before a final feedback.

It was working without any issue :)

fsaris, 2024-03-07:
Merged to main version https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/releases/tag/2024.3.0

closed 2024-03-07 by fsaris