fsaris/EspHome-AwoX-BLE-mesh-hub #83: 2024.3.1 connection issues

oxigen-dev, 2024-03-12:
Hi @fsaris ,

I'm facing a new issue with 2024.3.1. Unfortunately sometimes ESP can't connect to one of my ceilings.
Actual situation: only the `OFFICE` ceiling is turned on and ESP can't connect to it. After restarting ESP, connection is successful.

Can you please check it? Thanks!

Log:
```
INFO ESPHome 2024.2.2
INFO Reading configuration awox-ble-mesh-hub.yaml...
INFO Updating https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub.git@None
INFO Starting log output from awox-ble-mesh-hub/debug
INFO Connected to MQTT broker!
[21:52:43][W][component:215]: Components should block for at most 20-30ms.
[21:52:48][D][esp-idf:000]: W (86735692) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x100

[21:52:48][D][esp-idf:000]: W (86735700) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x100

[21:52:48][D][esp32_ble_client:172]: [1] [LIVING_ROOM_TV] ESP_GATTC_DISCONNECT_EVT, reason 256
[21:52:48][I][awox.connection:117]: IDLE
[21:52:48][D][awox.connection:138]: [1] [LIVING_ROOM_TV] ESP_GATTC_DISCONNECT_EVT, reason 256
[21:52:48][I][awox.mesh:043]: disconnected
[21:52:48][D][awox.mesh:407]: Delayed publish online/offline for 31615 - offline
[21:52:48][I][awox.connection:129]: READY_TO_CONNECT
[21:52:48][I][esp32_ble_client:067]: [0] [LIVING_ROOM_SERVING_TRAY] 0x00 Attempting BLE connection
[21:52:48][I][awox.connection:133]: CONNECTING
[21:52:52][D][awox.mesh:227]: Total devices: 6
[21:52:52][D][awox.mesh:230]: Available device LIVING_ROOM_SERVING_TRAY [19898] => rssi: -70
[21:52:52][D][awox.mesh:230]: Available device LIVING_ROOM_TV [31615] => rssi: -77
[21:52:52][D][awox.mesh:230]: Available device LIVING_ROOM_COUCH [32569] => rssi: -83
[21:52:52][D][awox.mesh:230]: Available device LIVING_ROOM_DINING [3969] => rssi: -85
[21:52:52][D][awox.mesh:230]: Available device OFFICE [8479] => rssi: -9999
[21:52:52][D][awox.mesh:230]: Available device KIDS_ROOM [0] => rssi: -9999
[21:52:52][D][awox.mesh:255]: Currently 0 mesh devices reachable through active connections (5 currently known and 5 fully recognized)
[21:52:52][D][awox.mesh:267]: Try to connecty to device LIVING_ROOM_TV [31615] no active connection found for this device
[21:52:52][I][awox.mesh:139]: Try to connect LIVING_ROOM_TV => rssi: -77
[21:52:52][I][awox.connection:117]: IDLE
[21:52:52][D][esp32_ble_client:110]: [1] [LIVING_ROOM_TV] Found device
[21:52:52][I][awox.connection:125]: DISCOVERED
[21:52:52][W][component:214]: Component awox_mesh took a long time for an operation (0.07 s).
[21:52:52][W][component:215]: Components should block for at most 20-30ms.
[21:53:12][D][sensor:094]: 'Awox BLE Mesh Hub Uptime Sensor': Sending state 86758.25781 s with 0 decimals of accuracy
[21:53:15][D][sensor:094]: 'Awox BLE Mesh Hub Heap Free': Sending state 64668.00000 B with 0 decimals of accuracy
[21:53:15][D][sensor:094]: 'Awox BLE Mesh Hub Heap Max Block': Sending state 43008.00000 B with 0 decimals of accuracy
[21:53:15][D][sensor:094]: 'Awox BLE Mesh Hub Loop Time': Sending state 102.00000 ms with 0 decimals of accuracy
[21:53:18][D][esp-idf:000]: W (86765740) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x100

[21:53:18][D][esp-idf:000]: W (86765747) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x100

[21:53:18][D][esp32_ble_client:172]: [0] [LIVING_ROOM_SERVING_TRAY] ESP_GATTC_DISCONNECT_EVT, reason 256
[21:53:18][I][awox.connection:117]: IDLE
[21:53:18][D][awox.connection:138]: [0] [LIVING_ROOM_SERVING_TRAY] ESP_GATTC_DISCONNECT_EVT, reason 256
[21:53:18][I][awox.mesh:043]: disconnected
[21:53:18][D][awox.mesh:407]: Delayed publish online/offline for 19898 - offline
[21:53:18][I][awox.connection:129]: READY_TO_CONNECT
[21:53:18][I][esp32_ble_client:067]: [1] [LIVING_ROOM_TV] 0x00 Attempting BLE connection
[21:53:18][I][awox.connection:133]: CONNECTING
[21:53:22][D][awox.mesh:227]: Total devices: 6
[21:53:22][D][awox.mesh:230]: Available device LIVING_ROOM_SERVING_TRAY [19898] => rssi: -70
[21:53:22][D][awox.mesh:230]: Available device LIVING_ROOM_TV [31615] => rssi: -77
[21:53:22][D][awox.mesh:230]: Available device LIVING_ROOM_COUCH [32569] => rssi: -83
[21:53:22][D][awox.mesh:230]: Available device LIVING_ROOM_DINING [3969] => rssi: -85
[21:53:22][D][awox.mesh:230]: Available device OFFICE [8479] => rssi: -9999
[21:53:22][D][awox.mesh:230]: Available device KIDS_ROOM [0] => rssi: -9999
[21:53:22][D][awox.mesh:255]: Currently 0 mesh devices reachable through active connections (5 currently known and 5 fully recognized)
[21:53:22][D][awox.mesh:267]: Try to connecty to device LIVING_ROOM_SERVING_TRAY [19898] no active connection found for this device
[21:53:22][I][awox.mesh:139]: Try to connect LIVING_ROOM_SERVING_TRAY => rssi: -70
[21:53:22][I][awox.connection:117]: IDLE
[21:53:22][D][esp32_ble_client:110]: [0] [LIVING_ROOM_SERVING_TRAY] Found device
[... 37 lines cut ...]

[21:54:17][D][esp32_ble_client:172]: [0] [LIVING_ROOM_SERVING_TRAY] ESP_GATTC_DISCONNECT_EVT, reason 256
[21:54:17][I][awox.connection:117]: IDLE
[21:54:17][D][awox.connection:138]: [0] [LIVING_ROOM_SERVING_TRAY] ESP_GATTC_DISCONNECT_EVT, reason 256
[21:54:17][I][awox.mesh:043]: disconnected
[21:54:17][D][awox.mesh:407]: Delayed publish online/offline for 19898 - offline
[21:54:17][I][awox.connection:129]: READY_TO_CONNECT
[21:54:17][I][esp32_ble_client:067]: [1] [LIVING_ROOM_TV] 0x00 Attempting BLE connection
[21:54:17][I][awox.connection:133]: CONNECTING
[21:54:17][D][awox.mesh:227]: Total devices: 6
[21:54:17][D][awox.mesh:230]: Available device LIVING_ROOM_SERVING_TRAY [19898] => rssi: -70
[21:54:17][D][awox.mesh:230]: Available device LIVING_ROOM_TV [31615] => rssi: -77
[21:54:17][D][awox.mesh:230]: Available device LIVING_ROOM_COUCH [32569] => rssi: -83
[21:54:17][D][awox.mesh:230]: Available device LIVING_ROOM_DINING [3969] => rssi: -85
[21:54:17][D][awox.mesh:230]: Available device OFFICE [8479] => rssi: -9999
[21:54:17][D][awox.mesh:230]: Available device KIDS_ROOM [0] => rssi: -9999
[21:54:17][D][awox.mesh:255]: Currently 0 mesh devices reachable through active connections (5 currently known and 5 fully recognized)
[21:54:17][D][awox.mesh:267]: Try to connecty to device LIVING_ROOM_SERVING_TRAY [19898] no active connection found for this device
[21:54:17][I][awox.mesh:139]: Try to connect LIVING_ROOM_SERVING_TRAY => rssi: -70
[21:54:17][I][awox.connection:117]: IDLE
[21:54:17][D][esp32_ble_client:110]: [0] [LIVING_ROOM_SERVING_TRAY] Found device
[21:54:17][I][awox.connection:125]: DISCOVERED
[21:54:17][W][component:214]: Component awox_mesh took a long time for an operation (0.09 s).
[21:54:17][W][component:215]: Components should block for at most 20-30ms.
```

fsaris, 2024-03-13:
@oxigen-dev are there no `Found Awox device ...` messages in log? 

I would have expected the RSSI of the other found devices also to be `-9999`

And when does this start, after a few minutes or after a few hours?

oxigen-dev, 2024-03-13:
> @oxigen-dev are there no `Found Awox device ...` messages in log? 
> 
> I would have expected the RSSI of the other found devices also to be `-9999`
> 
> And when does this start, after a few minutes or after a few hours?

After restarting the device next day same time it never worked. In the middle of the day it was still working.

oxigen-dev, 2024-03-13:
Today the very same happened. I've created a log before and after restarting the ESP.

Before restarting:
```
INFO ESPHome 2024.2.2
INFO Reading configuration awox-ble-mesh-hub.yaml...
INFO Updating https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub.git@None
INFO Starting log output from awox-ble-mesh-hub/debug
INFO Connected to MQTT broker!
[22:04:10][D][sensor:094]: 'Awox BLE Mesh Hub Uptime Sensor': Sending state 86676.90625 s with 0 decimals of accuracy
[22:04:11][D][esp-idf:000]: W (86682442) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x100

[22:04:11][D][esp-idf:000]: W (86682448) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x100

[22:04:11][D][esp32_ble_client:172]: [1] [LIVINGROOM TV] ESP_GATTC_DISCONNECT_EVT, reason 256
[22:04:11][I][awox.connection:117]: IDLE
[22:04:11][D][awox.connection:138]: [1] [LIVINGROOM TV] ESP_GATTC_DISCONNECT_EVT, reason 256
[22:04:11][I][awox.mesh:043]: disconnected
[22:04:11][D][awox.mesh:407]: Delayed publish online/offline for 31615 - offline
[22:04:11][I][awox.connection:129]: READY_TO_CONNECT
[22:04:11][I][esp32_ble_client:067]: [0] [LIVINGROOM SERVING TRAY] 0x00 Attempting BLE connection
[22:04:11][I][awox.connection:133]: CONNECTING
[22:04:16][D][awox.mesh:227]: Total devices: 5
[22:04:16][D][awox.mesh:230]: Available device LIVINGROOM SERVING TRAY [19898] => rssi: -70
[22:04:16][D][awox.mesh:230]: Available device LIVINGROOM TV [31615] => rssi: -78
[22:04:16][D][awox.mesh:230]: Available device LIVINGROOM COUCH [32569] => rssi: -9999
[22:04:16][D][awox.mesh:230]: Available device LIVINGROOM DINING [3969] => rssi: -9999
[22:04:16][D][awox.mesh:230]: Available device OFFICE [8479] => rssi: -9999
[22:04:16][D][awox.mesh:255]: Currently 0 mesh devices reachable through active connections (5 currently known and 5 fully recognized)
[22:04:16][D][awox.mesh:267]: Try to connecty to device LIVINGROOM TV [31615] no active connection found for this device
[22:04:16][I][awox.mesh:139]: Try to connect LIVINGROOM TV => rssi: -78
[22:04:16][I][awox.connection:117]: IDLE
[22:04:16][D][esp32_ble_client:110]: [1] [LIVINGROOM TV] Found device
[22:04:16][I][awox.connection:125]: DISCOVERED
[22:04:16][W][component:214]: Component awox_mesh took a long time for an operation (0.07 s).
[22:04:16][W][component:215]: Components should block for at most 20-30ms.
[22:04:31][D][sensor:094]: 'Awox BLE Mesh Hub Heap Free': Sending state 65152.00000 B with 0 decimals of accuracy
[22:04:31][D][sensor:094]: 'Awox BLE Mesh Hub Heap Max Block': Sending state 45056.00000 B with 0 decimals of accuracy
[22:04:31][D][sensor:094]: 'Awox BLE Mesh Hub Loop Time': Sending state 89.00000 ms with 0 decimals of accuracy
[22:04:41][D][esp-idf:000]: W (86712498) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x100

[22:04:41][D][esp-idf:000]: W (86712502) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x100

[22:04:41][D][esp32_ble_client:172]: [0] [LIVINGROOM SERVING TRAY] ESP_GATTC_DISCONNECT_EVT, reason 256
[22:04:41][I][awox.connection:117]: IDLE
[22:04:41][D][awox.connection:138]: [0] [LIVINGROOM SERVING TRAY] ESP_GATTC_DISCONNECT_EVT, reason 256
[22:04:41][I][awox.mesh:043]: disconnected
[22:04:41][D][awox.mesh:407]: Delayed publish online/offline for 19898 - offline
[22:04:41][I][awox.connection:129]: READY_TO_CONNECT
[22:04:41][I][esp32_ble_client:067]: [1] [LIVINGROOM TV] 0x00 Attempting BLE connection
[22:04:41][I][awox.connection:133]: CONNECTING
[22:04:46][D][awox.mesh:227]: Total devices: 5
[22:04:46][D][awox.mesh:230]: Available device LIVINGROOM SERVING TRAY [19898] => rssi: -70
[22:04:46][D][awox.mesh:230]: Available device LIVINGROOM TV [31615] => rssi: -78
[22:04:46][D][awox.mesh:230]: Available device LIVINGROOM COUCH [32569] => rssi: -9999
[22:04:46][D][awox.mesh:230]: Available device LIVINGROOM DINING [3969] => rssi: -9999
[22:04:46][D][awox.mesh:230]: Available device OFFICE [8479] => rssi: -9999
[22:04:46][D][awox.mesh:255]: Currently 0 mesh devices reachable through active connections (5 currently known and 5 fully recognized)
[22:04:46][D][awox.mesh:267]: Try to connecty to device LIVINGROOM SERVING TRAY [19898] no active connection found for this device
[22:04:46][I][awox.mesh:139]: Try to connect LIVINGROOM SERVING TRAY => rssi: -70
[22:04:46][I][awox.connection:117]: IDLE
[22:04:46][D][esp32_ble_client:110]: [0] [LIVINGROOM SERVING TRAY] Found device
[22:04:46][I][awox.connection:125]: DISCOVERED
[22:04:46][W][component:214]: Component awox_mesh took a long time for an operation (0.07 s).
[22:04:46][W][component:215]: Components should block for at most 20-30ms.
```

After restarting:
```
#[22:08:17][D][text_sensor:064]: 'Awox BLE Mesh Hub Device Info': Sending state '2024.2.2|Chip: ESP32 Features:WIFI_BGN,BLE,BT, Cores:2 Revision:3|ESP-IDF: 4.4.6|EFuse MAC: A8:42:E3:8D:B8:64|Reset: Software Reset CPU|Wakeup: Unknown'
[... 77 lines cut ...]
[22:08:24][D][awox.mesh:477]: Found OFFICE as connection
[22:08:24][D][awox.mesh.mqtt:626]: [8479] Process command state
[22:08:24][I][awox.connection:121]: 2 queued commands (debounce timer: -17, next command DD, for dest: 8479)
[22:08:24][D][awox.connection:541]: [0] [OFFICE] [8479] write_command packet DD => 0000101000000001
[22:08:24][I][awox.connection:115]: still 1 queued commands
[22:08:24][D][esp32_ble_client:110]: [0] [OFFICE] ESP_GATTC_WRITE_CHAR_EVT
[22:08:24][I][awox.connection:121]: 1 queued commands (debounce timer: -153, next command F1, for dest: 8479)
[... cut ...]
[22:08:24][D][awox.connection:541]: [0] [OFFICE] [8479] write_command packet F1 => 00111011
[22:08:24][D][esp32_ble_client:110]: [0] [OFFICE] ESP_GATTC_WRITE_CHAR_EVT
[22:08:24][D][esp32_ble_client:110]: [0] [OFFICE] ESP_GATTC_NOTIFY_EVT
[22:08:24][D][awox.connection:355]: online status report: mesh: 8479, on: 1, color_mode: 0, sequence_mode: 0, candle_mode: 0, w_b: 59, temp: 25, c_b: 46, rgb: FF0089, mode: 1 00000001
[22:08:24][I][awox.connection:458]: 8479: ON temp: 25 (59 %) ONLINE
[22:08:25][D][awox.mesh:088]: Found Awox device  - OFFICE. RSSI: -75 dB (total devices: 1)
```

oxigen-dev, 2024-03-18:
> Can you test the dev branch? I have logging set to verbose and for me it runs now more then a day. It should be reporting "Found device..." every few minutes

I've just started the flashing process. I'll come back in a few days. Thanks!

spider7611, 2024-03-18:
Hi,
I have problem too. I see in the history always connect then disconnect. Use dev esphome.
`INFO ESPHome 2024.4.0-dev
INFO Reading configuration /config/nappaliesp32.yaml...
INFO Starting log output from nappaliesp32/debug
INFO Connected to MQTT broker!
[18:53:43][D][awox.mesh:129]: No devices found to connect to
[18:54:00][D][awox.mesh:227]: Total devices: 2
[18:54:00][D][awox.mesh:230]: Available device A4:C1:38:A7:FA:55 [0] => rssi: -92
[18:54:00][D][awox.mesh:230]: Available device A4:C1:38:14:71:EB [29163] => rssi: -9999
[18:54:00][D][awox.mesh:255]: Currently 1 mesh devices reachable through active connections (1 currently known and 1 fully recognized)
[18:54:00][D][awox.mesh:129]: No devices found to connect to`

fsaris, 2024-03-18:
@spider7611 are you using the `dev` branch of this component?

oxigen-dev, 2024-03-18:
@fsaris Unfortunately the issue appeared again. I've created logs before and after restarting the ESP. Physical situation: all of the ceilings were switched off except for the `OFFICE`.
<details>
  <summary>Before restarting:</summary>
  <code>
INFO ESPHome 2024.2.2
INFO Reading configuration awox-ble-mesh-hub.yaml...
INFO Cloning https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub.git@None
INFO Starting log output from awox-ble-mesh-hub/debug
INFO Connected to MQTT broker!
[21:37:44][W][component:215]: Components should block for at most 20-30ms.
[21:37:58][D][esp-idf:000]: W (16371129) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x100
[21:37:58][D][esp-idf:000]: W (16371137) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x100
[21:37:58][D][esp32_ble_client:172]: [0] [SERVING_TRAY] ESP_GATTC_DISCONNECT_EVT, reason 256
[21:37:58][I][awox.connection:117]: IDLE
[21:37:58][D][awox.connection:138]: [0] [SERVING_TRAY] ESP_GATTC_DISCONNECT_EVT, reason 256
[21:37:58][I][awox.mesh:043]: disconnected
[21:37:58][D][awox.mesh:407]: Delayed publish online/offline for 19898 - offline
[21:37:58][I][awox.connection:129]: READY_TO_CONNECT
[21:37:58][I][esp32_ble_client:067]: [1] [COUCH] 0x00 Attempting BLE connection
[21:37:58][I][awox.connection:133]: CONNECTING
[21:38:00][D][sensor:094]: 'Awox BLE Mesh Hub Uptime Sensor': Sending state 16372.44336 s with 0 decimals of accuracy
[21:38:01][D][awox.mesh:227]: Total devices: 5
[21:38:01][D][awox.mesh:230]: Available device SERVING_TRAY [19898] => rssi: -70
[21:38:01][D][awox.mesh:230]: Available device COUCH [32569] => rssi: -74
[21:38:01][D][awox.mesh:230]: Available device DINING [3969] => rssi: -77
[21:38:01][D][awox.mesh:230]: Available device TV [31615] => rssi: -84
[21:38:01][D][awox.mesh:230]: Available device OFFICE [8479] => rssi: -9999
[21:38:01][D][awox.mesh:255]: Currently 0 mesh devices reachable through active connections (5 currently known and 5 fully recognized)
[21:38:01][D][awox.mesh:267]: Try to connecty to device SERVING_TRAY [19898] no active connection found for this device
[21:38:01][I][awox.mesh:139]: Try to connect SERVING_TRAY => rssi: -70
[21:38:01][I][awox.connection:117]: IDLE
[21:38:01][D][esp32_ble_client:110]: [0] [SERVING_TRAY] Found device
[21:38:01][I][awox.connection:125]: DISCOVERED
[21:38:01][W][component:214]: Component awox_mesh took a long time for an operation (0.07 s).
[21:38:01][W][component:215]: Components should block for at most 20-30ms.
[21:38:11][D][sensor:094]: 'Awox BLE Mesh Hub Heap Free': Sending state 65328.00000 B with 0 decimals of accuracy
[21:38:11][D][sensor:094]: 'Awox BLE Mesh Hub Heap Max Block': Sending state 45056.00000 B with 0 decimals of accuracy
[21:38:11][D][sensor:094]: 'Awox BLE Mesh Hub Loop Time': Sending state 94.00000 ms with 0 decimals of accuracy
[21:38:28][D][esp-idf:000]: W (16401180) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x100
[21:38:28][D][esp-idf:000]: W (16401187) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x100
[21:38:28][D][esp32_ble_client:172]: [1] [COUCH] ESP_GATTC_DISCONNECT_EVT, reason 256
[21:38:28][I][awox.connection:117]: IDLE
[21:38:28][D][awox.connection:138]: [1] [COUCH] ESP_GATTC_DISCONNECT_EVT, reason 256
[21:38:28][I][awox.mesh:043]: disconnected
[21:38:28][D][awox.mesh:407]: Delayed publish online/offline for 32569 - offline
[21:38:28][I][awox.connection:129]: READY_TO_CONNECT
[21:38:28][I][esp32_ble_client:067]: [0] [SERVING_TRAY] 0x00 Attempting BLE connection
[21:38:28][I][awox.connection:133]: CONNECTING
[21:38:31][D][awox.mesh:227]: Total devices: 5
[21:38:31][D][awox.mesh:230]: Available device SERVING_TRAY [19898] => rssi: -70
[21:38:31][D][awox.mesh:230]: Available device COUCH [32569] => rssi: -74
[21:38:31][D][awox.mesh:230]: Available device DINING [3969] => rssi: -77
[21:38:31][D][awox.mesh:230]: Available device TV [31615] => rssi: -84
[21:38:31][D][awox.mesh:230]: Available device OFFICE [8479] => rssi: -9999
[21:38:31][D][awox.mesh:255]: Currently 0 mesh devices reachable through active connections (5 currently known and 5 fully recognized)
[21:38:31][D][awox.mesh:267]: Try to connecty to device COUCH [32569] no active connection found for this device
[21:38:31][I][awox.mesh:139]: Try to connect COUCH => rssi: -74
[21:38:31][I][awox.connection:117]: IDLE
[21:38:31][D][esp32_ble_client:110]: [1] [COUCH] Found device
[21:38:31][I][awox.connection:125]: DISCOVERED
[21:38:31][W][component:214]: Component awox_mesh took a long time for an operation (0.07 s).
[21:38:31][W][component:215]: Components should block for at most 20-30ms.
</code>
</details>

<details>
  <summary>After restarting:</summary>
<code>
[21:42:21][I][mqtt:274]: MQTT Connected!
[21:42:21][I][app:062]: setup() finished successfully!
[... 163 lines cut ...]
[21:42:30][D][awox.mesh.mqtt:589]: [8479] Process command white_brightness 117
[21:42:30][D][awox.mesh:474]: Call connection for 8479
[21:42:30][D][awox.mesh:477]: Found OFFICE as connection
[21:42:30][D][awox.mesh.mqtt:626]: [8479] Process command state
[21:42:30][D][esp32_ble_client:110]: [0] [OFFICE] ESP_GATTC_WRITE_CHAR_EVT
[21:42:30][I][awox.connection:121]: 1 queued commands (debounce timer: -135, next command F1, for dest: 8479)
[21:42:30][I][awox.connection:121]: 1 queued commands (debounce timer: -115, next command F1, for dest: 8479)
[21:42:30][D][esp32_ble_client:110]: [0] [OFFICE] ESP_GATTC_NOTIFY_EVT
[21:42:30][D][awox.connection:439]: Group report, dev [8479] 4
[... cut ...]
[21:42:30][D][esp32_ble_client:110]: [0] [OFFICE] ESP_GATTC_WRITE_CHAR_EVT
[21:42:30][D][esp32_ble_client:110]: [0] [OFFICE] ESP_GATTC_NOTIFY_EVT
[21:42:30][D][awox.connection:383]: online status report: mesh: 8479, on: 1, color_mode: 0, sequence_mode: 0, candle_mode: 0, w_b: 59, temp: 25, c_b: 46, rgb: FF0089, mode: 1 00000001
[21:42:30][I][awox.connection:486]: 8479: ON temp: 25 (59 %) ONLINE
[21:42:31][I][awox.mesh.mqtt:082]: Publish connection info, 1 connections, 1 device online
</code>
</details>

fsaris, 2024-03-19:
@oxigen-dev looks that the `esp32_ble_tracker` stops looking for devices. Can you provide some more logs? Do you see during the day `Found Awox device` messages every few seconds/minutes? 

What settings do you use for `esp32_ble_tracker`?

Maybe you can catch all `awox-ble-mesh-hub/debug` topics in a sensor to get a full list. I'm curious if there are warnings before the  `esp32_ble_tracker` stops.

oxigen-dev, 2024-03-19:
> @oxigen-dev looks that the `esp32_ble_tracker` stops looking for devices. Can you provide some more logs? Do you see during the day `Found Awox device` messages every few seconds/minutes?
>
> What [...]

I'll logging for a whole day. Should I go with verbose?

fsaris, 2024-03-19:
> Should I go with verbose?

Could possible help to get more info about why scanning stops

poisonedy, 2024-03-20:
Hi @fsaris ,
I've found it's working but not if I power off the esp device. Also i've noticed if i stress it with Hyperion Addon, trying to reproduce colors from my screen I get same result. 
`[20:22:44][D][awox.mesh:230]: Available device [20:22:44][D][awox.mesh:230]: Available device A0:00:00:00:00:11 [0] => rssi: -78
[20:22:44][D][awox.mesh:255]: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
[20:22:44][D][awox.mesh:261]: Try to connecty to device A0:00:00:00:00:00 no mesh id known yet
[20:22:44][I][awox.mesh:139]: Try to connect A0:00:00:00:00:00 => rssi: -78
[20:22:44][I][awox.connection:117]: IDLE
[20:22:44][D][esp32_ble_client:110]: [0] [[20:22:44][D][awox.mesh:230]: Available device A0:00:00:00:00:11 [0] => rssi: -78
[20:22:44][D][awox.mesh:255]: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
[20:22:44][D][awox.mesh:261]: Try to connecty to device A0:00:00:00:00:00 no mesh id known yet
[20:22:44][I][awox.mesh:139]: Try to connect A0:00:00:00:00:00 => rssi: -78
[20:22:44][I][awox.connection:117]: IDLE
[20:22:44][D][esp32_ble_client:110]: [0] [A0:00:00:00:00:00] Found device
[20:22:44][D][esp32_ble_tracker:661]: Found device A0:00:00:00:00:00 RSSI=-73
[20:22:44][D][esp32_ble_tracker:682]:   Address Type: PUBLIC
[20:22:44][D][esp32_ble_tracker:684]:   Name: 'uXMuXltb'
[20:22:44][I][awox.connection:125]: DISCOVERED
[20:22:44][W][component:214]: Component awox_mesh took a long time for an operation (0.05 s).
[20:22:44][W][component:215]: Components should block for at most 20-30ms.
[20:22:44][D][esp32_ble_tracker:215]: Pausing scan to make connection...
[20:22:44][I][awox.connection:129]: READY_TO_CONNECT
[20:22:44][I][esp32_ble_client:067]: [0] [A0:00:00:00:00:00] 0x00 Attempting BLE connection
[20:22:44][I][awox.connection:133]: CONNECTING
[20:22:45][D][esp-idf:000]: W (303485) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[20:22:45][D][esp-idf:000]: W (303839) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[20:22:46][D][esp-idf:000]: W (304787) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[20:22:48][D][esp-idf:000]: W (305915) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x3e

[20:22:48][D][esp-idf:000]: W (305920) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[20:22:48][D][esp32_ble_client:172]: [0] [A0:00:00:00:00:00] ESP_GATTC_DISCONNECT_EVT, reason 62
[20:22:48][I][awox.connection:117]: IDLE
[20:22:48][D][awox.connection:138]: [0] [A0:00:00:00:00:00] ESP_GATTC_DISCONNECT_EVT, reason 62
[20:22:48][I][awox.mesh:043]: disconnected
[20:22:48][D][esp32_ble_tracker:266]: Starting scan...
[20:22:48][D][awox.mesh:117]: scan end
[20:22:48][D][awox.mesh:227]: Total devices: 2
[20:22:48][D][awox.mesh:230]: Available device A0:00:00:00:00:00 [0] => rssi: -78
[20:22:48][D][awox.mesh:230]: Available device A0:00:00:00:00:11 [0] => rssi: -78
[20:22:48][D][awox.mesh:255]: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
[20:22:48][D][awox.mesh:261]: Try to connecty to device A0:00:00:00:00:00 no mesh id known yet
[20:22:48][I][awox.mesh:139]: Try to connect A0:00:00:00:00:00 => rssi: -78
[20:22:48][I][awox.connection:117]: IDLE
[20:22:48][D][esp32_ble_client:110]: [0] [A0:00:00:00:00:00] Found device
[20:22:48][D][esp32_ble_tracker:661]: Found device A0:00:00:00:00:00 RSSI=-73
[20:22:48][D][esp32_ble_tracker:682]:   Address Type: PUBLIC
[20:22:48][D][esp32_ble_tracker:684]:   Name: 'uXMuXltb'
[20:22:48][I][awox.connection:125]: DISCOVERED
[20:22:48][W][component:214]: Component awox_mesh took a long time for an operation (0.05 s).
[20:22:48][W][component:215]: Components should block for at most 20-30ms.
[20:22:48][D][esp32_ble_tracker:215]: Pausing scan to make connection...
[20:22:48][D][esp32_ble_tracker:215]: Pausing scan to make connection...
[20:22:48][D][esp-idf:000]: E (306926) BT_BTM: BTM_BleScan scan not active

[20:22:48][D][esp-idf:000]: W (306931) BT_APPL:  bta_dm_ble_scan stop scan failed, status=0x6

[20:22:49][D][esp-idf:000]: W (307757) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[20:22:50][D][esp-idf:000]: W (308510) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[20:22:50][D][esp-idf:000]: W (308687) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[20:22:51][D][esp-idf:000]: W (309800) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x3e

[20:22:51][D][esp-idf:000]: W (309805) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[20:22:51][D][esp32_ble_client:172]: [0] [A0:00:00:00:00:00] ESP_GATTC_DISCONNECT_EVT, reason 62
[... 181 lines cut ...]
[20:22:51][D][awox.mesh:117]: scan end
[20:22:52][D][awox.mesh:088]: Found Awox device uXMuXltb - A0:00:00:00:00:11. RSSI: -78 dB (total devices: 2)
[20:22:53][D][awox.mesh:088]: Found Awox device uXMuXltb - A0:00:00:00:00:00. RSSI: -72 dB (total devices: 2)
[20:22:53][D][awox.mesh:088]: Found Awox device uXMuXltb - A0:00:00:00:00:00. RSSI: -72 dB (total devices: 2)
[20:22:53][D][awox.mesh:088]: Found Awox device  - A0:00:00:00:00:11. RSSI: -82 dB (total devices: 2)
[20:22:55][D][awox.mesh:227]: Total devices: 2
[20:22:55][D][awox.mesh:230]: Available device A0:00:00:00:00:00 [0] => rssi: -72
[20:22:55][D][awox.mesh:230]: Available device A0:00:00:00:00:11 [0] => rssi: -82
[20:22:55][D][awox.mesh:255]: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
[... cut ...]
[20:22:55][W][component:214]: Component awox_mesh took a long time for an operation (0.05 s).
[20:22:55][W][component:215]: Components should block for at most 20-30ms.
[20:22:55][D][esp32_ble_tracker:215]: Pausing scan to make connection...
[20:22:55][I][awox.connection:129]: READY_TO_CONNECT
[20:22:55][I][esp32_ble_client:067]: [0] [A0:00:00:00:00:00] 0x00 Attempting BLE connection
[20:22:55][I][awox.connection:133]: CONNECTING
[20:22:55][D][esp-idf:000]: W (312396) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
`

oxigen-dev, 2024-03-20:
> > Should I go with verbose?
> 
> Could possible help to get more info about why scanning stops

Hello @fsaris ,

I was logging the whole day with log level Verbose. Please find the file attached.

- Everything was off at 19:43.
- Office ceiling was turned on at 19:44, this is the first time I've experienced the issue (Ceiling stayed offline in HA). It was then turned off.
- TV Ceiling and Serving tray Ceiling were turned on at 20:11 (they can be turned on with the same switch). Nothing happened, everything stayed Offline. Everything was turned off.
- Right after this, Couch Ceiling and Dining Ceiling were turned on  (they can be turned on with the same switch). Connection was successful, they appeared as Online in HA.
- I've turned on the Office Ceiling (Couch and Dining were still on), connection was successful.
- I've turned off everything so everything went to Offline in HA.
- I've turned on just the Office Ceiling again and it stayed Offline in HA.

If I understand correctly, the Hub thinks sometimes that the physically turned off Ceilings are available (eg. RSSI=-71) while the one which is physically turned off has RSSI=-9999.
[awox-20240320.zip](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/14673229/awox-20240320.zip)

fsaris, 2024-03-20:
> I've found it's working but not if I power off the esp device and I plug in again. 

So after a reboot it fails? What version are you using? Did you try the `dev` branch already?

Stressing the esp will probably not help, it's only a low specs device

oxigen-dev, 2024-03-20:
> @oxigen-dev looks that the `esp32_ble_tracker` stops looking for devices. Can you provide some more logs? Do you see during the day `Found Awox device` messages every few seconds/minutes?
>
> What [...]

```
esp32_ble_tracker:
  scan_parameters:
    active: false
```

poisonedy, 2024-03-21:
> > I've found it's working but not if I power off the esp device and I plug in again.
> 
> So after a reboot it fails? What version are you using? Did you try the `dev` branch already?
> 
> Stressing the esp will probably not help, it's only a low specs device

Using the dev branch.

Yes, after reboot (disconnecting esp32 and plugging it again), or after some stress.
Workaround: Flash again the esp32 with the same code, and then again it finds the mesh.

What i mean about "stress the device"? I mean send commands (change color, brightness,..) trough Home Assistant API repeatedly because I'm using Hyperion, it changes the color of the light based on the colors of my computer screen, so I don't think  it's esp32 having problems to process it by specs. It works some time (5 minutes, 10 minutes...) and then it stops working. (looks like esp32 reboots and then it cannot find the mesh). Then I flash it again and it works for some more time, but still stops again. Looks like some kind of overflow because the esp32 reboots, and it happens early or late (early if I send lots of commands with Hyperion, late with a normal use without Hyperion).

I would love to help so if you need more tests or feedback don't hesitate to ask.

fsaris, 2024-03-21:
```
esp32_ble_tracker:
  scan_parameters:
    active: false
```

@oxigen-dev any reason to have `active: false`?

oxigen-dev, 2024-03-21:
> ```
> esp32_ble_tracker:
>   scan_parameters:
>     active: false
> ```
> 
> @oxigen-dev any reason to have `active: false`?

Main reason: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/blob/main/awox-ble-mesh-hub.yaml

It was working perfectly for months, before big changes in ESPHome and Awox Hub.

fsaris, 2024-03-21:
😄 

Could you try with `active: true`?

poisonedy, 2024-03-21:
I there,
After some more testing I decided to assign one esp32 per rgb using the mac filter. Now it seems to be more stable, even stressing it. 
The esp32 is able to stay connected to the mesh more time than if i use one esp32 for two rgb, besides that I got some disconnections that i ve workaround flashing again.
My config as follows:
```
esphome:
  name: awox-esp32-mesh-control
  friendly_name: AwoX esp32 mesh control

esp32:
  board: esp32dev
  framework:
    type: esp-idf
# Enable logging
logger:
status_led:
  # Match this with the board your using
  pin: GPIO2

ota:
  password: "838ca87c56b8"

wifi:
  ssid: "4000D"
  password: "shadowed"
  manual_ip:
    static_ip: 192.168.1.110
    gateway: 192.168.1.1
    subnet: 255.255.255.0

external_components:
  source:
    type: git
    url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub.git
    ref: dev

# Enable logging

mqtt:
  broker: 192.168.1.111
  username: shadowed
  password: shadowed

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: True

awox_mesh:
  mesh_name: "uXMuXltb"
  mesh_password: "5f8cd677"
  max_connections: 2
  min_rssi: -100

  allowed_mac_addresses:
    - A4:C1:38:00:00:00

  device_info:
    - product_id: 0x96
      device_type: RGB
      name: EGLO RGB+TW
      model: EGLO-RGB-TW
      manufacturer: EGLO
    

```

oxigen-dev, 2024-03-22:
> 😄 
> 
> Could you try with `active: true`?

@fsaris I've tested it and it didn't help.

TotalPanther317, 2024-03-22:
Hi, I recently had the same issue when updating to 2024.3.1. For some reason it fixed itself by reinstalling esphome, deleting the esp's flash and reinstalling it onto the esp again. Also check if the mqtt host ip is still correct.

oxigen-dev, 2024-03-23:
> Hi, I recently had the same issue when updating to 2024.3.1. For some reason it fixed itself by reinstalling esphome, deleting the esp's flash and reinstalling it onto the esp again. Also check if the mqtt host ip is still correct.

A ladder was needed for this, but I'm now testing :). Thanks!

fsaris, 2024-03-23:
@oxigen-dev I update the debug logging a little. Could you try the latest dev version if you are still running into issues?

oxigen-dev, 2024-03-23:
> @oxigen-dev I update the debug logging a little. Could you try the latest dev version if you are still running into issues?  

I didn't have luck with erasing and reflashing, now I'm testing the current dev branch.

oxigen-dev, 2024-03-23:
Hi @fsaris ,

With the current dev branch connecting to the ceilings can take sometimes minutes although until now I didn't see unsuccessful connections.

Eg. one ceiling is turned on but I see this for minutes:
```
[22:14:52][D][awox.mesh:235]: Total devices: 5
[22:14:52][D][awox.mesh:238]: Available device A4:C1:[redacted] [8479] => rssi: -9999
[22:14:52][D][awox.mesh:238]: Available device A4:C1:[redacted] [19898] => rssi: -9999
[22:14:52][D][awox.mesh:238]: Available device A4:C1:[redacted] [32569] => rssi: -9999
[22:14:52][D][awox.mesh:238]: Available device A4:C1:[redacted] [3969] => rssi: -9999
[22:14:52][D][awox.mesh:238]: Available device A4:C1:[redacted] [31615] => rssi: -9999
[22:14:52][D][awox.mesh:263]: Currently 0 mesh devices reachable through active connections (5 currently known and 5 fully recognized)
```

fsaris, 2024-03-24:
The logic to mark devices as unavailable was changed a little. It was depending on new found devices. But now it will mark them RSSI -9999 if no devices were "found" by the BLE device tracker after 20seconds.

Can you post some more logs?

I would expect lines like these every 5 a 10 seconds:

```
[10:26:59][D][awox.mesh:085]: Skipped device 4IMBLHPA - A4:C1:38:55:1E:82. RSSI: -63, not in mac_addresses_allowed
[10:26:59][D][awox.mesh:092]: Found Awox device 4IMBLHPA - A4:C1:38:55:0C:A1. RSSI: -80 dB (total devices: 9)
[10:26:59][D][awox.mesh:092]: Found Awox device 4IMBLHPA - A4:C1:38:5A:1F:8E. RSSI: -86 dB (total devices: 9)
```

fsaris, 2024-03-24:
Also the `set_state . ..` logs will tell me a little bit more about what's happening with your ESP32. Somehow the `esp_ble_tracker` looks to get in a state that blocks the scan logic

fsaris, 2024-03-24:
@oxigen-dev just as double check, you did a clean build? (Clear build files).

oxigen-dev, 2024-03-24:
> @oxigen-dev just as double check, you did a clean build? (Clear build files).

I was doing `sudo rm -r .esphome` to delete build files then `sudo docker run --rm --privileged -v "${PWD}":/config -it ghcr.io/esphome/esphome run awox-ble-mesh-hub.yaml`

TotalPanther317, 2024-03-24:
Maybe you should do it through the esphome gui?

oxigen-dev, 2024-03-24:
> The logic to mark devices as unavailable was changed a little. It was depending on new found devices. But now it will mark them RSSI -9999 if no devices were "found" by the BLE device tracker after [...]

Please find a new log here.
<details>
  <summary>Log</summary>
  <code>
[10:20:58][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -90, address_prefix mismatch
[10:20:58][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -90, address_prefix mismatch
[10:20:59][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -90, address_prefix mismatch
[10:20:59][D][awox.mesh:085]: Skipped device  - [HALL_CEILING]. RSSI: -91, not in mac_addresses_allowed
[10:21:00][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -89, address_prefix mismatch
[10:21:00][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -96, address_prefix mismatch
[10:21:01][D][awox.mesh:235]: Total devices: 5
[10:21:01][D][awox.mesh:238]: Available device [OFFICE_CEILING] [8479] => rssi: -9999
[10:21:01][D][awox.mesh:238]: Available device [SERVING_TRAY_CEILING] [19898] => rssi: -9999
[10:21:01][D][awox.mesh:238]: Available device [COUCH_CEILING] [32569] => rssi: -9999
[10:21:01][D][awox.mesh:238]: Available device [DINING_CEILING] [3969] => rssi: -9999
[10:21:01][D][awox.mesh:238]: Available device [TV_CEILING] [31615] => rssi: -9999
[10:21:01][D][awox.mesh:263]: Currently 0 mesh devices reachable through active connections (5 currently known and 5 fully recognized)
[10:21:01][D][awox.mesh:133]: No devices found to connect to
[10:21:01][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -95, address_prefix mismatch
[10:21:02][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -96, address_prefix mismatch
[10:21:02][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -97, address_prefix mismatch
[10:21:02][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -97, address_prefix mismatch
[10:21:02][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -96, address_prefix mismatch
[10:21:03][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -91, address_prefix mismatch
[10:21:03][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -90, address_prefix mismatch
[10:21:04][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -91, address_prefix mismatch
[10:21:04][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -90, address_prefix mismatch
[10:21:05][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -102, address_prefix mismatch
[10:21:05][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -92, address_prefix mismatch
[10:21:06][D][awox.mesh:235]: Total devices: 5
[10:21:06][D][awox.mesh:238]: Available device [OFFICE_CEILING] [8479] => rssi: -9999
[10:21:06][D][awox.mesh:238]: Available device [SERVING_TRAY_CEILING] [19898] => rssi: -9999
[10:21:06][D][awox.mesh:238]: Available device [COUCH_CEILING] [32569] => rssi: -9999
[10:21:06][D][awox.mesh:238]: Available device [DINING_CEILING] [3969] => rssi: -9999
[10:21:06][D][awox.mesh:238]: Available device [TV_CEILING] [31615] => rssi: -9999
[10:21:06][D][awox.mesh:263]: Currently 0 mesh devices reachable through active connections (5 currently known and 5 fully recognized)
[10:21:06][D][awox.mesh:133]: No devices found to connect to
[10:21:07][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -96, address_prefix mismatch
[10:21:07][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -91, address_prefix mismatch
[10:21:08][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -91, address_prefix mismatch
[10:21:08][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -90, address_prefix mismatch
[10:21:09][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -90, address_prefix mismatch
[10:21:09][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -90, address_prefix mismatch
[10:21:09][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -90, address_prefix mismatch
[10:21:10][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -89, address_prefix mismatch
[10:21:10][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -101, address_prefix mismatch
[10:21:11][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -97, address_prefix mismatch
[10:21:11][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -99, address_prefix mismatch
[10:21:11][D][awox.mesh:235]: Total devices: 5
[10:21:11][D][awox.mesh:238]: Available device [OFFICE_CEILING] [8479] => rssi: -9999
[10:21:11][D][awox.mesh:238]: Available device [SERVING_TRAY_CEILING] [19898] => rssi: -9999
[10:21:11][D][awox.mesh:238]: Available device [COUCH_CEILING] [32569] => rssi: -9999
[10:21:11][D][awox.mesh:238]: Available device [DINING_CEILING] [3969] => rssi: -9999
[10:21:11][D][awox.mesh:238]: Available device [TV_CEILING] [31615] => rssi: -9999
[10:21:11][D][awox.mesh:263]: Currently 0 mesh devices reachable through active connections (5 currently known and 5 fully recognized)
[10:21:11][D][awox.mesh:133]: No devices found to connect to
[10:21:11][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -91, address_prefix mismatch
[10:21:11][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -97, address_prefix mismatch
[10:21:11][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -98, address_prefix mismatch
[10:21:12][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -100, address_prefix mismatch
[10:21:12][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -91, address_prefix mismatch
[... cut ...]
[10:22:45][D][sensor:094]: 'Awox BLE Mesh Hub Heap Max Block': Sending state 40960.00000 B with 0 decimals of accuracy
[10:22:45][D][sensor:094]: 'Awox BLE Mesh Hub Loop Time': Sending state 263.00000 ms with 0 decimals of accuracy
[10:22:46][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -94, address_prefix mismatch
[10:22:47][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -96, address_prefix mismatch
[10:22:47][D][awox.mesh:079]: Skipped device  - [redacted]. RSSI: -96, address_prefix mismatch
</code>
</details>

oxigen-dev, 2024-03-24:
> Maybe you should do it through the esphome gui?

I don't really see the added value :)

TotalPanther317, 2024-03-24:
Well, it seems as the yaml file is not configured correctly. Can you set this `address_prefix: A4:C1` in you .yaml file?

oxigen-dev, 2024-03-24:
> Well, it seems as the yaml file is not configured correctly. Can you set this `address_prefix: A4:C1` in you .yaml file?

This is what I was using: `address_prefix: A4:C1`

TotalPanther317, 2024-03-24:
This seems odd, maybe just remove the line completely?

oxigen-dev, 2024-03-24:
> This seems odd, maybe just remove the line completely?

Actually it was working fine for months with older versions of the Awox Hub and ESPHome this is why I think the problem is not the config file or the flashing process.

TotalPanther317, 2024-03-24:
Do you have another board on hand to flash this to?

oxigen-dev, 2024-03-24:
> Do you have another board on hand to flash this to?

Unfortunately not. I think that board should be OK, I've even completely erased it, then cleaned the build files, rebuild and reflashed it and nothing changed.

TotalPanther317, 2024-03-24:
Have you also uninstalled ALL esphome installations (also dev/beta), rebooted and reinstalled?

oxigen-dev, 2024-03-24:
> Have you also uninstalled ALL esphome installations (also dev/beta), rebooted and reinstalled?

I removed the esphome docker image before flashing so it was using the latest release version.

TotalPanther317, 2024-03-24:
I'm starting to run out of solutions... Only one last thing: Do you use the dev branch?

oxigen-dev, 2024-03-24:
> I'm starting to run out of solutions... Only one last thing: Do you use the dev branch?

Yes :(
```
external_components:
  source: github://fsaris/EspHome-AwoX-BLE-mesh-hub@dev
```

TotalPanther317, 2024-03-24:
Try this:
```
external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub
      ref: dev
```

oxigen-dev, 2024-03-24:
> Try this:
> 
> ```
> external_components:
>   - source:
>       type: git
>       url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub
>       ref: dev
> ```

I've checked now the official documentation at https://esphome.io/guides/configuration-types.html and
`# shorthand form github://username/repository/[folder/]file-path.yml[@branch-or-tag]` should be correct. I've already tried dev branches several times and it worked fine :(

fsaris, 2024-03-24:
You are using the dev branch else you wouldn't get the updated debug logging 🙂

It's probably related to the fact that your mesh devices are not always online. But not sure yet what changed in the latest esphome versions that causes this issue now. 

Does your other device have this issue too?

TotalPanther317, 2024-03-24:
I just remembered that especially this board loves to cut the wifi connection at some random interval. If you use this, try another one.
[image]

oxigen-dev, 2024-03-24:
> You are using the dev branch else you wouldn't get the updated debug logging 🙂
> 
> It's probably related to the fact that your mesh devices are not always online. But not sure yet what changed in the latest esphome versions that causes this issue now. 
> 
> Does your other device have this issue too? 

I have 3 hubs, 2 of them are controlling 2 (2x2) Ceilings, the problematic one is controlling now 5 in total. The other two are working fine. If I revert back to the old ESPHome version I'll loose the effect functionality :(

oxigen-dev, 2024-03-24:
> I just remembered that especially this board, in my experience, loves to cut the wifi connection at some random interval. If you use this, try another one.
> [image]
> 

I'm using a bigger one with external antenna.

fsaris, 2024-03-24:
> If I revert back to the old ESPHome version I'll loose the effect functionality :(

Removing this line should be enough to let it work with the older ESPHome version. https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/blob/948dcc0bb9b8e0f810d70cad32b96af32ad26267/components/awox_mesh/mesh_connection.cpp#L57

Curious if it doesn't give the same issues with that setup

oxigen-dev, 2024-03-24:
> Removing this line should be enough to let it work with the older ESPHome version. [...]

I've modified the code from the main branch and used an older ESPHome. It was just a short test, but the Hub was not even able to connect to the Ceiling. Now I'm back to an older ESPHome and older Awox Hub and it seems to work better. Ofcourse now I've lost some new functionality.

oxigen-dev, 2024-03-24:
> I've modified the code from the main branch and used an older ESPHome. It was just a short test, but the Hub was not even able to connect to the Ceiling. Now I'm back to an older ESPHome and older [...]

Now I have the same issue with the older versions.. RSSI is -9999 even if the Ceiling is on. I have really no idea what to do now :'(. I'll go back to the current dev branch and newest ESPHome.

oxigen-dev, 2024-03-24:
Just after my post it connected. This took some minutes...

fsaris, 2024-04-21:
@oxigen-dev still issues with connecting or can we close this issue?

oxigen-dev, 2024-04-21:
> @oxigen-dev still issues with connecting or can we close this issue?

@fsaris Now I'm on the dev branch and in the last few weeks it seemed to work well. As now I don't have problems, I'd propose to close this issue. Thanks!

closed 2024-04-21 by oxigen-dev