fsaris/EspHome-AwoX-BLE-mesh-hub #84: Groups cannot be controlled after some time

michalpulda, 2024-03-13:
Hi!

First, big thank you for implementing (above other things) support for groups. This feature has really changed how I perceive my Eglo lights (I actually don't want to get rid of them anymore). Sincere thanks for this!

Anyhow, after a few hours the lights won't respond to group commands anymore. Please find an excerpt from the log attached (`logger level: VERBOSE`). After restart it works again. Actually, I have set up an automation to restart the ESPHome devices every day at 5.30 AM. Some days I don't see the problem, so days I do.

I'm still on [this version](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/releases/tag/2024.2.1), so I will try to update and see if an update helps. I haven't seen anything that might help in the release notes.

If there is anything I can help with, please let me know.

Thanks a lot!

[20234-03-13_logs_esp32u-ble-awox_logs.txt](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/files/14593688/20234-03-13_logs_esp32u-ble-awox_logs.txt)

fsaris, 2024-03-13:
Sounds related to  #79, and that is fixed with the newer version.

If not than probably related to #83. That issue I'm still investigating

michalpulda, 2024-03-13:
Thanks, I have updated to the latest version. Will let you know if the issue persist.

michalpulda, 2024-03-18:
I can confirm that the issue still persists even with the last release. How can I help? Thanks!

fsaris, 2024-03-18:
Can you test the `dev` branch? I have logging set to verbose and for me it runs now more then a day. It should be reporting "Found device..." every few minutes 

To install dev use:

```
external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub
      ref: dev
```

fsaris, 2024-03-23:
duplicate of #83

closed 2024-03-23 by fsaris

michalpulda, 2024-03-30:
After few days of testing, I can confirm that it works well with the dev branch. Thanks!