fsaris/EspHome-AwoX-BLE-mesh-hub #86: Cannot update due to insufficient flash

danieldur, 2024-04-26:
First of all thank you for this great component!
I've been using it since a few year ago and it worked flawlessly. Now, after ignoring some esphome update notifications I decided that it is time to update my ESP32. But bad luck struck: I cannot update.

Here is my config:

```
esphome:
  name: "esp1"
  friendly_name: "esp1"

esp32:
  board: esp32dev
  framework:
    type: arduino
    version: 2.0.6 #https://github.com/espressif/arduino-esp32/releases

#status_led:
#  pin: GPIO2

ota:
  password: "***"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
   
external_components:
  source: github://fsaris/EspHome-AwoX-BLE-mesh-hub
  refresh: 1d

mqtt:
  broker: ***.duckdns.org
  username: mqtt_user
  password: XxX

#esp32_ble_tracker:
#  scan_parameters:
#    active: false

awox_mesh:
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password
  device_info:
    - product_id: 0x30
      device_type: RGB
      name: panel
      model: EPanel_120
      manufacturer: EGLO
      icon: mdi:wall-sconce-flat

#bluetooth_proxy:
#  active: true

# Enable logging
#logger:

#debug:
```

Notice the size reduction attempts in the commented out sections

And here is the update log:

```
INFO ESPHome 2024.4.1
INFO Reading configuration /config/esphome/esp1.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing esp1 (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- ArduinoJson @ 6.18.5
RAM:   [==        ]  19.8% (used 64976 bytes from 327680 bytes)
Flash: [==========]  103.3% (used 1895609 bytes from 1835008 bytes)
Error: The program size (1895609 bytes) is greater than maximum allowed (1835008 bytes)
*** [checkprogsize] Explicit exit, status 1
========================= [FAILED] Took 13.73 seconds =========================
```

Now the question is: did this component grow in size in the recent time?

fsaris, 2024-04-27:
With the Arduino framework you will keep running into this issue.

That's why we recommend to use esp-idf see
https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/discussions/71

fsaris, 2024-04-27:
Could be that the component grow a little over the last year but tried to reduce that again. But maybe esphome grow itself to

closed 2024-04-27 by fsaris