fsaris/EspHome-AwoX-BLE-mesh-hub #88: Long username crashes esphome device

radovanobal, 2024-05-26:
Using a longer username will cause esphome device to reboot at "D][awox.connection:251]: combine mesh name + password:" step. 

[23:06:31]abort() was called at PC 0x402128a3 on core 1
[23:06:31]
[23:06:31]
[23:06:31]Backtrace:0x400837c9:0x3ffd0f200x400943f5:0x3ffd0f40 0x4009986d:0x3ffd0f60 0x402128a3:0x3ffd0fe0 0x402128ea:0x3ffd1000 0x402129e3:0x3ffd1020 0x4022319b:0x3ffd1040 0x40213975:0x3ffd1060 0x40213cae:0x3ffd1080 0x40213d1d:0x3ffd10a0 0x400e5c8f:0x3ffd10c0 0x400e6164:0x3ffd1120 0x400e6dc7:0x3ffd11c0 0x400e76a3:0x3ffd12e0 0x40249c16:0x3ffd1360 0x400eb469:0x3ffd1380 0x40249bc1:0x3ffd13a0 0x400e9e8d:0x3ffd13c0 0x4024a145:0x3ffd13e0 0x4024a1a9:0x3ffd1400 0x400f3ad9:0x3ffd1420 0x400f5efa:0x3ffd1450 0x400fe6f5:0x3ffd1470 
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400837c9: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
WARNING Decoded 0x400943f5: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
WARNING Decoded 0x4009986d: abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
WARNING Decoded 0x402128a3: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/HOST-aarch64-linux-gnu/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:47
WARNING Decoded 0x402128ea: std::terminate() at /builds/idf/crosstool-NG/.build/HOST-aarch64-linux-gnu/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:57
WARNING Decoded 0x402129e3: __cxa_throw at /builds/idf/crosstool-NG/.build/HOST-aarch64-linux-gnu/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:95
WARNING Decoded 0x4022319b: std::__throw_length_error(char const*) at /builds/idf/crosstool-NG/.build/HOST-aarch64-linux-gnu/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/functexcept.cc:78 (discriminator 2)
WARNING Decoded 0x40213975: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_check_length(unsigned int, unsigned int, char const*) const at /builds/idf/crosstool-NG/.build/HOST-aarch64-linux-gnu/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/no-rtti/libstdc++-v3/include/bits/basic_string.h:312
WARNING Decoded 0x40213cae: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_replace_aux(unsigned int, unsigned int, unsigned int, char) at /builds/idf/crosstool-NG/.build/HOST-aarch64-linux-gnu/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/no-rtti/libstdc++-v3/include/bits/basic_string.tcc:396
WARNING Decoded 0x40213d1d: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::append(unsigned int, char) at /builds/idf/crosstool-NG/.build/HOST-aarch64-linux-gnu/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/no-rtti/libstdc++-v3/include/bits/basic_string.h:1281
WARNING Decoded 0x400e5c8f: esphome::awox_mesh::MeshConnection::combine_name_and_password[abi:cxx11]() const at /home/radovan/.homeassistant/esphome/.esphome/build/homeproxy/src/esphome/components/awox_mesh/mesh_connection.cpp:254
WARNING Decoded 0x400e6164: esphome::awox_mesh::MeshConnection::key_encrypt(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const at /home/radovan/.homeassistant/esphome/.esphome/build/homeproxy/src/esphome/components/awox_mesh/mesh_connection.cpp:271
WARNING Decoded 0x400e6dc7: esphome::awox_mesh::MeshConnection::setup_connection() at /home/radovan/.homeassistant/esphome/.esphome/build/homeproxy/src/esphome/components/awox_mesh/mesh_connection.cpp:222
WARNING Decoded 0x400e76a3: esphome::awox_mesh::MeshConnection::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /home/radovan/.homeassistant/esphome/.esphome/build/homeproxy/src/esphome/components/awox_mesh/mesh_connection.cpp:148
WARNING Decoded 0x40249c16: esphome::esp32_ble_tracker::ESP32BLETracker::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /home/radovan/.homeassistant/esphome/.esphome/build/homeproxy/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:404 (discriminator 2)
WARNING Decoded 0x400eb469: non-virtual thunk to esphome::esp32_ble_tracker::ESP32BLETracker::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*)
WARNING Decoded 0x40249bc1: esphome::esp32_ble::ESP32BLE::real_gattc_event_handler_(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /home/radovan/.homeassistant/esphome/.esphome/build/homeproxy/src/esphome/components/esp32_ble/ble.cpp:344 (discriminator 2)
WARNING Decoded 0x400e9e8d: esphome::esp32_ble::ESP32BLE::loop() at /home/radovan/.homeassistant/esphome/.esphome/build/homeproxy/src/esphome/components/esp32_ble/ble.cpp:294
WARNING Decoded 0x4024a145: esphome::Component::call_loop() at /home/radovan/.homeassistant/esphome/.esphome/build/homeproxy/src/esphome/core/component.cpp:77
WARNING Decoded 0x4024a1a9: esphome::Component::call() at /home/radovan/.homeassistant/esphome/.esphome/build/homeproxy/src/esphome/core/component.cpp:104
WARNING Decoded 0x400f3ad9: esphome::Application::loop() at /home/radovan/.homeassistant/esphome/.esphome/build/homeproxy/src/esphome/core/application.cpp:74 (discriminator 2)
WARNING Decoded 0x400f5efa: loop() at /home/radovan/.homeassistant/esphome/.esphome/build/homeproxy/src/main.cpp:320
WARNING Decoded 0x400fe6f5: loopTask(void*) at /home/radovan/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
[23:06:37]
[23:06:37]
[23:06:37]
[23:06:37]
[23:06:37]ELF file SHA256: 0000000000000000
[23:06:37]
[23:06:37]Rebooting...
[23:06:37]ets Jul 29 2019 12:21:46
[23:06:37]
[23:06:37]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[23:06:37]configsip: 0, SPIWP:0xee
[23:06:37]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[23:06:37]mode:DIO, clock div:2
[23:06:37]load:0x3fff0030,len:1184
[23:06:37]load:0x40078000,len:13132
[23:06:37]load:0x40080400,len:3036
[23:06:37]entry 0x400805e4

fsaris, 2024-06-06:
Are you sure you used the correct mesh username? It should be something of 8 characters

fsaris, 2024-09-01:
_note to self, add a max length validation to the yaml config for the mesh credentials_

radovanobal, 2024-09-01:
Sorry, forgot to respond. I used the same username I have in my home assistant integration for awox mesh. The username I used was an email.

fsaris, 2024-09-01:
See the readme, you need to fetch the mesh credentials for instance using this form https://fsaris.github.io/EspHome-AwoX-BLE-mesh-hub/awoxh-mesh-credentials-tool/

You use your awox account to fetch the mesh credentials

radovanobal, 2024-09-13:
it looks like I need to work on my English reading compilation. I apologise for the misunderstanding, it works fine when getting the proper credentials.

closed 2024-10-19 by fsaris