fsaris/EspHome-AwoX-BLE-mesh-hub #89: Disconnect with error ESP_GATTC_DISCONNECT_EVT

Stefan3005, 2024-06-03:
Hello,

I have a problem with the application.
It finds my two Awox Light Panels, but the connection failed with: ESP_GATTC_DISCONNECT_EVT.

Months ago, I got the component runnig. Now that I have a new HA Installation on another device, it don't work anymore.

Here are my logs:
[logs_awox-and-pager_run.txt](https://github.com/user-attachments/files/15534279/logs_awox-and-pager_run.txt)

And this is my yaml:
[awox-and-pager.txt](https://github.com/user-attachments/files/15534351/awox-and-pager.txt)

Awox-BLE version: 2024.3.1
ESPHome version: 2024.5.4
HA Core version: 2024.5.5

Thanks for your help and this awesome component.

Greetings
Stefan

fsaris, 2024-06-06:
Not sure, looks that the Bluetooth layer isn't able to setup a connection.

Did you have it running on this device with the previous version?

What device are you using?

Stefan3005, 2024-06-24:
Sorry for my delayed response, I was on vacation.
I use an esp-wroom-32. 
I think that is the reason, because I tried another ESP and with that one it worked.

fsaris, 2024-09-01:
Okay, so probably related to the build quality of the esp32 board

closed 2024-09-01 by fsaris