fsaris/EspHome-AwoX-BLE-mesh-hub #91: ESPHome AwoX entity greyed out

danieldur, 2024-07-02:
Hello there Frans,
Sorry to bother but I am loosing my mind here trying to figure things out. 
I had been using the ESPhome integration for quite some time and constantly delayed updates. Then, one day decided to jump the gun and update... but since then the integration doesn't work. So I started investigating, reconfiguring... nothing...

Here is my configuration: 

`esphome:
  name: esp1
  friendly_name: esp1

esp32:
  board: esp32dev
  framework:
    type: esp-idf

status_led:
  pin: GPIO2

# Enable logging
logger:

# Enable Home Assistant API
# api:

# Allow Over-The-Air updates
ota:
  - platform: esphome
    password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub
      ref: main

#esp32_ble_tracker:
#  scan_parameters:
#    active: false

awox_mesh:
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password
  device_info:
  # Example device type, see device info in HomeAssistant or MQTT message to find the 'product_id'
    - product_id: 0x30
      device_type: RGB
      name: panel
      model: EPanel_120
      manufacturer: EGLO

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_user
  password: !secret mqtt_password
`

And here are the logs after installing it on my ESP (this exact model: https://www.berrybase.de/esp32-nodemcu-development-board): https://pastebin.com/Udhq4N3q

The configuration is brand new as at one point, while investigating, I managed to delete the old config.

I deleted the previous devices inside the MQTT integration and, after a reinstall, the panel is visible there (Connected via [Unnamed device]).

Any ideas what I might have done wrong?

Thank you!

fsaris, 2024-07-27:
Are you sure this is your current yaml? Because I think you need the `esp32_ble_tracker` to be there to let it build.

Since earlier this year there is a min RSSI config 

```
min_rssi: -90 # to prevent that the mesh tries to connect to devices that are out of range
```

Recording to your logs your devices report a RSSI around -90 so maybe they are out of range? 

You could try with a lower `min_rssi` value

danieldur, 2024-08-29:
Hello Frans,
Sorry for my late reply. Your suggestion worked, but only after moving the ESP board closer to the light panel.
Something must have changed either in the ESP, your add-on or something in my home that I didn't notice. Because the board used to work OK with the panel from about 2 meters and one wall away, Now I had to move the board in the same room with the light, about 2 meters away and raise the sensibility to maximum (-100 dB).
Thank you very much for the add-on and the support!

closed 2024-09-01 by fsaris