fsaris/EspHome-AwoX-BLE-mesh-hub #92: One lamp does not properly get set up

chefexperte, 2024-07-05:
I have three EFueva_300s lamps. I have had one for years, that one works perfectly fine. 
Then I bought an old one, that one also started working after connecting it. 
Now I have bought another new lamp of the series, but this one does not properly get recognised. 
It finds three devices, but says that only two are currently known and fully recognised. 

Logs:

```
[03:15:20][D][awox.mesh:230]: Available device A4:C1:38:A3:2D:72 [11634] => rssi: -90
[03:15:20][D][awox.mesh:230]: Available device A4:C1:38:BD:CA:93 [0] => rssi: -9999
[03:15:20][D][awox.mesh:230]: Available device A4:C1:38:81:0B:5F [2911] => rssi: -9999
[03:15:20][D][awox.mesh:255]: Currently 1 mesh devices reachable through active connections (2 currently known and 2 fully recognized)
[03:15:20][D][awox.mesh:129]: No devices found to connect to
[03:15:21][D][awox.mesh:088]: Found Awox device  - A4:C1:38:A3:2D:72. RSSI: -90 dB (total devices: 3)
[03:15:24][D][awox.mesh:088]: Found Awox device  - A4:C1:38:A3:2D:72. RSSI: -89 dB (total devices: 3)
[03:15:25][D][awox.mesh:088]: Found Awox device  - A4:C1:38:A3:2D:72. RSSI: -89 dB (total devices: 3)
```

My script:

```
esphome:
  name: esp32-board
  friendly_name: ESP32 board

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    

# Enable logging
logger:

mqtt:
  broker: 192.168.0.XXX
  username: mqtt
  password: !secret mqtt_password

ota:
  - platform: esphome
    id: ota_pass
    password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Board Fallback Hotspot"
    password: "xxx"

external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub
      ref: main

esp32_ble_tracker:
  scan_parameters:
    active: false

awox_mesh:
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password

  device_info:
    - product_id: 0x42
      device_type: RGB
      name: EGLO Surface square
      model: EFueva_300s
      manufacturer: EGLO
      icon: mdi:lightbulb
    - product_id: 0x42
      device_type: RGB
      name: EGLO Surface square
      model: EFueva_300s
      manufacturer: EGLO
      icon: mdi:lightbulb
    - product_id: 0x42
      device_type: RGB
      name: EGLO Surface square
      model: EFueva_300s
      manufacturer: EGLO
      icon: mdi:lightbulb
```

Entities from MQTT: 
[image]

I have tried deleting MQTT+Moquitto broker and reinstalling them, but that did not help. I reflashed the esp board multiple times, that did not help either. I updated ESPHome and deleted the .esphome build folder as well and tried again. Didn't help either. 
I don't really know what to do, I don't get what the problem is.

chefexperte, 2024-07-05:
Okay reading through the log I also found this: 

```
[03:28:41][D][awox.mesh:267]: Try to connecty to device A4:C1:38:XX:XX:5F [2911] no active connection found for this device
[03:28:41][I][awox.mesh:139]: Try to connect A4:C1:38:XX:XX:5F => rssi: -71
[03:28:41][I][awox.connection:117]: IDLE
[03:28:41][D][esp32_ble_client:110]: [0] [A4:C1:38:XX:XX:5F] Found device
[03:28:41][D][esp32_ble_tracker:669]: Found device A4:C1:38:XX:XX:5F RSSI=-67
[03:28:41][D][esp32_ble_tracker:690]:   Address Type: PUBLIC
[03:28:41][D][esp32_ble_tracker:692]:   Name: 'XXXXXXXX'
[03:28:41][I][awox.connection:125]: DISCOVERED
[03:28:41][W][component:237]: Component awox_mesh took a long time for an operation (65 ms).
[03:28:41][W][component:238]: Components should block for at most 30 ms.
[03:28:41][D][esp32_ble_tracker:219]: Pausing scan to make connection...
[03:28:41][I][awox.connection:129]: READY_TO_CONNECT
[03:28:41][I][esp32_ble_client:067]: [0] [A4:C1:38:XX:XX:5F] 0x00 Attempting BLE connection
[03:28:41][I][awox.connection:133]: CONNECTING
[03:28:42][D][esp32_ble_client:110]: [0] [A4:C1:38:XX:XX:5F] ESP_GATTC_CONNECT_EVT
[03:28:42][D][esp32_ble_client:110]: [0] [A4:C1:38:XX:XX:5F] ESP_GATTC_OPEN_EVT
[03:28:42][I][awox.connection:137]: CONNECTED
[03:28:42][D][esp32_ble_tracker:270]: Starting scan...
[03:28:42][D][awox.mesh:117]: scan end
[03:28:42][D][awox.mesh:088]: Found Awox device XXXXXXXX - A4:C1:38:XX:XX:93. RSSI: -95 dB (total devices: 3)
[03:28:43][D][esp32_ble_client:306]: [0] [A4:C1:38:XX:XX:5F] Event 46
[03:28:43][D][esp32_ble_client:110]: [0] [A4:C1:38:XX:XX:5F] ESP_GATTC_SEARCH_CMPL_EVT
[03:28:43][I][esp32_ble_client:227]: [0] [A4:C1:38:XX:XX:5F] Connected
[03:28:43][I][awox.connection:147]: Connected....
[03:28:43][D][awox.connection:251]: combine mesh name + password: XXXXXXXX:XXXXXXXX
[03:28:43][D][awox.connection:234]: Listen for notifications
[03:28:43][D][awox.connection:241]: Enable notifications
[03:28:43][I][awox.mesh.mqtt:082]: Publish connection info, 2 connections, 2 device online
[03:28:43][D][esp32_ble_client:188]: [0] [A4:C1:38:XX:XX:5F] cfg_mtu status 0, mtu 23
[03:28:43][D][esp32_ble_client:110]: [0] [A4:C1:38:XX:XX:5F] ESP_GATTC_REG_FOR_NOTIFY_EVT
[03:28:43][W][esp32_ble_client:275]: [0] [A4:C1:38:XX:XX:5F] esp_ble_gattc_get_descr_by_char_handle error, status=10
[03:28:43][D][esp32_ble_client:110]: [0] [A4:C1:38:XX:XX:5F] ESP_GATTC_WRITE_CHAR_EVT
[03:28:43][W][component:237]: Component esp32_ble took a long time for an operation (54 ms).
[03:28:43][W][component:238]: Components should block for at most 30 ms.
[03:28:43][D][esp32_ble_client:110]: [0] [A4:C1:38:XX:XX:5F] ESP_GATTC_READ_CHAR_EVT
[03:28:43][I][awox.connection:179]: Response OK, let's go
[03:28:43][D][awox.connection:251]: combine mesh name + password: XXXXXXXX:XXXXXXXX
[03:28:43][I][awox.connection:184]: [0] [A4:C1:38:XX:XX:5F] session key 1100001001100100000000000100000010010111101111100111100011000100001001001100111011001110010011101000111101011010000001XXXXXXXXXX
[03:28:43][D][awox.connection:551]: [0] [A4:C1:38:XX:XX:5F] request status update
[03:28:43][D][awox.connection:541]: [0] [A4:C1:38:XX:XX:5F] [65535] write_command packet DA => 00010000
[03:28:43][D][esp32_ble_client:110]: [0] [A4:C1:38:XX:XX:5F] ESP_GATTC_WRITE_CHAR_EVT
[03:28:43][D][esp32_ble_client:110]: [0] [A4:C1:38:XX:XX:5F] ESP_GATTC_NOTIFY_EVT
[03:28:43][D][awox.connection:355]: online status report: mesh: 2911, on: 1, color_mode: 0, sequence_mode: 0, candle_mode: 0, w_b: 67, temp: 127, c_b: 53, rgb: FF0000, mode: 1 00000001
[03:28:43][I][awox.connection:458]: 2911: ON temp: 127 (67 %) ONLINE
[03:28:43][D][awox.mesh:407]: Delayed publish online/offline for 2911 - online
[03:28:43][D][esp32_ble_client:110]: [0] [A4:C1:38:XX:XX:5F] ESP_GATTC_WRITE_CHAR_EVT
[03:28:43][W][component:237]: Component esp32_ble took a long time for an operation (95 ms).
[03:28:43][W][component:238]: Components should block for at most 30 ms.
[03:28:43][D][esp32_ble_client:110]: [0] [A4:C1:38:XX:XX:5F] ESP_GATTC_NOTIFY_EVT
[03:28:43][D][awox.connection:379]: status report: mesh: 2911, on: 1, color_mode: 0, sequence_mode: 0, candle_mode: 0, w_b: 67, temp: 127, c_b: 53, rgb: FF0000, mode: 1 00000001
[03:28:43][I][awox.connection:458]: 2911: ON temp: 127 (67 %) ONLINE
[03:28:43][D][awox.mesh:088]: Found Awox device  - A4:C1:38:XX:XX:5F. RSSI: -68 dB (total devices: 3)
[03:28:44][D][awox.mesh:088]: Found Awox device XXXXXXXX - A4:C1:38:XX:XX:93. RSSI: -95 dB (total devices: 3)
[03:28:44][D][awox.mesh:088]: Found Awox device  - A4:C1:38:XX:XX:5F. RSSI: -69 dB (total devices: 3)
[03:28:44][D][awox.mesh:088]: Found Awox device  - A4:C1:38:XX:XX:72. RSSI: -88 dB (total devices: 3)
[03:28:45][D][awox.mesh:088]: Found Awox device XXXXXXXX - A4:C1:38:XX:XX:93. RSSI: -86 dB (total devices: 3)
[03:28:45][D][awox.mesh:088]: Found Awox device  - A4:C1:38:XX:XX:5F. RSSI: -69 dB (total devices: 3)
[03:28:45][D][awox.mesh:088]: Found Awox device XXXXXXXX - A4:C1:38:XX:XX:93. RSSI: -85 dB (total devices: 3)
[03:28:46][D][awox.mesh:088]: Found Awox device XXXXXXXX - A4:C1:38:XX:XX:93. RSSI: -94 dB (total devices: 3)
[03:28:46][I][awox.mesh.mqtt:117]: Publish online/offline for 2911 - online
[03:28:46][D][awox.mesh:088]: Found Awox device XXXXXXXX - A4:C1:38:XX:XX:93. RSSI: -85 dB (total devices: 3)
[03:28:47][D][awox.mesh:088]: Found Awox device XXXXXXXX - A4:C1:38:XX:XX:93. RSSI: -94 dB (total devices: 3)
[... cut ...]
[03:28:49][D][awox.mesh:088]: Found Awox device  - A4:C1:38:XX:XX:93. RSSI: -85 dB (total devices: 3)
[03:28:49][D][awox.mesh:088]: Found Awox device XXXXXXXX - A4:C1:38:XX:XX:93. RSSI: -87 dB (total devices: 3)
[03:28:50][D][awox.mesh:088]: Found Awox device  - A4:C1:38:XX:XX:72. RSSI: -86 dB (total devices: 3)
```

Where `esp_ble_gattc_get_descr_by_char_handle error, status=10` might(?) be related to the issue. Besides that, it says 2 devices online and 2 connections, but 3 total devices. Also, why 2 connections?

The old lamp has HW1.0 and the others HW4.0, Firmware is all up to date.

fsaris, 2024-07-12:
Are the lights in reach of each other? Else the mesh will not be able to work.

The component will setup 1 connection and then checks each device that is part of it's mesh (and is in reach). When it sees/scans a Bluetooth address that isn't part of that mesh it sets up a second connection to also be to control that device and the devices in reach of that device.

That is the reason for the 2 connections (you could bump it to max 3 connections when needed, see the config options).

The error is probably caused by a weak connection

chefexperte, 2024-07-12:
The light that does not get recognised has a distance of around 5-8 Meters with no wall in between. So I think they should be in reach of each other. Is there a definite way I can check if that is or is not the problem? 
I will also check the config options and see if that helps.

chefexperte, 2024-07-25:
Okay so 

> you could bump it to max 3 connections when needed, see the config options

did help, and I also tried positioning my ESP board better. But the bluetooth connection IS really weak. The lamps are only a few meters apart with no walls or anything obstructing in the way, but that seems to be too much. :(
Maybe the EFueva_300s has a weak bluetooth chip? I'd expect my ESP board to have bad bluetooth, it was cheap. but the lamps?

fsaris, 2024-07-27:
If you test it using the official app and position yourself in a position that the app is only in reach of 1 device are you able to control the other light's?

uwecash, 2024-08-11:
I have a very similar problem. My two lamps are about 3 meters away from the ESP. One of the two keeps losing connection to the ESP. But it was the same with the smartphone app.
My question would therefore be whether it would be possible to use a second Bluetooth proxy, as a repeater so to speak?
What should the configuration look like?

fsaris, 2024-09-01:
Yes you can use a second module. That's why you can limit the Mac addresses and mesh IDs to let a specific device/light only talk to a specific esp32 module

fsaris, 2024-09-01:
See the readme and example config for the config options 
https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub?tab=readme-ov-file#allowed_mac_addresses-list-of-mac-addresses---optional

closed 2024-09-01 by fsaris