fsaris/EspHome-AwoX-BLE-mesh-hub #93: Motion sensor

mrsofa, 2024-07-21:
Hello,

I’ve successfully connected an Eglo smart bulb with my ESP32. I also have a couple of Bluetooth motion sensors, but they don’t seem to display in the MQTT integration as the bulb does.

I noticed that these sensors are not listed in the configuration examples. I was wondering if it would be possible to add such a device.

The model is:

Art No: 97475
Type: Bluetooth PIR Sensor

fsaris, 2024-07-21:
Hi, I expect that we should be able to get them to work. But currently they are not supported and as I don't have such a device I'm not able to test them.

Do you see the devices (mac addresses) in the logs?

fsaris, 2024-07-21:
Are the sensors ble mesh devices? Else it will not work

mrsofa, 2024-07-27:
Hey, sorry for the late reply, i am having dificulties with the esp32 itself, it keeps rebooting on me..
The sensor is as i understand a ble mesh device yes, there are 3 mac adresses in the logs and under the inegration i have one bulb, an unknow device and a group - that is probably the sensor and bulb as they are linked in the awox app.

I can paste the logs here because they don't tell me alot what's going on.
[logs_awox-ble-mesh-hub_logs.txt](https://github.com/user-attachments/files/16399122/logs_awox-ble-mesh-hub_logs.txt)

fsaris, 2024-09-01:
Looks that the Bluetooth stack crashes. Can you share your yaml config?

mrsofa, 2024-09-01:
Here you go
[config.txt](https://github.com/user-attachments/files/16829297/config.txt)

fsaris, 2024-09-01:
Config looks fine, looks the esp32 board is giving you issues. Maybe you can test with another one? There are more reports of some boards crashing on the ble stack an others just running fine with the same config

mrsofa, 2024-09-02:
Oh wow, that worked yes, looks like Aliexpress esp boards are a little weird after all.

Also about the Motion sensor, it's probably a BT one as I can't see the mac in the logs. I can see a group in the mqtt, because I have the lamp an sensor grouped, and they still work, but as I said, no found device what so ever. You can check the logs but i think this one wont work.
[logs_awox-ble-mesh-hub-2_logs.txt](https://github.com/user-attachments/files/16840926/logs_awox-ble-mesh-hub-2_logs.txt)

And thanks again for the easy fix for the bluetooth crashing

fsaris, 2024-09-29:
It looks that the device with address `A4:C1:38:92:0C:D2` is responding with data the code doesn't expect. Maybe use the `allowed_mac_addresses` config to only allow mac-addresses you know work

fsaris, 2024-10-19:
@mrsofa the crash can also be due to some unhandled edge case that's was already fixed in the `dev` branch and not is also part of the `main` branch.

Closing the issue for now, can always be re-opened when more info is provided

closed 2024-10-19 by fsaris