fsaris/EspHome-AwoX-BLE-mesh-hub #94: Device authentication error: known mesh credentials are not excepted by the device

RobinWeitzel, 2024-08-03:
Hi,

I tried installing this component both on an ESP32 WROOM and WROVER (just to make sure it is not the device) but in both cases the ESP find my lamp but fails to connect to it.
I took the credentials from the form (using the ones with "mesh") and I tried removing and adding the lamp to my account multiple times as well as deleting my account, creating a new one and then using those new credentials.

However, it still fails to authenticate.
Are some lamps not supported?
This is the lamp I am using (link is in german): https://www.eglo.com/at/deckenleuchte-turcona-c-32355.html

Here is the full error I get:
> [04:33:52][D][awox.mesh:088]: Found Awox device oujqtErw - A4:C1:38:97:5E:D4. RSSI: -83 dB (total devices: 1)
[04:33:53][D][awox.mesh:088]: Found Awox device oujqtErw - A4:C1:38:97:5E:D4. RSSI: -83 dB (total devices: 1)
[04:33:54][D][awox.mesh:088]: Found Awox device oujqtErw - A4:C1:38:97:5E:D4. RSSI: -82 dB (total devices: 1)
[04:33:55][D][awox.mesh:088]: Found Awox device oujqtErw - A4:C1:38:97:5E:D4. RSSI: -82 dB (total devices: 1)
[04:33:56][D][awox.mesh:088]: Found Awox device oujqtErw - A4:C1:38:97:5E:D4. RSSI: -80 dB (total devices: 1)
[04:33:56][D][awox.mesh:227]: Total devices: 1
[04:33:56][D][awox.mesh:230]: Available device A4:C1:38:97:5E:D4 [0] => rssi: -80
[04:33:56][D][awox.mesh:255]: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
[04:33:56][D][awox.mesh:261]: Try to connecty to device A4:C1:38:97:5E:D4 no mesh id known yet
[04:33:56][I][awox.mesh:139]: Try to connect A4:C1:38:97:5E:D4 => rssi: -80
[04:33:56][I][awox.connection:117]: IDLE
[04:33:56][D][esp32_ble_client:110]: [0] [A4:C1:38:97:5E:D4] Found device
[04:33:56][D][esp32_ble_tracker:669]: Found device A4:C1:38:97:5E:D4 RSSI=-84
[04:33:56][D][esp32_ble_tracker:690]:   Address Type: PUBLIC
[04:33:56][D][esp32_ble_tracker:692]:   Name: 'oujqtErw'
[04:33:56][I][awox.connection:125]: DISCOVERED
[04:33:56][D][esp32_ble_tracker:219]: Pausing scan to make connection...
[04:33:56][I][awox.connection:129]: READY_TO_CONNECT
[04:33:56][I][esp32_ble_client:067]: [0] [A4:C1:38:97:5E:D4] 0x00 Attempting BLE connection
[04:33:56][I][awox.connection:133]: CONNECTING
[04:33:58][D][esp32_ble_client:110]: [0] [A4:C1:38:97:5E:D4] ESP_GATTC_CONNECT_EVT
[04:33:58][D][esp32_ble_client:110]: [0] [A4:C1:38:97:5E:D4] ESP_GATTC_OPEN_EVT
[04:33:58][I][awox.connection:137]: CONNECTED
[04:33:58][D][esp32_ble_tracker:270]: Starting scan...
[04:33:58][D][awox.mesh:117]: scan end
[04:34:00][D][esp32_ble_client:306]: [0] [A4:C1:38:97:5E:D4] Event 46
[04:34:00][D][esp32_ble_client:110]: [0] [A4:C1:38:97:5E:D4] ESP_GATTC_SEARCH_CMPL_EVT
[04:34:00][I][esp32_ble_client:227]: [0] [A4:C1:38:97:5E:D4] Connected
[04:34:00][I][awox.connection:147]: Connected....
[04:34:00][D][awox.connection:251]: combine mesh name + password: oujqtErw:8dd0cbfe
[04:34:00][D][awox.connection:234]: Listen for notifications
[04:34:00][D][awox.connection:241]: Enable notifications
[04:34:00][I][awox.mesh.mqtt:072]: Publish mesh connection status: online
[04:34:00][I][awox.mesh.mqtt:082]: Publish connection info, 1 connections, 0 device online
[04:34:00][D][esp32_ble_client:188]: [0] [A4:C1:38:97:5E:D4] cfg_mtu status 0, mtu 247
[04:34:00][D][esp32_ble_client:110]: [0] [A4:C1:38:97:5E:D4] ESP_GATTC_REG_FOR_NOTIFY_EVT
[04:34:00][W][esp32_ble_client:275]: [0] [A4:C1:38:97:5E:D4] esp_ble_gattc_get_descr_by_char_handle error, status=10
[04:34:00][D][esp32_ble_client:110]: [0] [A4:C1:38:97:5E:D4] ESP_GATTC_WRITE_CHAR_EVT
[04:34:00][W][component:237]: Component esp32_ble took a long time for an operation (54 ms).
[04:34:00][W][component:238]: Components should block for at most 30 ms.
[04:34:00][D][esp32_ble_client:110]: [0] [A4:C1:38:97:5E:D4] ESP_GATTC_READ_CHAR_EVT
[04:34:00][E][awox.connection:191]: Device authentication error: known mesh credentials are not excepted by the device. Did you re-pair them to your Awox app with a different account?
[04:34:00][I][awox.connection:197]: [0] [A4:C1:38:97:5E:D4] response 00001110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[04:34:00][I][esp32_ble_client:084]: [0] [A4:C1:38:97:5E:D4] Disconnecting.
[04:34:00][I][awox.connection:113]: DISCONNECTING
[04:34:00][I][awox.mesh:043]: disconnected
[04:34:00][I][awox.mesh.mqtt:072]: Publish mesh connection status: offline
[04:34:00][I][awox.mesh.mqtt:082]: Publish connection info, 0 connections, 0 device online
[04:34:00][D][esp32_ble_client:110]: [0] [] ESP_GATTC_WRITE_CHAR_EVT
[04:34:00][D][esp32_ble_client:110]: [0] [] ESP_GATTC_CLOSE_EVT
[04:34:00][I][awox.connection:117]: IDLE
[04:34:00][W][component:237]: Component esp32_ble took a long time for an operation (72 ms).
[04:34:00][W][component:238]: Components should block for at most 30 ms.
[04:34:01][D][awox.mesh:227]: Total devices: 1
[04:34:01][D][awox.mesh:230]: Available device A4:C1:38:97:5E:D4 [0] => rssi: -80
[04:34:01][D][awox.mesh:255]: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
[04:34:01][D][awox.mesh:261]: Try to connecty to device A4:C1:38:97:5E:D4 no mesh id known yet
[04:34:01][I][awox.mesh:139]: Try to connect A4:C1:38:97:5E:D4 => rssi: -80
[04:34:01][I][awox.connection:117]: IDLE
[04:34:01][D][esp32_ble_client:110]: [0] [A4:C1:38:97:5E:D4] Found device
[... 64 lines cut ...]
[04:34:09][D][esp32_ble_client:306]: [0] [A4:C1:38:97:5E:D4] Event 46
[04:34:09][D][esp32_ble_client:110]: [0] [A4:C1:38:97:5E:D4] ESP_GATTC_SEARCH_CMPL_EVT
[04:34:09][I][esp32_ble_client:227]: [0] [A4:C1:38:97:5E:D4] Connected
[04:34:09][I][awox.connection:147]: Connected....
[04:34:09][D][awox.connection:251]: combine mesh name + password: oujqtErw:8dd0cbfe
[04:34:09][D][awox.connection:234]: Listen for notifications
[04:34:09][D][awox.connection:241]: Enable notifications
[04:34:09][I][awox.mesh.mqtt:072]: Publish mesh connection status: online
[04:34:09][I][awox.mesh.mqtt:082]: Publish connection info, 1 connections, 0 device online
[04:34:09][D][esp32_ble_client:188]: [0] [A4:C1:38:97:5E:D4] cfg_mtu status 0, mtu 247
[04:34:09][D][esp32_ble_client:110]: [0] [A4:C1:38:97:5E:D4] ESP_GATTC_REG_FOR_NOTIFY_EVT
[04:34:09][W][esp32_ble_client:275]: [0] [A4:C1:38:97:5E:D4] esp_ble_gattc_get_descr_by_char_handle error, status=10
[04:34:09][D][esp32_ble_client:110]: [0] [A4:C1:38:97:5E:D4] ESP_GATTC_WRITE_CHAR_EVT
[04:34:09][W][component:237]: Component esp32_ble took a long time for an operation (53 ms).
[04:34:09][W][component:238]: Components should block for at most 30 ms.
[04:34:09][D][esp32_ble_client:110]: [0] [A4:C1:38:97:5E:D4] ESP_GATTC_READ_CHAR_EVT
[04:34:09][E][awox.connection:191]: Device authentication error: known mesh credentials are not excepted by the device. Did you re-pair them to your Awox app with a different account?
[04:34:09][I][awox.connection:197]: [0] [A4:C1:38:97:5E:D4] response 00001110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[04:34:09][I][esp32_ble_client:084]: [0] [A4:C1:38:97:5E:D4] Disconnecting.
[04:34:09][I][awox.connection:113]: DISCONNECTING
[04:34:09][I][awox.mesh:043]: disconnected
[04:34:09][I][awox.mesh.mqtt:072]: Publish mesh connection status: offline
[04:34:09][I][awox.mesh.mqtt:082]: Publish connection info, 0 connections, 0 device online
[04:34:09][D][esp32_ble_client:110]: [0] [] ESP_GATTC_WRITE_CHAR_EVT
[04:34:09][D][esp32_ble_client:110]: [0] []

fsaris, 2024-08-04:
The Turcona-C doesn't seem to be a BLE-mesh device. Only the ble mesh devices are supported by this component

RobinWeitzel, 2024-08-05:
Ah, I see. That makes sense. It is Zigbee enabled so I will try with a ZigBee bridge.

Thanks
Robin

closed 2024-08-05 by RobinWeitzel