fsaris/EspHome-AwoX-BLE-mesh-hub #101: EPanel_120 not detected

benedikt-wueller, 2024-09-19:
Hi, I tried using both the legacy integration (https://github.com/fsaris/home-assistant-awox) using an ESP32 bluetooth proxy as well as this one installed on an M5Stack Atom Lite. However, the legacy integration was unable to connect to the panel and using this module no discovery messages are sent over MQTT even when the M5 is placed directly next to the `EPanel_120` (Version 2.2.6 - HW 1.0).

I just noticed the legacy integration mentions support for HW 4.0 only
> EPanel_120 (HW v4.0)

Am I correct to assume HW 1.0 is currently not supported by either integration or is there something wrong with my setup?
If it is currently unsupported, are there any plans to support it in the future?

fsaris, 2024-09-19:
For the legacy integration I was under the assumption that the hardware version was related but currently I'm not sure if it's related.

Are you sure you have the ble mesh EPanel_120? Isn't it the older Bluetooth version?

This component only works for the ble (Bluetooth) mesh devices

benedikt-wueller, 2024-09-19:
Honestly I'm not sure it's the mesh version but that might be why I'm facing issues. I'll try to investigate this further but worst case I'll have to continue to use the awox app

fsaris, 2024-09-19:
See #97

closed 2024-09-19 by fsaris

benedikt-wueller, 2024-10-06:
Just in case someone stumbles upon the same issue in the future, I finally figured it out. I am in fact using the ble mesh panel. The issue was related to the MQTT broker I was using. Even though the logs reported a successful connection to the MQTT broker, neither the broker nor the ESPHome logs ever received any debug logs from the esp32 board.

Once I switched from the HA `Mosquitto broker` add-on to a self-hosted mosquitto broker using docker, the module immediately started working and my panel got discovered.

fsaris, 2024-10-13:
> Once I switched from the HA Mosquitto broker add-on to a self-hosted mosquitto broker using docker, the module immediately started working, logs started to pour in and my panel got discovered successfully.

Looks that you had some retained messages in the mosquitto db that gave you some trouble. Probably clearing the db would be enough instead of switching to a different server