fsaris/EspHome-AwoX-BLE-mesh-hub #103: Long delay during operation

blueJack92, 2024-10-09:
There is an enormous delay of up to several minutes when operating the light functions (on/off, dimming) in Home Assistant.

What can be done to make this happen much faster?

Here is my yaml file:
`substitutions:
  name: "awox-ble-mesh-hub"
  friendly_name: "AwoX BLE mesh hub"

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_user
  password: !secret mqtt_password

# Enable Home Assistant API
#api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub
      ref: main
  #  ref: dev
  # - source:
  #     type: local
  #     path: components

esp32_ble_tracker:
  scan_parameters:
    active: false

awox_mesh:
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password

  # Optional options

  # address_prefix: A4:C1
  # min_rssi: -90 # to prevent that the mesh tries to connect to devices that are out of range
  # max_connections: 2 # max 3 connections

  # You can give a list of mesh_ids that only should be handled by this device
  # Example when you want to setup multiple ESP modules to only control a part of the mesh) be sure to also set allowed_mac_addresses option.

  # allowed_mesh_ids:
  #   - 12345
  #   - 23456
  #   - 6758

  # You can give a list of mac_addresses that are allowed to connect to (to exclude some devices that are often offline or for the allowed_mesh_ids option)

  # allowed_mac_addresses:
  #   - A4:C1:38:8A:BA:11
  #   - A4:C1:38:65:AD:22

  device_info:
    - product_id: 0x96
      device_type: RGB
      manufacturer: EGLO
      name: Marghera-C Pendelleuchte
      model: EGLO-RGB-TW
      icon: mdi:wall-sconce-flat
`

blueJack92, 2024-10-12:
Ok, I think the problem was caused by the Eglo gateways. Since I have removed these, the control system works very quickly to immediately for the most part.

Sometimes there is a delay of a few seconds. Could it be that the distance to the ESP hub is too great?

fsaris, 2024-10-12:
Can you check the logs of the esp during your testing? Then you should be able to pinpoint a little bit better where the most delay is.

closed 2024-10-19 by fsaris