fsaris/EspHome-AwoX-BLE-mesh-hub #104: Groups are always displayed as online + Color is not updated

blueJack92, 2024-10-12:
First of all, a big thank you to you for developing this component! I'm thrilled with how great the solution works in conjunction with Home Assistant. :)

Regarding my problem: The groups are recognized correctly and listed in MQTT. If the lights are supplied with power, the control of the group works without any problems. However, the color of the light of the group is not adjusted if the color has been changed in the Awox app. It would be desirable to check the color of the luminaire and transfer it to the group.

The second problem, which is more annoying, is that all groups are reported as online, even if the lamps are offline. It would be necessary to check here that if all the lamps in the group are offline, the group must also be marked as offline.

Or do I have a configuration error somewhere here? Thanks for your help!

fsaris, 2024-10-12:
> However, the color of the light of the group is not adjusted if the color has been changed in the Awox app. It would be desirable to check the color of the luminaire and transfer it to the group.

The state of the lights is updated, bit the group isn't?

What would you expect that the behaviour of the group would be of only one light in the group would be adjusted?

> The second problem, which is more annoying, is that all groups are reported as online, even if the lamps are offline. It would be necessary to check here that if all the lamps in the group are offline, the group must also be marked as offline.

Though this was already the case, but need to check this

blueJack92, 2024-10-12:
>The state of the lights is updated, bit the group isn't?
>
>What would you expect that the behaviour of the >group would be of only one light in the group would be >adjusted?

If possible, I would change the state of the group if both lights have been changed with the same values. If only one lights has been changed, the status remains as before.

In my case, however, the group status is not changed at all, even if all the lights have the same values.

> Though this was already the case, but need to check this
Unfortunately, this is not the case for me. Can I provide logs? The lights are correctly recognized as offline. However, the group is not marked as offline.

fsaris, 2024-10-12:
> > Though this was already the case, but need to check this

> Unfortunately, this is not the case for me. Can I provide logs? The lights are correctly recognized as offline. However, the group is not marked as offline.

Just tested and currently the group is shown online if one of the devices is online and offline if all devices are offline. Same for on/off.

Can you test current `main` version? That should now also update the group if all devices in the group have the same brightness, color etc

For reference https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/pull/105/commits/b692585083580fc6983cd031e9d3eec93d16ee35

blueJack92, 2024-10-12:
Thank you very much for your quick reply - and that you've even already provided an update!

I have tested it, but now I have the following problem: My groups no longer have any color at all. The brightness is set to 0 %. However, the lights shine with a warm tone and 65 % brightness.

I also have the following error messages in the log:
```
Invalid color mode 'brightness' received for entity light.eglo_rgb_tw_5Platform mqtt does not generate unique IDs. ID awox-connection-0-connected already exists - ignoring binary_sensor.none_connection_0_connected
Platform mqtt does not generate unique IDs. ID awox-connection-1-connected already exists - ignoring binary_sensor.none_connection_1_connected

```

blueJack92, 2024-10-12:
If possible and if you think this makes sense, I would like the following: If the lights from a group have different settings, the setting of (any) light should be used for the display of the group.

In my opinion, it is not intuitive if the group has a setting that does not correspond to any state of a light in this group.

fsaris, 2024-10-13:
okay, was a little bit to fast yesterday with my changes :)

Did some more testing today and now I see the group updating when the values are equal for each light.

Updating the group to the value of the latest change in the group is not that great IMO. Not sure how this is handled in other extensions, zigbee2mqtt for instance or in HA when creating own groups?

For reference: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/commit/42a4cea47bfcb9c13264218d11c32a4554784fc0

blueJack92, 2024-10-13:
Many thanks for the quick implementation! :) Unfortunately my groups are still without values ... I have reflashed the ESP32 and use the main repo.

Should I provide you with any logs? Unfortunately I don't see anything with a group in the logs ...

fsaris, 2024-10-13:
Are you sure you installed the latest version? By default the external$component is only refetched once a day (https://esphome.io/components/external_components.html#external-components-refresh).

If that doesn't work enable debug logging (if possible in verbose). It should show what values are not equal.

blueJack92, 2024-10-13:
Perfect! You were right. I don't have the latest version.

It works superbly now! Now the groups are also shown as offline when all luminaires in the group are offline. Have you optimized anything in this respect? Before, the groups were always displayed online, even if all the lights were offline.

fsaris, 2024-10-13:
> Have you optimized anything in this respect? Before, the groups were always displayed online, even if all the lights were offline.

Yes, found a small issue there during testing.

> I will buy you several beers for your support. > Thanks and cheers! :)

👍🏻

> I have (unfortunately) discovered a small (visual) problem: The groups are displayed as “Online” in HA every few seconds. Then they are displayed as “Offline” again for a few seconds. This only happens with the groups, lights are not affected.

Can you check the esphome logs? 
It should show what message from the mesh triggers the group to be offline/online. 

> In any case, the logs show that no devices were found:
> Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
> 
> EDIT: At that time, I did not supply a single Awox Light with power.

And also in this case the group was shown as online?

blueJack92, 2024-10-13:
I can't find anything about a group in the logs ... Enclosed you will find my logs.

[logs_esphome-web-f08670_logs (1).txt](https://github.com/user-attachments/files/17356827/logs_esphome-web-f08670_logs.1.txt)

fsaris, 2024-10-13:
This is when all the devices are offline?

blueJack92, 2024-10-13:
Yes, correct. All devices are offline. Nevertheless, the groups are displayed as online every few seconds (for a few seconds).

blueJack92, 2024-10-13:
[image]

blueJack92, 2024-10-13:
What is also interesting is that when I switch on a particular light, group 8 is displayed as online. However, the activated light is not in group 8 - and nothing happens when I want to control group 8.

I will try to get logs tomorrow.

fsaris, 2024-10-15:
@blueJack92 looks that the groups are not correctly set offline when no devices are connected. Need to do some more testing.

Logs would help to identify what exactly happens in your situation.

blueJack92, 2024-10-15:
Yes, I think there is still a problem somewhere.

I don't know exactly how the mesh network is/will be set up. In my case, when I de-energize a specific group of lights (Group 8), all devices are suddenly offline. As if the meshnet was set up from these devices. However, no new way is then sought to reconnect to other devices.

I am now testing with two ESP32 devices, as I have two floors. Perhaps it is also due to the connection between the individual lights being too weak.

Unfortunately I don't see anything in the logs regarding groups ... :(

blueJack92, 2024-10-15:
`
[21:27:36][D][awox.mesh:565]: Call connection for 13720
[21:27:36][I][awox.mesh:576]: No active connection for 13720, we trigger message on all could be also a group
[21:27:36][D][awox.mesh:156]: Request info again for 13395
[21:27:36][D][awox.mesh:565]: Call connection for 13395
[21:27:36][I][awox.mesh:576]: No active connection for 13395, we trigger message on all could be also a group
`
These are two lights from a group that is displayed as online, although both lights are not supplied with power.

`[21:27:37][I][awox.connection:121]: 4 queued commands (debounce timer: 2323826, next command EA, for dest: 12173)`

You can find the entire log in the appendix. As soon as I disconnect the group containing the light with the mesh id 1273 from the power supply, none of the lights can be controlled any more. I suspect that this id is causing the problems. The ESP is not looking for any other connection to other lights.
[logs_esphome-web-2652cc_logs.txt](https://github.com/user-attachments/files/17383691/logs_esphome-web-2652cc_logs.txt)

blueJack92, 2024-10-19:
Unfortunately, I still have the problem that a light is often not available. I don't know what the problem is. The ESP is 3 m away as the crow flies. As soon as I supply group 8 with power, the light appears as available.
I have already increased the max_connections to 3 to rule out the possibility that the mesh is not being set up optimally.
The RSSI is specified as -96. Even lights that are 1 m away from the ESP are shown with max. -70.

Incidentally, the groups are now displayed correctly. Where there is a problem is the display in HA (when a light is switched off, it appears again shortly afterwards as “on”, only to appear again as “off” in a few seconds) - but this has nothing to do with this component.

blueJack92, 2024-10-19:
I have a hunch what the problem might be. Group 8 consists of 6 Eglo bulbs, and I also have 4 ceiling lights on the lower floor. 

Could this cause problems if the mesh network is set up from there? It might also be useful to have an option to exclude a certain group/lights from being able to set up a mesh from there.

I have no idea whether this is the cause, but this is the (only) difference to the upper floor, where everything works perfectly.

fsaris, 2024-10-19:
Do the groups show online even when there is no active connection at all?

> Could this cause problems if the mesh network is set up from there?

Yes, when a some devices in a mesh are turned off it could be that some of the old messages are still published by the devices that are online and have the wrong value. 

If the mesh is broken into 2 sub meshed that are not linked together anymore because a light in the middle is offline device updates of one part of the mesh are not communicated anymore with the other part. And the ESP will not be able to tell correctly what the status is.

>It might also be useful to have an option to exclude a certain group/lights from being able to set up a mesh from there.

you can use the `allowed_mac_addresses` config option for this. This enables you too instruct what devices can be used as "mesh connection". (https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub?tab=readme-ov-file#allowed_mac_addresses-list-of-mac-addresses---optional)

Possible solution could be to position the ESP in the middle of the mesh so it could setup multiple connections (config: `max_connections`) to be able to get the correct status of as much devices that are online.

Or break you mesh into multiple parts, use multiple ESP devices and configure each one to only connect to a specific set of devices `allowed_mac_addresses` and only communicate/control a specific set of devices `allowed_mesh_ids`.

fsaris, 2024-10-19:
From your logs:

```
[21:27:38][D][awox.mesh:235]: Total devices: 6
[21:27:38][D][awox.mesh:238]: Available device A4:C1:38:BC:4A:E7 [19175] => rssi: -9999
[21:27:38][D][awox.mesh:238]: Available device A4:C1:38:92:2F:8D [12173] => rssi: -9999
[21:27:38][D][awox.mesh:238]: Available device A4:C1:38:C2:AC:A0 [11424] => rssi: -9999
[21:27:38][D][awox.mesh:238]: Available device A4:C1:38:4A:CB:B6 [19382] => rssi: -9999
[21:27:38][D][awox.mesh:238]: Available device A4:C1:38:CC:BF:2B [16171] => rssi: -9999
[21:27:38][D][awox.mesh:238]: Available device A4:C1:38:71:73:8E [29582] => rssi: -9999
[21:27:38][D][awox.mesh:263]: Currently 0 mesh devices reachable through active connections (10 currently known and 10 fully recognized)
```

So all your devices are offline or out-of-range. It knows 6 address, so 6 devices were at one point in range. And it found in total 10 devices in the mesh.

When this situation happens (`Currently 0 mesh devices reachable through active connections`) are the groups than online?

blueJack92, 2024-10-20:
>Or break you mesh into multiple parts, use multiple ESP devices and configure each one to only connect to a specific set of devices allowed_mac_addresses and only communicate/control a specific set of devices allowed_mesh_ids.
Did I understand that correctly? allowed_mac_addresses is there to prevent a DIRECT connection to the ESP? A connection via mesh is still possible? 

allowed_mesh_ids is there to define who is allowed into the mesh. In other words, who is allowed to connect to the ESP via detours (via another light)?

I am currently already using the two settings. However, I have specified all mesh IDs and all MAC addresses of the lights to be connected in the yaml. I use two ESPs - upper and lower floor.

One last question: Does it make sense to increase the max_connections to 3? My ESPs are located relatively centrally in the apartment.

fsaris, 2024-10-20:
> Did I understand that correctly? allowed_mac_addresses is there to prevent a DIRECT connection to the ESP? A connection via mesh is still possible?

Yes, 

> allowed_mesh_ids is there to define who is allowed into the mesh. In other words, who is allowed to connect to the ESP via detours (via another light)?

Yes

So if you split your mesh over 2 ESP32 devices also split the `mesh_ids` over the 2. There should be no overlap.

> I am currently already using the two settings. However, I have specified all mesh IDs and all MAC addresses of the lights to be connected in the yaml (I use two ESPs - upper and lower floor). Would it therefore make sense to allow only 1 of the 6 bulbs to connect directly to the ESP? All 6 are always turned on at the same time anyway, so a single direct connection would be enough.

If they ar all in reach of the esp I would assign indeed only 1 of the addresses to keep the traffic and connection attempts to a minimum. 

> One last question: Does it make sense to increase the max_connections to 3? My ESPs are located relatively centrally in the apartment.

Only if the devices are not in reach of each other, and you need more active connections to control all the devices in the mesh.

Because the esp will try to connect to a device until all the mesh id's are connected (the once defined in the config or if no config for each it once received a message on the mesh). 

Once the ESP is connected to the first device it checks if all mesh ids are connected. If not, it tries to connect to the next device to find more devices.

blueJack92, 2024-10-21:
Thank you! I have now implemented this accordingly and will keep an eye on it.

However, I now have the problem again that a group is displayed as online, although all lights are displayed as "Not available".

Can you take a look at the code for this? Can you perhaps optimize something here?

Thanks for your support!

blueJack92, 2024-10-21:
```
[17:37:43][D][awox.mesh:238]: Available device A4:C1:38:A9:92:2F [4655] => rssi: 4294967209
[17:37:43][D][awox.mesh:238]: Available device A4:C1:38:71:73:8E [29582] => rssi: 4294957297
[17:37:43][D][awox.mesh:238]: Available device A4:C1:38:5B:B4:53 [13395] => rssi: 4294957297
[17:37:43][D][awox.mesh:263]: Currently 1 mesh devices reachable through active connections (9 currently known and 9 fully recognized)
```

The two lights (ending with 8E and 53) are displayed as "Not available" in HA but the groups are displayed as "Available". The logs are displayed the devices as "Available", but the lights are offline with no power.

blueJack92, 2024-10-21:
> So the devices are offline in HA but the group is shown online. All devices of the group are offline? Are all the devices of the group linked to the same ESP32?

The affected lights are connected to the same ESP. They may only connect to this ESP.  Yes, all lights in both groups are displayed as offline. However, both groups are displayed as online.

> For both situations, Is there any active connection on that moment?

Both are the same situation.

>Number of connections and what mesh_ids are online

max_connections = 2
According to the ESP log, the two Mesh IDs are available. In HA, however, they are correctly displayed as offline for devices. However, the group to which the two devices belong is displayed as online. They are also available according to ESP.

blueJack92, 2024-10-24:
`[07:03:26][D][awox.mesh:235]: Total devices: 3
[07:03:26][D][awox.mesh:238]: Available device A4:C1:38:6F:79:20 [31008] => rssi: 4294967220
[07:03:26][D][awox.mesh:238]: Available device A4:C1:38:F0:DC:AE [23726] => rssi: 4294967212
[07:03:26][D][awox.mesh:238]: Available device A4:C1:38:33:22:86 [8838] => rssi: 4294957297
[07:03:26][D][awox.mesh:263]: Currently 2 mesh devices reachable through active connections (4 currently known and 4 fully recognized)`

Same problem with the other ESP. Two lights are supplied with power. All luminaires from one group are de-energized. Nevertheless, the group is displayed as online. The individual lights in the group are all correctly displayed as offline.

Meaning: The online detection of the lights works, the online detection of the groups does not.

fsaris, 2024-10-24:
Can you provide some more logs from the ESP?

From the time around bringing the devices online and offline. I'm interested in the processed commands and messages.

Preferably from both esp devices

blueJack92, 2024-10-24:
Do I have to increase the sensitivity of the logs for this? I will try to get logs of it today.

fsaris, 2024-10-24:
Debug logging should be enough. The debug and info messages should tell enough

blueJack92, 2024-10-26:
```
[07:35:00][D][awox.mesh:133]: No devices found to connect to
[07:35:03][D][awox.mesh:236]: Total devices: 2
[07:35:03][D][awox.mesh:239]: Available device A4:C1:38:6F:79:20 [31008] => rssi: -9999
[07:35:03][D][awox.mesh:239]: Available device A4:C1:38:06:3E:8B [16011] => rssi: -9999
[07:35:03][D][awox.mesh:264]: Currently 0 mesh devices reachable through active connections (2 currently known and 2 fully recognized)
[07:35:03][D][awox.mesh:133]: No devices found to connect to
[07:35:08][D][awox.mesh:236]: Total devices: 2
[07:35:08][D][awox.mesh:239]: Available device A4:C1:38:6F:79:20 [31008] => rssi: -9999
[07:35:08][D][awox.mesh:239]: Available device A4:C1:38:06:3E:8B [16011] => rssi: -9999
[07:35:08][D][awox.mesh:264]: Currently 0 mesh devices reachable through active connections (2 currently known and 2 fully recognized)
[07:35:08][D][awox.mesh:133]: No devices found to connect to
[07:35:13][D][awox.mesh:236]: Total devices: 2
[07:35:13][D][awox.mesh:239]: Available device A4:C1:38:6F:79:20 [31008] => rssi: -78
[07:35:13][D][awox.mesh:239]: Available device A4:C1:38:06:3E:8B [16011] => rssi: -9999
[07:35:13][D][awox.mesh:264]: Currently 0 mesh devices reachable through active connections (2 currently known and 2 fully recognized)
[07:35:13][D][awox.mesh:276]: Try to connecty to device A4:C1:38:6F:79:20 [31008] no active connection found for this device
[07:35:13][I][awox.mesh:144]: Try to connect A4:C1:38:6F:79:20 => rssi: -78
[07:35:13][I][awox.connection:150]: [0] set_state SEARCHING
[07:35:13][D][esp32_ble_client:110]: [0] [A4:C1:38:6F:79:20] Found device
[07:35:13][D][esp32_ble_tracker:671]: Found device A4:C1:38:6F:79:20 RSSI=-79
[07:35:13][D][esp32_ble_tracker:692]:   Address Type: PUBLIC
[07:35:13][D][esp32_ble_tracker:694]:   Name: '7OGT5QyD'
[07:35:13][I][awox.connection:150]: [0] set_state DISCOVERED
[07:35:13][D][esp32_ble_tracker:219]: Pausing scan to make connection...
[07:35:13][I][awox.connection:150]: [0] set_state READY_TO_CONNECT
[07:35:13][I][esp32_ble_client:067]: [0] [A4:C1:38:6F:79:20] 0x00 Attempting BLE connection
[07:35:13][I][awox.connection:150]: [0] set_state CONNECTING
[07:35:15][D][esp32_ble_client:110]: [0] [A4:C1:38:6F:79:20] ESP_GATTC_CONNECT_EVT
[07:35:15][D][esp32_ble_client:110]: [0] [A4:C1:38:6F:79:20] ESP_GATTC_OPEN_EVT
[07:35:15][I][awox.connection:150]: [0] set_state CONNECTED
[07:35:15][D][esp32_ble_tracker:270]: Starting scan...
[07:35:15][D][awox.mesh:119]: scan end
[07:35:17][D][esp32_ble_client:306]: [0] [A4:C1:38:6F:79:20] Event 46
[07:35:17][D][esp32_ble_client:110]: [0] [A4:C1:38:6F:79:20] ESP_GATTC_SEARCH_CMPL_EVT
[07:35:17][I][esp32_ble_client:227]: [0] [A4:C1:38:6F:79:20] Connected
[07:35:17][I][awox.connection:148]: Connected....
[07:35:17][D][awox.connection:280]: combine mesh name + password: 7OGT5QyD:c449ddc4
[07:35:17][D][awox.connection:259]: Listen for notifications
[07:35:17][D][awox.connection:270]: Enable notifications
[07:35:17][I][awox.mesh.mqtt:072]: Publish mesh connection status: online
[07:35:17][I][awox.mesh.mqtt:082]: Publish connection info, 1 connections, 1 device online
[07:35:17][D][esp32_ble_client:188]: [0] [A4:C1:38:6F:79:20] cfg_mtu status 0, mtu 23
[07:35:17][D][esp32_ble_client:110]: [0] [A4:C1:38:6F:79:20] ESP_GATTC_WRITE_CHAR_EVT
[07:35:17][D][esp32_ble_client:110]: [0] [A4:C1:38:6F:79:20] ESP_GATTC_REG_FOR_NOTIFY_EVT
[07:35:17][W][esp32_ble_client:275]: [0] [A4:C1:38:6F:79:20] esp_ble_gattc_get_descr_by_char_handle error, status=10
[07:35:17][D][esp32_ble_client:110]: [0] [A4:C1:38:6F:79:20] ESP_GATTC_READ_CHAR_EVT
[07:35:17][I][awox.connection:180]: Response OK, let's go
[07:35:17][D][awox.connection:280]: combine mesh name + password: 7OGT5QyD:c449ddc4
[07:35:17][I][awox.connection:185]: [0] [A4:C1:38:6F:79:20] session key 10110101110100100000000101001111001111010000001011001101101101001100011001001101101001101101010110000101101011000111010110000111
[07:35:17][D][awox.connection:580]: [0] [A4:C1:38:6F:79:20] request status update
[07:35:17][D][awox.connection:570]: [0] [A4:C1:38:6F:79:20] [65535] write_command packet DA => 00010000
[07:35:17][D][esp32_ble_client:110]: [0] [A4:C1:38:6F:79:20] ESP_GATTC_WRITE_CHAR_EVT
[07:35:17][D][esp32_ble_client:110]: [0] [A4:C1:38:6F:79:20] ESP_GATTC_WRITE_CHAR_EVT
[07:35:18][D][esp32_ble_client:110]: [0] [A4:C1:38:6F:79:20] ESP_GATTC_NOTIFY_EVT
[07:35:18][D][awox.connection:408]: status report: mesh: 31008, on: 1, color_mode: 0, sequence_mode: 0, candle_mode: 0, w_b: 127, temp: 102, c_b: 100, rgb: FF0000, mode: 1 00000001
[07:35:18][I][awox.connection:487]: device 31008: ON temp: 102 (127 %) ONLINE
[07:35:18][D][awox.mesh:524]: Sync group state, group 0: (32768)ON temp: 102 (127 %%) ONLINE
[07:35:18][D][awox.mesh:419]: Delayed publish online/offline for 31008 - online
[07:35:18][D][esp32_ble_client:110]: [0] [A4:C1:38:6F:79:20] ESP_GATTC_NOTIFY_EVT
[07:35:18][D][awox.connection:384]: online status report: mesh: 31008, on: 1, color_mode: 0, sequence_mode: 0, candle_mode: 0, w_b: 127, temp: 102, c_b: 100, rgb: FF0000, mode: 1 00000001
[07:35:18][I][awox.connection:487]: device 31008: ON temp: 102 (127 %) ONLINE
[07:35:18][D][awox.mesh:524]: Sync group state, group 0: (32768)ON temp: 102 (127 %%) ONLINE
[07:35:21][I][awox.mesh.mqtt:117]: Publish online/offline for device 31008 - online
[07:35:33][D][awox.mesh:236]: Total devices: 2
[07:35:33][D][awox.mesh:239]: Available device A4:C1:38:6F:79:20 [31008] => rssi: -77
[... cut ...]
[07:36:13][D][awox.mesh:133]: No devices found to connect to
```
Now I have the logs. I don't quite understand them ...

`[07:35:37][D][awox.mesh:291]: Clear RSSI for A4:C1:38:6F:79:20 [31008] not found the last 20 seconds`
This light was online the whole time. And also controllable. Nevertheless, it is claimed that it was not found. In the next lines it also appears again with an RSSI value.

The groups are still displayed as online. Although according to MQTT in HA all lights are displayed as offline or “not available”.

blueJack92, 2024-10-26:
If I have switched the lights of a group on and off once, it is correctly displayed as “Not available”.

However, all groups are displayed as online first. Perhaps it would be possible to mark all groups as offline by default. When they become online, they are marked correctly anyway.

blueJack92, 2024-11-10:
Unfortunately, the problem still exists. If Lights is de-energized, the group remains online.

However, the problem has now arisen that the light is sometimes displayed as "unavailable", even though the light can be controlled via the associated group. This shows the correct status of the light.

blueJack92, 2024-11-10:
What I have observed and can also reproduce: Some of my lights are supplied with continuous current, others are switched on or off by a switch.

If one of the switch-controlled lights is switched on, the status of some specific lights (which are of course supplied with power) is set to "unavailable". However, this is obviously not correct, as I can still control the lights via the associated group.

fsaris, 2024-11-11:
Having some of the devices offline, and missing some of the message in the mesh that other devices are online/offline can give strange values in HA. You probably see the same in the official AwoX app (at least that is my experience).

If you would monitor the logs closely you will probably see messages of devices that would be offline even that you know that they are online. 

> A general question: Do the individual lights try to connect to other lights independently? Or do they stick to my specified Mesh IDs and Mac addresses?

Yes, the lights do not know of the ESP config. So they try to connect to all lights in the mesh. 

> Second question: Lights become unavailable for a few seconds. After about 10 seconds they are back again. Could it be that a new "mesh path" has been searched for and found?

There is a loop that tries to fetch the device state of all devices that are currently marked as offline in the mesh every 5 seconds. https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/blob/aa687bce9b93e9a2dbd9fc563ea4091a3e7ac241/components/awox_mesh/awox_mesh.cpp#L155 

> Would it be possible to set that the lights are only displayed as unavailable after a certain period of time? With Zigbee, devices in HA are only set as "unavailable" after 6 hours by default.

This is now by intention a very short time. When you "power down" a light you want to see this (almost) immediately in HA. 

There is small timeout when a device goes offline (could maybe be made configurable) to check if there isn't a message again that a light/device is online. 
https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/blob/aa687bce9b93e9a2dbd9fc563ea4091a3e7ac241/components/awox_mesh/awox_mesh.cpp#L163

If you could provide more logs then maybe we can pinpoint what's going on and if extending the timeout would help in your case.

blueJack92, 2024-11-11:
My problem is that the current solution is very unreliable. I use Shelly in Detatched Mode and switch the Eglo/AwoX lights via the light switches in the room. If a light becomes unavailable, I can no longer switch the light.

So it becomes a frustrating experience when the lights can't be turned on or off several times a day. :(

fsaris, 2024-11-11:
I get the frustrations :)

I first had the situation that some lights were only "online" (powered) when in use. But I changed that due to reasons you describe to be always online. And I have now a Shelly i3 behind the switch that controls the light through HA.

Please share some logs so we can pinpoint the origin of your issues.

Maybe a solution could be to split your devices in 2 separate mesh networks. But that would make it harder to use the official AwoX app.

blueJack92, 2024-11-11:
:))

I already use two ESP32s - one upstairs and one downstairs. The upper one causes almost no problems. But the lower one is constantly causing problems. Lights go offline and stay offline for hours - until I restart the ESP. However, other lights in this mesh can be operated via HA.
I have now ordered another ESP32, which is supposed to have a particularly long Bluetooth range (up to 80m).
There's not much in the logs: the light goes offline because it hasn't been reported for 20 seconds.

Translated with DeepL.com (free version)

fsaris, 2024-11-11:
> There's not much in the logs: the light goes offline because it hasn't been reported for 20 seconds.

is this the part of the mesh with the devices that aren't always online? And you are sure the connection is lost and not that the ESP reboots? You see in the logs that the connection is broken?

blueJack92, 2024-11-11:
> is this the part of the mesh with the devices that aren't always online?
Yes.

I am now using this config for my esp32:

```
substitutions:
  name: "awox-ble-mesh-hub-hallway-l"
  friendly_name: AwoX BLE Mesh Hub Hallway Lower

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: False
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf

# Enable logging
logger:
  # baud_rate: 9600
  # level: VERBOSE
  logs:
    mdns: INFO
    ota: INFO
    json: INFO
    # component: DEBUG
    mqtt.sensor: INFO
    sensor: WARN
  #   esp32_ble: DEBUG
  #   esp32_ble_client: DEBUG
  #   esp32_ble_tracker: DEBUG
  #   awox.mesh.mqtt: DEBUG
  #   awox.mesh: DEBUG
  #   awox.connection: DEBUG

debug:
  update_interval: 5s

text_sensor:
  - platform: debug
    reset_reason:
      name: "Reset Reason"

sensor:
  - platform: debug
    free:
      name: "Heap Free"
    loop_time:
      name: "Loop Time"

# MQTT configuration
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_user
  password: !secret mqtt_password

  # Disable retained discovery messages until you are got the config correct
  discovery_retain: False

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

# Wi-Fi connection settings
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Set a static IP address
  manual_ip:
    static_ip: 192.168.188.226
    gateway: 192.168.188.1
    subnet: 255.255.255.0
    dns1: 192.168.188.1

# Button to trigger device restart
button:
  - platform: restart
    name: ${name} restart

# Load external components from a Git repository
external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub
      ref: main
    refresh: 0s

# Enable BLE tracking for nearby devices
esp32_ble_tracker:
  scan_parameters:
    active: False

# AwoX BLE mesh configuration
awox_mesh:
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password

  # Optional options
  address_prefix: A4:C1
  min_rssi: -80 # to prevent that the mesh tries to connect to devices that are out of range
  max_connections: 2 # max 3 connections

  # You can give a list of mesh_ids that only should be handled by this device
  # Example: When you want to set up multiple ESP modules (to control only a part of the mesh), be sure to also set the allowed_mac_addresses option.
  allowed_mesh_ids:
    - 29582
    - 11424
    - 12173
    - 19175
    - 19382
    - 16171
    - 27045
    - 4655
    - 13720
    - 13395

  # You can give a list of mac_addresses that are allowed to connect to (to exclude some devices that are often offline or for the allowed_mesh_ids option)
  allowed_mac_addresses:
    - A4:C1:38:71:73:8E
    - A4:C1:38:C2:AC:A0
    - A4:C1:38:92:2F:8D
    - A4:C1:38:BC:4A:E7
    - A4:C1:38:4A:CB:B6
    - A4:C1:38:CC:BF:2B
    - A4:C1:38:5A:E9:A4
    - A4:C1:38:A9:92:2F
    - A4:C1:38:F6:B5:98
    - A4:C1:38:5B:B4:53

  device_info:
    - product_id: 0xA6
      device_type: WHITE_TEMP
      name: EGLOBulb
      model: ESMLFm-w6-TW
      manufacturer: EGLO
      icon: mdi:lightbulb

    - product_id: 0x96
      device_type: RGB
      name: EGLO RGB+TW
      model: EGLO-RGB-TW
      manufacturer: EGLO
      icon: mdi:ceiling-light

    - product_id: 0xA9
      device_type: RGB
      name: EGLO RGB+TW
      model: EGLO-RGB-TW-IPSU
      manufacturer: EGLO
      icon: mdi:wall-sconce-flat
```