fsaris/EspHome-AwoX-BLE-mesh-hub #108: Device authentication error

josask, 2024-11-08:
Hi @fsaris, thank you very much for this component! 
Can you help me with this?

```
[15:54:41][D][awox.mesh:236]: Total devices: 1
[15:54:42][D][awox.mesh:239]: Available device A4:C1:38:xx:xx:xx [0] => rssi: -87
[15:54:42][D][awox.mesh:264]: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
[15:54:42][D][awox.mesh:270]: Try to connecty to device A4:C1:38:xx:xx:xx no mesh id known yet
[15:54:42][I][awox.mesh:144]: Try to connect A4:C1:38:xx:xx:xx => rssi: -87
[15:54:42][I][awox.connection:150]: [0] set_state SEARCHING
[15:54:42][D][esp32_ble_client:110]: [0] [A4:C1:38:xx:xx:xx] Found device
[15:54:42][D][esp32_ble_tracker:671]: Found device A4:C1:38:xx:xx:xx RSSI=-89
[15:54:42][D][esp32_ble_tracker:692]:   Address Type: PUBLIC
[15:54:42][D][esp32_ble_tracker:694]:   Name: 'itsOK'
[15:54:42][I][awox.connection:150]: [0] set_state DISCOVERED
[15:54:42][D][esp32_ble_tracker:219]: Pausing scan to make connection...
[15:54:42][I][awox.connection:150]: [0] set_state READY_TO_CONNECT
[15:54:42][I][esp32_ble_client:067]: [0] [A4:C1:38:xx:xx:xx] 0x00 Attempting BLE connection
[15:54:42][I][awox.connection:150]: [0] set_state CONNECTING
[15:54:42][D][esp-idf:000][BTU_TASK]: W (138179) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[15:54:46][D][esp32_ble_client:110]: [0] [A4:C1:38:xx:xx:xx] ESP_GATTC_CONNECT_EVT
[15:54:46][D][esp32_ble_client:110]: [0] [A4:C1:38:xx:xx:xx] ESP_GATTC_OPEN_EVT
[15:54:46][I][awox.connection:150]: [0] set_state CONNECTED
[15:54:46][D][esp32_ble_tracker:270]: Starting scan...
[15:54:46][D][awox.mesh:119]: scan end
[15:54:47][D][awox.mesh:236]: Total devices: 1
[15:54:47][D][awox.mesh:239]: Available device A4:C1:38:xx:xx:xx [0] => rssi: -87
[15:54:47][D][awox.mesh:264]: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
[15:54:47][D][awox.mesh:133]: No devices found to connect to
[15:54:48][D][esp32_ble_client:306]: [0] [A4:C1:38:xx:xx:xx] Event 46
[15:54:48][D][esp32_ble_client:110]: [0] [A4:C1:38:xx:xx:xx] ESP_GATTC_SEARCH_CMPL_EVT
[15:54:48][I][esp32_ble_client:227]: [0] [A4:C1:38:xx:xx:xx] Connected
[15:54:48][I][awox.connection:148]: Connected....
[15:54:48][D][awox.connection:280]: combine mesh name + password: itsOK:itsOKpass
[15:54:48][D][awox.connection:259]: Listen for notifications
[15:54:48][D][awox.connection:270]: Enable notifications
[15:54:48][I][awox.mesh.mqtt:072]: Publish mesh connection status: online
[15:54:48][I][awox.mesh.mqtt:082]: Publish connection info, 1 connections, 0 device online
[15:54:48][D][esp32_ble_client:188]: [0] [A4:C1:38:xx:xx:xx] cfg_mtu status 0, mtu 23
[15:54:48][D][esp32_ble_client:110]: [0] [A4:C1:38:xx:xx:xx] ESP_GATTC_WRITE_CHAR_EVT
[15:54:48][D][esp32_ble_client:110]: [0] [A4:C1:38:xx:xx:xx] ESP_GATTC_REG_FOR_NOTIFY_EVT
[15:54:49][W][esp32_ble_client:275]: [0] [A4:C1:38:xx:xx:xx] esp_ble_gattc_get_descr_by_char_handle error, status=10
[15:54:49][W][component:237]: Component esp32_ble took a long time for an operation (53 ms).
[15:54:49][W][component:238]: Components should block for at most 30 ms.
[15:54:49][D][esp32_ble_client:110]: [0] [A4:C1:38:xx:xx:xx] ESP_GATTC_READ_CHAR_EVT
[15:54:49][E][awox.connection:192]: Device authentication error: known mesh credentials are not excepted by the device. Did you re-pair them to your Awox app with a different account?
[15:54:49][I][awox.connection:198]: [0] [A4:C1:38:xx:xx:xx] response 00001110010001001001000011100001100011101001000110101100000110011010010000000000000000000000000000000000000000000000000000000000
[15:54:49][I][esp32_ble_client:084]: [0] [A4:C1:38:xx:xx:xx] Disconnecting.
[15:54:49][I][awox.connection:150]: [0] set_state DISCONNECTING
[15:54:49][I][awox.mesh:043]: disconnected
[15:54:49][I][awox.mesh.mqtt:072]: Publish mesh connection status: offline
[15:54:49][I][awox.mesh.mqtt:082]: Publish connection info, 0 connections, 0 device online
[15:54:49][D][esp-idf:000][BTU_TASK]: W (143633) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
```

fsaris, 2024-11-08:
You are sure you have the correct mesh credentials?

It can happen sometimes that a authentication request fails, but with the correct credentials this normally is gone an a second attempt.

torwag, 2024-11-25:
Hi, I have basically the same problem. The connect.z light does not connect and I get the error message:
```
[awox.connection:192]: Device authentication error: known mesh credentials are not excepted by the device. Did you re-pair them to your Awox app with a different account?
```
That is when I use the mesh credentials extracted by the form. 
However, I also get a second entry for credentials marked zigbee, same name but different password. If I use this, I get the message   
```
[awox.mesh.mqtt:082]: Publish connection info, 1 connections, 0 device online
```
but, it seems there is no connection to the light itself.
I checked the credentials several times and copied them verbatim into the config. They should be right.
I still use the AWOX HomeConnect App. Maybe this is a problem?

What would be the best way to debug this, reset the light? Delete the account and create a new one? Would love to get a small next step to try hint.

Little side question, I already run an esp32 to receive data from BLE temperature sensors. Would it be possible to merge the AwoX-BLE and the BLE stuff into a single esp32, taken that it seems that we already scratch on the memory limit?

fsaris, 2024-11-25:
connect.z are ZigBee lights and are not compatible with this component

fsaris, 2024-11-25:
It's indeed also Bluetooth but it's a different protocol

torwag, 2024-11-25:
What a pity. Is there any information about that protocol?

fsaris, 2024-11-25:
Just use a ZigBee Dongle with ZigBee2mqtt https://www.zigbee2mqtt.io/supported-devices/#v=EGLO,Eglo,AwoX
See also https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/discussions/41

torwag, 2024-11-26:
Yes, I did. I was trying to avoid it as it is the only zigbee device in my home. But it works with zigbee.

Thanks for your help and support and thank you for the integration.