fsaris/EspHome-AwoX-BLE-mesh-hub #112: Exception in _state_received when handling msg on...

Dillton, 2024-12-12:
Hello @fsaris.

I have noticed that I am receiving sometimes such error messages in HA.
Have you seen somethingg like this before?

Logger: homeassistant.components.mqtt.client
Source: util/logging.py:95
integration: MQTT (documentation, issues)
First occurred: 01:22:09 (66 occurrences)
Last logged: 10:54:22

Exception in _state_received when handling msg on 'awox-ble-mesh-hub/32770/state': 'OFF' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/entity.py", line 1470, in _message_callback msg_callback(msg) ~~~~~~~~~~~~^^^^^ File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 418, in _state_received values = json_loads_object(msg.payload) File "/usr/src/homeassistant/homeassistant/util/json.py", line 56, in json_loads_object value: JsonValueType = json_loads(obj) ~~~~~~~~~~^^^^^ File "/usr/src/homeassistant/homeassistant/util/json.py", line 42, in json_loads return orjson.loads(obj) # type:ignore[no-any-return] ~~~~~~~~~~~~^^^^^ orjson.JSONDecodeError: unexpected character: line 1 column 1 (char 0)
Exception in _state_received when handling msg on 'awox-ble-mesh-hub/32772/state': 'OFF' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/entity.py", line 1470, in _message_callback msg_callback(msg) ~~~~~~~~~~~~^^^^^ File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 418, in _state_received values = json_loads_object(msg.payload) File "/usr/src/homeassistant/homeassistant/util/json.py", line 56, in json_loads_object value: JsonValueType = json_loads(obj) ~~~~~~~~~~^^^^^ File "/usr/src/homeassistant/homeassistant/util/json.py", line 42, in json_loads return orjson.loads(obj) # type:ignore[no-any-return] ~~~~~~~~~~~~^^^^^ orjson.JSONDecodeError: unexpected character: line 1 column 1 (char 0)
Exception in _state_received when handling msg on 'awox-ble-mesh-hub/32768/state': 'OFF' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/entity.py", line 1470, in _message_callback msg_callback(msg) ~~~~~~~~~~~~^^^^^ File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 418, in _state_received values = json_loads_object(msg.payload) File "/usr/src/homeassistant/homeassistant/util/json.py", line 56, in json_loads_object value: JsonValueType = json_loads(obj) ~~~~~~~~~~^^^^^ File "/usr/src/homeassistant/homeassistant/util/json.py", line 42, in json_loads return orjson.loads(obj) # type:ignore[no-any-return] ~~~~~~~~~~~~^^^^^ orjson.JSONDecodeError: unexpected character: line 1 column 1 (char 0)

fsaris, 2024-12-14:
Do you have RGB lights or plugs?

Dillton, 2024-12-14:
Yes 8x Eglo spot 120 in groups like 3 + 3 + 2.
I have found few hours after opening this thread that there were some changes made regarding groups is main branch so I've switched to main and so far so good but.

Dillton, 2024-12-14:
I have one more question not related to this topic but to config of max_connections where currently we can enable up to 3 separate connections and ESP can connect to up to 3 devices so the mesh is more stable right?
I have noticed many times that lights going frequently offline for a few seconds (but only devices in office or bedroom see picture below) so I have enabled 3 connections but only 1 connection is established most of the time, sometimes I saw 2 connections for a few hours but thats it.
So the question is could we expand the connection config so we can strictly say always connection 1 to device A,  2 to device B and 3 to device C?
Below is approximate layout of my appartment and the idea of the additional config:
[image]

If I remember correctly from my past observations the ESP is mostly connected to device A.
Let me know what do you think then I would open a separate thread.

Dillton, 2024-12-14:
> Hello @fsaris.
>
> I have noticed that I am receiving sometimes such error messages in HA. Have you seen somethingg like this before?
>
> Logger: homeassistant.components.mqtt.client Source: [...]

Update for IDs:
I have checked the IDs and the errors are for sure related to groups.
32770 - Bedroom group
32772 - Office group
32768 - Living room group

fsaris, 2024-12-14:
> Update for IDs: I have checked the IDs and the errors are for sure related to groups. 32770 - Bedroom group 32772 - Office group 32768 - Living room group

You have probably some retained mqtt messages with invalid config setup. A mismatch in what is configured in the discovery message and in the received state.

Try removing all retained messages and restart the esp device.

fsaris, 2024-12-14:
> If I remember correctly from my past observations the ESP is mostly connected to device A.
Let me know what do you think then I would open a separate thread

It will check if all known devices are reachable through the current connection(s) and only when it misses a device an additional connection will be setup.

Dillton, 2024-12-18:
> You have probably some retained mqtt messages with invalid config setup. A mismatch in what is configured in the discovery message and in the received state.
>
> Try removing all retained messages and [...]

MQTT cleaned and running on main branch. So far no errors in log. Closing the issue for now.

Dillton, 2024-12-21:
Hello, seems it's not solved.
Today it has appeared again at the time when ESP was restarting due to ESPHome update.
So apparently this error occurs sometimes after restart.

fsaris, 2024-12-21:
Can you share your yaml config(s) and the exact topics from the errors in HA

Dillton, 2024-12-23:
Sure. Once this occurs again I will share everything I can.

Dillton, 2024-12-23:
> It will check if all known devices are reachable through the current connection(s) and only when it misses a device an additional connection will be setup.

Regarding this second topic as you can see from history below the closest devices (living room) are whole time connected without any issue but the rest is going unavailable every few minutes sometimes even every few seconds. ESP restart solves this for a few days.
[image]

fsaris, 2025-02-14:
closing for now. please provide more info (config+logs) if you need more assistance

closed 2025-02-14 by fsaris