fsaris/EspHome-AwoX-BLE-mesh-hub #116: Awox EPanel-120-10, Awox smart control

danielp777, 2024-12-15:
Hello there.....

i bought esp32, i got no working solution... many hours wasted....

I have more Awox EPanel-120-10, with the controls SMPWBm10GEa...

Do i have the correct solution when i trymthis ESPHome-Awox-BLE-mesh-hub??

log in MQTT for the mesh-network:
searcher: searcher_re:
    0: re.compile('bluetooth')
    1: EOF
2024-12-15 18:57:50.931 WARNING (AwoxMeshCommands-khClL4HZ) [custom_components.awox.awox_mesh] [khClL4HZ] Command failed, retry 1
2024-12-15 18:57:50.932 WARNING (AwoxMeshCommands-khClL4HZ) [custom_components.awox.awox_mesh] [khClL4HZ] Command failed, retry 2
2024-12-15 18:57:50.932 WARNING (AwoxMeshCommands-khClL4HZ) [custom_components.awox.awox_mesh] [khClL4HZ] Command failed, retry 3
2024-12-15 18:57:50.940 ERROR (MainThread) [custom_components.awox.awox_mesh] Error fetching awox data: No device connected
2024-12-15 18:58:51.699 ERROR (SyncWorker_13) [custom_components.awox.bluetoothctl] Failed to start scan: Timeout exceeded.

i hope there is a solution for my Panels... Thanks!
Daniel

fsaris, 2024-12-15:
What rssi values do you get for your devices?

And what is the exact type of your Panels?

danielp777, 2024-12-15:
i think i have no devices found, so no rssi values...

2024-12-15 19:19:26.624 WARNING (AwoxMeshCommands-khClL4HZ) [custom_components.awox.awox_mesh] [khClL4HZ] Command failed, retry 1
2024-12-15 19:19:26.624 WARNING (AwoxMeshCommands-khClL4HZ) [custom_components.awox.awox_mesh] [khClL4HZ] Command failed, retry 2
2024-12-15 19:19:26.625 WARNING (AwoxMeshCommands-khClL4HZ) [custom_components.awox.awox_mesh] [khClL4HZ] Command failed, retry 3
2024-12-15 19:20:26.368 ERROR (SyncWorker_0) [custom_components.awox.bluetoothctl] Failed to start scan: Timeout exceeded.

it looks like, that the mesh can not be discovered?

my devices:

EGLOPanel by AwoX
EPanel-120-10, Version 2.2.6(HW 1.0)
EPanel_600, Version 2.2.17(HW 1.0)

danielp777, 2024-12-16:
Hi,

now i tested so much and hope for help :-)

thanks 
Daniel

danielp777, 2024-12-23:
Other esp32 and it worked...

fsaris, 2024-12-24:
Guess this is resolved then

closed 2024-12-24 by fsaris