fsaris/EspHome-AwoX-BLE-mesh-hub #119: BLE Connection Fails

jekkel, 2025-02-17:
Since quite a while any newly compiled firmware fails to connect to the AwoX LED panels despite the discovery seems to work fine. This is similar to the issue I described in #74 .
I have ran the old firmware built with EspHome 2024.1.4 on [37a4e6b6f07c70bba14135ddfb6b544f80c1db70](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/commit/b794e523b13d8c39e80acc6535b694a91bf0dbf5) until recently which worked rather stable (occasional restarts required though) but due to a changed network configuration I need to recompile the firmware running into the aforementioned issue.

I do have some debugging info enabled in the yaml and increased the ble scan window a bit:

```yaml

esp32_ble_tracker:
  scan_parameters:
    active: false
    window: 150ms

debug:
  update_interval: 5s

text_sensor:
  - platform: debug
    reset_reason:
      name: "Reset Reason"
        
sensor:
  - platform: debug
    free:
      name: "Heap Free"
    loop_time:
      name: "Loop Time"
```

But beside that, I use a pretty default config. I captured the log and what I think where the issue shows is this repeating part:
```
[19:55:52][D][esp32_ble_tracker:273]: Starting scan...
[19:55:52][D][awox.mesh:119]: scan end
[19:55:52][D][awox.mesh:236]: Total devices: 3
[19:55:52][D][awox.mesh:238]: Available device A4:C1:38:98:16:AB [0] => rssi: -79
[19:55:52][D][awox.mesh:238]: Available device A4:C1:38:4D:E6:FD [0] => rssi: -87
[19:55:52][D][awox.mesh:238]: Available device A4:C1:38:F1:E8:CE [0] => rssi: -98
[19:55:52][D][awox.mesh:261]: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
[19:55:52][D][awox.mesh:270]: Try to connecty to device A4:C1:38:98:16:AB no mesh id known yet
[19:55:52][I][awox.mesh:143]: Try to connect A4:C1:38:98:16:AB => rssi: -79
[19:55:52][I][awox.connection:150]: [0] set_state SEARCHING
[19:55:52][D][esp32_ble_client:110]: [0] [A4:C1:38:98:16:AB] Found device
[19:55:52][D][esp32_ble_tracker:677]: Found device A4:C1:38:98:16:AB RSSI=-82
[19:55:52][D][esp32_ble_tracker:698]:   Address Type: PUBLIC
[19:55:52][D][esp32_ble_tracker:700]:   Name: 'G0hGF8B0'
[19:55:52][I][awox.connection:150]: [0] set_state DISCOVERED
[19:55:52][W][component:237]: Component awox_mesh took a long time for an operation (65 ms).
[19:55:52][W][component:238]: Components should block for at most 30 ms.
[19:55:52][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[19:55:52][I][awox.connection:150]: [0] set_state READY_TO_CONNECT
[19:55:52][I][esp32_ble_client:066]: [0] [A4:C1:38:98:16:AB] 0x00 Attempting BLE connection
[19:55:52][I][awox.connection:150]: [0] set_state CONNECTING
[19:55:52][D][esp-idf:000][BTU_TASK]: W (617707) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[19:55:52][D][esp-idf:000][BTU_TASK]: W (618066) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[19:55:53][D][esp-idf:000][BTU_TASK]: W (618816) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[19:55:54][D][esp-idf:000][BTU_TASK]: W (620118) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x3e

[19:55:54][D][esp-idf:000][BTU_TASK]: W (620122) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x3e

[19:55:54][D][esp-idf:000][BTU_TASK]: W (620124) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[19:55:54][D][esp32_ble_client:171]: [0] [A4:C1:38:98:16:AB] ESP_GATTC_DISCONNECT_EVT, reason 62
[19:55:54][I][awox.connection:150]: [0] set_state IDLE
[19:55:54][D][awox.connection:138]: [0] [A4:C1:38:98:16:AB] ESP_GATTC_DISCONNECT_EVT, reason 62
[19:55:54][I][awox.mesh:043]: disconnected
[19:55:54][D][esp32_ble_tracker:273]: Starting scan...
[19:55:54][D][awox.mesh:119]: scan end
```

Find the full log attached.

I happily provide any further data or information and can recompile with any changes requested. Right now I am unable to control the LED panels.

Thanks for all the efforts by the way! I really appreciate!

fsaris, 2025-02-17:
Seems similar to https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/discussions/118#discussioncomment-12203318

Try a clean wired install

jekkel, 2025-02-17:
Wow what a fast response! Could you please elaborate (one more time) what's a ? From my understanding a "clean wired install" is:

* in ESPHome: "Clean Build Files"
* in ESPHome: "Install" and then download 
* Flash via USB/Serial, I did it via `esptool`

I'm sure I did it that way. 

But your pointer was helpful nevertheless as I double checked potential issues for a crashing BLE stack and found out I could have chosen a different board (`esp32dev` instead of `nodemcu-32s`). I did the whole procedure again with a different board I can confirm the issue to be (almost) gone when the boards reboots after flashing via USB.

Still occasionally the BT stack seems to crash every now and then and I found out my previous USB power source for the board is not sufficient anymore, i.e. prevents BLE connections.

I'll close the ticket for now as it doesn't seem to be a AwoX issue in any way. Maybe adding some more details about BLE stability troubleshooting in e.g. the README.md could be a good idea.

closed 2025-02-17 by jekkel

fsaris, 2025-02-17:
Thats indeed what I ment with a clean install.

The power supply is indeed also something to take into account.

Will update the readme