fsaris/EspHome-AwoX-BLE-mesh-hub #121: Issues with adding a EGLO Salobrena-C BLE lamp

Dev-Sniper, 2025-03-01:
I have a couple of EGLO Salobrena-C lamps from 2021 that can be connected to the Awox Smart Control / Awox Home Control App via BLE. The firmware on them is up to date (Version 2.6.17) and they can be reset and reconnected to the official Apps. I´ve talked to EGLOs customer support so I know that these lamps have BLE connectivity but can´t be controlled by a zigbee hub. When resetting the lamps they show up with the name "unpaired" on Android phones (but not on iPhones and a bluetooth dongle I tried to use didn´t get them either). Thus I´ve switched to a regular ESP-32 (ESP32-WROOM32D) that I flashed with the ESPHome "Bluetooth proxy" prebuilt image. That works and the device can be found in the ESPHome Dashboard of my Homeassistant instance. Then I went through the [instructions](https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/discussions/71#discussion-6234371).

Step 1: the ESP can be flashed and connects to the wireless network

Step 2: adding the logger works (although I don´t have the "Logs" -> "wireless" option, but the log button on the added device shows the logs wether it´s connected via wifi or usb)

the output is:

`INFO ESPHome 2025.2.1
INFO Reading configuration /config/esphome/awox-ble-mesh-hub.yaml...
INFO Starting log output from [IP] using esphome API
INFO Successfully connected to awox-ble-mesh-hub @ [IP] in 0.100s
INFO Successful handshake with awox-ble-mesh-hub @ [IP] in 0.151s
[17:04:09][I][app:100]: ESPHome version 2025.2.1 compiled on [Date & time]
[17:04:09][C][wifi:600]: WiFi:
[17:04:09][C][wifi:428]:   Local MAC: [MAC]
[17:04:09][C][wifi:433]:   SSID: [SSID]
[17:04:09][C][wifi:436]:   IP Address: [IP]
[17:04:09][C][wifi:439]:   BSSID: [BSSID]
[17:04:09][C][wifi:441]:   Hostname: 'awox-ble-mesh-hub'
[17:04:09][C][wifi:443]:   Signal strength: -56 dB ▂▄▆█
[17:04:09][V][wifi:445]:   Priority: 0.0
[17:04:09][C][wifi:447]:   Channel: 1
[17:04:09][C][wifi:448]:   Subnet: [Subnet]
[17:04:09][C][wifi:449]:   Gateway: [Gateway]
[17:04:09][C][wifi:450]:   DNS1: [DNS]
[17:04:09][C][wifi:451]:   DNS2: [DNS]
[17:04:09][C][logger:177]: Logger:
[17:04:09][C][logger:178]:   Max Level: VERY_VERBOSE
[17:04:09][C][logger:179]:   Initial Level: VERY_VERBOSE
[17:04:09][C][logger:181]:   Log Baud Rate: 115200
[17:04:09][C][logger:182]:   Hardware UART: UART0
[17:04:09][C][mdns:116]: mDNS:
[17:04:09][C][mdns:117]:   Hostname: awox-ble-mesh-hub
[17:04:09][V][mdns:118]:   Services:
[17:04:09][V][mdns:120]:   - _esphomelib, _tcp, 6053
[17:04:09][V][mdns:122]:     TXT: friendly_name = awox-ble-mesh-hub
[17:04:10][V][mdns:122]:     TXT: version = 2025.2.1
[17:04:10][V][mdns:122]:     TXT: mac = [mac]
[17:04:10][V][mdns:122]:     TXT: platform = ESP32
[17:04:10][V][mdns:122]:     TXT: board = esp32dev
[17:04:10][V][mdns:122]:     TXT: network = wifi
[17:04:10][V][mdns:122]:     TXT: api_encryption = [api_encryption]
[17:04:10][C][esphome.ota:073]: Over-The-Air updates:
[17:04:10][C][esphome.ota:074]:   Address: awox-ble-mesh-hub.local:3232
[17:04:10][C][esphome.ota:075]:   Version: 2
[17:04:10][C][esphome.ota:078]:   Password configured
[17:04:10][C][safe_mode:018]: Safe Mode:
[17:04:10][C][safe_mode:019]:   Boot considered successful after 60 seconds
[17:04:10][C][safe_mode:021]:   Invoke after 10 boot attempts
[17:04:10][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[17:04:10][C][api:140]: API Server:
[17:04:10][C][api:141]:   Address: awox-ble-mesh-hub.local:6053
[17:04:10][C][api:143]:   Using noise encryption: YES
[17:04:15][VV][scheduler:225]: Running interval '' with interval=60000 last_execution=160620 (now=220621)`

Step 3: Skipped because I already set the framework type to "esp-idf" during Step 1

Step 4 & 5: Setting up the MQTT worker... this might or might not be the issue. I have created a MQTT Broker instance using the HomeAssistant Mosquitto Add-on. I´ve added the credentials for this broker to to config file and removed the homeassistant api parts. I then flashed the new config to the ESP32 and got this log:

`INFO ESPHome 2025.2.1
INFO Reading configuration /config/esphome/awox-ble-mesh-hub.yaml...
INFO Starting log output from awox-ble-mesh-hub/debug
INFO Connected to MQTT broker!`

So I don´t get information on the WiFi network anymore but I did get information regarding MQTT.
I haven´t used MQTT before so that´s not my area of expertise but I have published a message through the MQTT broker instance to the topic "awox-ble-mesh-hub/debug" and it did show up in the logs for my device so I would assume that MQTT and the wireless connection both work. Just for fun I published a message to "awox-ble-mesh-hub/test" and that didn´t get picked up by the device but by the MQTT broker that I previously configured to subscribe to "awox-ble-mesh-hub/#" so in theory I should see every MQTT message within my MQTT broker instance even if my device isn´t subscribed to a topic.

Step 6: I removed the captive portal part and copied the external_components section from the example YAML file. Then I installed the config on the ESP and got the same results as in Step 5. Tested the MQTT broker again, everything seems to work. I then reset one of the lamps. It showed up as unpaired on an android phone but neither my MQTT broker (which is still subscribed to "/#" nor my device logs changed. I also turned on two of the lamps within the mesh but that didn´t lead to any results either. So I´m not really sure if my devices are just incompatible with the add-on or if I messed up anything. 

Here is the final YAML configuration file that´s on my ESP right now:
```yaml
esphome:
  name: awox-ble-mesh-hub
  friendly_name: awox-ble-mesh-hub

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:
  level: VERY_VERBOSE

mqtt:
  broker: [broker name]
  username: [user name]
  password: [user password]

  # Disable retained discovery messages until you are got the config correct
  discovery_retain: True
  
ota:
  - platform: esphome
    password: [ota password]

wifi:
  ssid: [ssid]
  password: [password]

external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub
      ref: main
  #  ref: dev
  # - source:
  #     type: local
  #     path: components

esp32_ble_tracker:
  scan_parameters:
    active: false **_(I have tried setting this to true just in case it might change anything but the results were the same)_**

awox_mesh:
  mesh_name: [mesh name retrieved via the form linked in the github repo]
  mesh_password: [password retrieved via the form linked in the github repo]
```

Before every build I run "Clean MQTT" and "Clean Build Files" within ESP Home and check that the configuration is valid. iBeacon managed to find devices that have a name starting with the mesh name but it can´t do anything with the added devices (name schema is "[mesh name] [4 digit ID with alphanumerical characters]" example: awoxMesh D18F

fsaris, 2025-03-01:
The zigbee variants are not supported. They use a different protocol voor the ble connection.
I can recommend ZigBee2mqtt in your case

Dev-Sniper, 2025-03-01:
It´s not a zigbee compatible device though. I asked EGLOs customer support what the devices supports because I´ve already got a zigbee controller which I thought I could use but it only works with BLE. So I can´t use Zigbee2mqtt. I only added the comment regarding zigbee because I´ve noticed in other issues that some people had zigbee compatible devices so their problems were solveable with that. But that won´t work with my devices

fsaris, 2025-03-02:
Do you have a log with the awox_mesh enabled?

fsaris, 2025-03-02:
Resetting isn't needed, they need to be part of the mesh with the credentials you added to the yaml.

There should also be something about the awox scanning etc

Does the device show online in the esphome dashboard?

Dev-Sniper, 2025-03-02:
The ESP32 is shown as online within ESPHome and it can receive messages from the MQTT Broker. But it doesn´t detect any of the devices. Yeah I kinda wondered why it doesn´t show information about the wifi connection etc. anymore but maybe that´s in a different log file? Or could it be that the discovered devices don´t get published in a "awox-ble-mesh-hub/..." topic?

Dev-Sniper, 2025-03-02:
By the way just to be sure. I got the following data from the form:
```
{"mesh_name":"[name]","mesh_password":"[password]","service":"zigbee"}
{"mesh_name":"[name]","mesh_password":"[other password]","service":"mesh"}
```
my YAML currently has `mesh_name: [name]` and `mesh_password: [other password]` since the other entry references zigbee and the form states that the credentials for `service: mesh` should be used. So I assume that this part of the configuration is correct?

Dev-Sniper, 2025-03-03:
Then it seems like I‘ve got the MQTT and the mesh settings correct. But regardless of how often I reboot / flash the ESP it doesn‘t show more than the minimal log listing which topic it‘s subscribed to and that it managed to connect to the MQTT broker

Dev-Sniper, 2025-03-03:
Well it‘s currently set to very verbose so setting it to debug probably won‘t change anything but I‘ll try that later

fsaris, 2025-03-03:
It could as very verbose can result in memory issues and crash/restart the device

Dev-Sniper, 2025-03-03:
I tried setting it to debug mode but that didn´t lead to a different output. it´s still the short log + any messages I manually publish

fsaris, 2025-03-03:
With awox_mesh disabled in the config do you see then more in the logs? It should report about ble etc .
Are you sure you are looking at the correct logs?

Did you try to read the logs over a wired connection?

Dev-Sniper, 2025-03-10:
I replaced my logging part of the config with your debug settings but it didn´t change anything. So no new logs.