fsaris/EspHome-AwoX-BLE-mesh-hub #122: Compatible with bluetooth proxy?

racinmat, 2025-03-10:
Hi, is this component compatible with bluetooth proxy, or do I need to use a separate HW for a bluetooth proxy?
I noticed there is
```
esp32_ble_tracker:
  scan_parameters:
    active: false
```
in https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/blob/main/awox-ble-mesh-hub.yaml#L64-L66
but 
```
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true
```
in https://esphome.io/components/bluetooth_proxy.html so I guess they are not compatible?

fsaris, 2025-03-13:
No currently not because then you need to enable the API and mqtt both. And that will probably give issues.

There is a idea to convert this component to switch from mqtt to the api but that's currently not high on my list

fsaris, 2025-03-13:
https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/issues/59

closed 2025-03-13 by fsaris

racinmat, 2025-03-13:
I see, thanks.  @michalpulda  how do you select to which device does what connect? If I have some awox devices and then few non-awox devices, how to force them to connect to the correct device?

michalpulda, 2025-03-13:
In my setup, Awox devices will only connect to the AwoX BLE mesh hub, not to a bluetooth proxy.

racinmat, 2025-03-14:
How can you achieve such things? I'm curious. And how do you make sure the non-awox will only connect to the bluetooth proxy?