fsaris/EspHome-AwoX-BLE-mesh-hub #124: problem with esp 32 awox

Edwardder3, 2025-06-26:
Hallo i have a problem and i dont now what the problem is can you try to help 

my english is not so goot so sory for that but i gat thith error 

Logger: homeassistant.components.mqtt.client
Quelle: util/logging.py:167
Integration: MQTT (Dokumentation, Probleme)
Erstmals aufgetreten: 20:31:23 (4 Vorkommnisse)
Zuletzt protokolliert: 21:26:40

Exception in _state_received when handling msg on 'beleuchtung/32776/state': 'OFF' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/entity.py", line 1524, in _message_callback msg_callback(msg) ~~~~~~~~~~~~^^^^^ File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 289, in _state_received values = json_loads_object(msg.payload) File "/usr/src/homeassistant/homeassistant/util/json.py", line 56, in json_loads_object value: JsonValueType = json_loads(obj) ~~~~~~~~~~^^^^^ File "/usr/src/homeassistant/homeassistant/util/json.py", line 42, in json_loads return orjson.loads(obj) # type:ignore[no-any-return] ~~~~~~~~~~~~^^^^^ orjson.JSONDecodeError: unexpected character: line 1 column 1 (char 0)
Exception in _state_received when handling msg on 'beleuchtung/32781/state': 'OFF' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/entity.py", line 1524, in _message_callback msg_callback(msg) ~~~~~~~~~~~~^^^^^ File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 289, in _state_received values = json_loads_object(msg.payload) File "/usr/src/homeassistant/homeassistant/util/json.py", line 56, in json_loads_object value: JsonValueType = json_loads(obj) ~~~~~~~~~~^^^^^ File "/usr/src/homeassistant/homeassistant/util/json.py", line 42, in json_loads return orjson.loads(obj) # type:ignore[no-any-return] ~~~~~~~~~~~~^^^^^ orjson.JSONDecodeError: unexpected character: line 1 column 1 (char 0)
Exception in _state_received when handling msg on 'beleuchtung/32775/state': 'OFF' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/entity.py", line 1524, in _message_callback msg_callback(msg) ~~~~~~~~~~~~^^^^^ File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 289, in _state_received values = json_loads_object(msg.payload) File "/usr/src/homeassistant/homeassistant/util/json.py", line 56, in json_loads_object value: JsonValueType = json_loads(obj) ~~~~~~~~~~^^^^^ File "/usr/src/homeassistant/homeassistant/util/json.py", line 42, in json_loads return orjson.loads(obj) # type:ignore[no-any-return] ~~~~~~~~~~~~^^^^^ orjson.JSONDecodeError: unexpected character: line 1 column 1 (char 0)

fsaris, 2025-06-26:
@Edwardder3 Do you have this set to `False`? https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub/blob/aa687bce9b93e9a2dbd9fc563ea4091a3e7ac241/awox-ble-mesh-hub.yaml#L41 

Then please restart home assistant and after that the esp that should fix it. Else you will need to reset/remove this invalid message from your MQTT broker.

You probably switch the device type to the correct one by setting the device info.

fsaris, 2025-07-05:
This looks like a disconnect of the connection. Since it only happens once I expected it reconnected again and it fixed itself. Or is it still giving you issues?

Edwardder3, 2025-07-09:
[image]

fsaris, 2025-08-05:
@Edwardder3 can you share your esphome YAML config? And how many devices are connected to the ESPHome?

Edwardder3, 2025-08-06:
i have 36 devices conectet to the esp

Edwardder3, 2025-08-06:
ok i try it