fsaris/EspHome-AwoX-BLE-mesh-hub #125: Esphome 2025.7.x problem after update

Dillton, 2025-07-17:
Hello @fsaris 

After some time we have again some change in Esphome which breakt this great AwoX integration.
The ESP simply won't connect to the mesh even if the devices are discovered properly.

Unfortunately I didn't notice any error in the log but I believe you will be able to fix it.
For now the hotfix is to downgrade to 2025.6.3 and reinstall the ESP.

Bushrider, 2025-07-17:
Same for me, happy holiday !

Edwardder3, 2025-07-17:
i have the same problem and i install all new and get after install this 

00:25:38][I][awox.mesh:143]: Try to connect A4:C1:38:9A:56:B0 => rssi: -86
[00:25:38][I][awox.connection:150]: [1] set_state SEARCHING
[00:25:38][D][esp32_ble_client:208]: [1] [A4:C1:38:9A:56:B0] Found device
[00:25:38][D][esp32_ble_tracker:807]: Found device A4:C1:38:9A:56:B0 RSSI=-91
[00:25:38][D][esp32_ble_tracker:828]:   Address Type: PUBLIC
[00:25:38][I][awox.connection:150]: [1] set_state DISCOVERED
[00:25:38][D][esp32_ble_tracker:116]: connecting: 1, discovered: 1, searching: 0, disconnecting: 0
[00:25:46][D][awox.mesh:290]: Clear RSSI for A4:C1:38:7E:25:4F [0] not found the last 20 seconds
[00:25:49][D][awox.mesh:290]: Clear RSSI for A4:C1:38:9A:D7:3C [0] not found the last 20 seconds
[00:25:51][D][awox.mesh:290]: Clear RSSI for A4:C1:38:44:65:EB [0] not found the last 20 seconds
[00:25:52][D][awox.mesh:290]: Clear RSSI for A4:C1:38:47:E9:D3 [0] not found the last 20 seconds
[00:25:52][D][awox.mesh:290]: Clear RSSI for A4:C1:38:5A:EF:93 [0] not found the last 20 seconds
[00:25:53][D][awox.mesh:290]: Clear RSSI for A4:C1:38:F4:C5:DB [0] not found the last 20 seconds
[00:25:53][D][awox.mesh:290]: Clear RSSI for A4:C1:38:9C:CE:BD [0] not found the last 20 seconds
[00:25:53][D][awox.mesh:290]: Clear RSSI for A4:C1:38:9A:56:B0 [0] not found the last 20 seconds
[00:25:53][D][awox.mesh:290]: Clear RSSI for A4:C1:38:9C:96:14 [0] not found the last 20 seconds
[00:25:53][D][awox.mesh:290]: Clear RSSI for A4:C1:38:9A:DF:60 [0] not found the last 20 seconds
[00:25:53][D][awox.mesh:290]: Clear RSSI for A4:C1:38:1B:3A:94 [0] not found the last 20 seconds
[00:25:53][D][awox.mesh:290]: Clear RSSI for A4:C1:38:9A:75:4B [0] not found the last 20 seconds
[00:25:53][D][awox.mesh:290]: Clear RSSI for A4:C1:38:2B:C7:9A [0] not found the last 20 seconds
[00:26:23][I][safe_mode:042]: Boot seems successful; resetting boot loop counter
[00:26:23][D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed

and nothing works so when i try to read the logs nothing happens 

happy holiday :)

Dillton, 2025-07-18:
> i have the same problem and i install all new and get after install this
>
> 00:25:38][I][awox.mesh:143]: Try to connect A4:C1:38:9A:56:B0 => rssi: -86 [00:25:38][I][awox.connection:150]: [1] set_state [...]

To be honest I have noticed the same in log but wasn't sure if could be related.
Happy holiday @fsaris

Trellmor, 2025-07-22:
@Jinguba install a previous backup/snapshot in Home Assistant or install 2025.6 from e.g. https://github.com/khenderick/esphome-legacy-addons

Edwardder3, 2025-07-22:
I am realy new at all this things but i think the change is the new IDF versions. The ESP-IDF 4.x support removed - ESP-IDF 5.3.2+ now required so i install an old ESP-HOME and i have the same Error. I use the awox addon for the time but it dont works well so i think the hole code must be changed.

WMax77, 2025-07-29:
Hello evryone, it seems I've the same problem. Older version worked but not stable over longer time (loosing connection) so I tryed an update and since tem I'm not able to get it running.

My config (with some changes against my old one, using allowed_mesh_ids/allowed_mac_addresses)

```python

  substitutions:
  name: esphome-web-23819c
  friendly_name: ESPHome-AWOX-BLE

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    

# Enable logging
logger:

# Enable Home Assistant API
#api:

mqtt:
  broker: 10.0.0.--
  username: !secret mqtt_user
  password: !secret mqtt_pass
  
  discovery_retain: False

# Allow Over-The-Air updates
ota:
  platform: esphome
# Allow provisioning Wi-Fi via serial
#improv_serial:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Set up a wifi access point
  #ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
#captive_portal:

#dashboard_import:
#  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
#  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
#web_server:

external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub
      ref: main
  #  ref: dev
  # - source:
  #     type: local
  #     path: components

esp32_ble_tracker:
  scan_parameters:
    active: false

awox_mesh:
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password

  allowed_mesh_ids:
    - 22419

  allowed_mac_addresses:
    - A4:C1:38:59:F8:FC
    - A4:C1:38:6A:57:93

  device_info:
 # Example device type, see device info in HomeAssistant or MQTT message to find the 'product_id'
    - product_id: 0x35
      device_type: RGB
      name: Giron-C 17W
      model: EGLO 32589
      manufacturer: EGLO
      icon: mdi:wall-sconce-flat
    
    - product_id: 0x42
      device_type: RGB
      name: EGLO Surface square
      model: EFueva_300s
      manufacturer: EGLO

And the log I get always:

[21:35:48][D][awox.mesh:236]: Total devices: 2
[21:35:48][D][awox.mesh:238]: Available device A4:C1:38:6A:57:93 [0] => rssi: -60
[21:35:48][D][awox.mesh:238]: Available device A4:C1:38:59:F8:FC [0] => rssi: -91
[21:35:48][D][awox.mesh:261]: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
[21:35:48][D][awox.mesh:133]: No devices found to connect to
[21:35:53][D][awox.mesh:236]: Total devices: 2
[21:35:54][D][awox.mesh:238]: Available device A4:C1:38:6A:57:93 [0] => rssi: -60
[21:35:54][D][awox.mesh:238]: Available device A4:C1:38:59:F8:FC [0] => rssi: -91
[21:35:54][D][awox.mesh:261]: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
[21:35:54][D][awox.mesh:133]: No devices found to connect to
[21:35:58][D][awox.mesh:236]: Total devices: 2
[21:35:58][D][awox.mesh:238]: Available device A4:C1:38:6A:57:93 [0] => rssi: -60
[21:35:58][D][awox.mesh:238]: Available device A4:C1:38:59:F8:FC [0] => rssi: -91
[21:35:58][D][awox.mesh:261]: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
[21:35:58][D][awox.mesh:133]: No devices found to connect to
[21:36:02][D][awox.mesh:290]: Clear RSSI for A4:C1:38:59:F8:FC [0] not found the last 20 seconds
[21:36:03][D][awox.mesh:236]: Total devices: 2
[21:36:03][D][awox.mesh:238]: Available device A4:C1:38:6A:57:93 [0] => rssi: -60
[21:36:03][D][awox.mesh:238]: Available device A4:C1:38:59:F8:FC [0] => rssi: -9999
[21:36:04][D][awox.mesh:261]: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
[21:36:04][D][awox.mesh:133]: No devices found to connect to
[21:36:04][D][awox.mesh:290]: Clear RSSI for A4:C1:38:6A:57:93 [0] not found the last 20 seconds
[21:36:09][D][awox.mesh:236]: Total devices: 2
[21:36:09][D][awox.mesh:238]: Available device A4:C1:38:6A:57:93 [0] => rssi: -9999
[21:36:09][D][awox.mesh:238]: Available device A4:C1:38:59:F8:FC [0] => rssi: -9999
[21:36:09][D][awox.mesh:261]: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
[21:36:09][D][awox.mesh:133]: No devices found to connect to
[21:36:13][D][awox.mesh:236]: Total devices: 2
[21:36:13][D][awox.mesh:238]: Available device A4:C1:38:6A:57:93 [0] => rssi: -9999
[21:36:13][D][awox.mesh:238]: Available device A4:C1:38:59:F8:FC [0] => rssi: -9999
[21:36:13][D][awox.mesh:261]: Currently 0 mesh devices reachable through active connections (0 currently known and 0 fully recognized)
[21:36:13][D][awox.mesh:133]: No devices found to connect to

Very interesting for me is that the rssi is qite ok at the beginning but change to -9999 and never chane again.
Never get any cange in the log after rssi: -9999 regardles how long I do logging

By now it seems it can only be solved by getting back one year ore more.

Greatings from Austria, have a nice hollyday.

fsaris, 2025-07-29:
> By now it seems it can only be solved by getting back one year ore more.

Why one year back? 2025.6.* should work

fsaris, 2025-08-05:
Issue is caused/related to https://github.com/esphome/esphome/pull/9089 The `mesh_connection::set_state()` needs to call `BLEClientBase::set_state()` to ensure the internal loop is enabled again so the connection can be set and after that the BLE scanner can proceed again to check for RSSI values.

fsaris, 2025-08-05:
@Dillton main should work now

Bushrider, 2025-08-05:
I can confirm "main" is working now, thanks a lot !

WMax77, 2025-08-05:
You'r great @fsaris !!
I can confirm it is working now.
Btw. I got also the warnigs during install, seems the don't matter. Install was ok, it's working now!

Thanks a lot.

spider7611, 2025-08-06:
> [@spider7611](https://github.com/spider7611) did you try to clean the build files? Can you share your config?

Yes, always clean it before update or install.
Here is my config:
`esphome:
  name: nappaliesp32

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:
  level: WARN

#esp32_ble_tracker:

#bluetooth_proxy:
#  active: true
# Enable Home Assistant API
# api:

ota:
  - platform: esphome
    password: "e101bc4d9ee2203391662c67bed78faa"

wifi:
  networks:
  - ssid: !secret wifi_ssid2
    password: !secret wifi_password2    
    priority: 1
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
    priority: 0    

  manual_ip:
      static_ip: 192.168.1.51
      gateway: 192.168.1.1
      subnet: 255.255.255.0

mqtt:
  broker: 192.168.1.242

sensor:
  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:17:50:0F"
#    bindkey: "eef418daf699a0c188f3bfd17e4565d9"
    temperature:
      name: "Nappali Temperature"
      accuracy_decimals: 1
    humidity:
      name: "Nappali Humidity"
      accuracy_decimals: 0
    battery_level:
      name: "Nappali Battery Level"

  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:A7:FA:55"
#    bindkey: "eef418daf699a0c188f3bfd17e4565d9"
    temperature:
      name: "Furdoszoba Temperature"
      accuracy_decimals: 1
    humidity:
      name: "Furdoszoba Humidity"
      accuracy_decimals: 0
    battery_level:
      name: "Furdoszoba Battery Level"

  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:52:57:24"
#    bindkey: "eec11478fe41eeab0b02bad554efe21f"
    temperature:
      name: "Haloszoba TemperatureBL"
      accuracy_decimals: 1
    humidity:
      name: "Haloszoba HumidityBL"
      accuracy_decimals: 0
    battery_level:
      name: "Haloszoba Battery LevelBL"  
      
text_sensor:

  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: NappaliESP32 IP Address
    ssid:
      name: SSID

external_components:
  - source:
      type: git
      url: https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub
      ref: main
  #  ref: dev
  # - source:
  #     type: local
  #     path: components

esp32_ble_tracker:
  scan_parameters:
    active: false
    interval: 
      minutes: 1

awox_mesh:
  mesh_name: !secret mesh_name
  mesh_password: !secret mesh_password

  allowed_mac_addresses:
    - A4:C1:38:14:71:EB
    - A4:C1:38:3E:5E:F7

  device_info:
    # Example device type, see device info in HomeAssistant or MQTT message to find the 'product_id'
    product_id: 0xA9
    device_type: RGB
    name: EGLO RGB+TW
    model: EGLO-RGB-TW-IPSU
    manufacturer: EGLO`