fsaris/EspHome-AwoX-BLE-mesh-hub #127: Issue on compiling 'SEARCHING' is not a member of 'esphome::esp32_ble_tracker::ClientState'

ThePatzen, 2025-09-17:
Hi!

I get a strange error on compiling:

``In file included from src/esphome/components/awox_mesh/mesh_connection.cpp:5:
src/esphome/components/awox_mesh/mesh_connection.h: In member function 'virtual void esphome::awox_mesh::MeshConnection::set_state(esphome::esp32_ble_tracker::ClientState)':
src/esphome/components/awox_mesh/mesh_connection.h:121:44: error: 'SEARCHING' is not a member of 'esphome::esp32_ble_tracker::ClientState'
  121 |       case esp32_ble_tracker::ClientState::SEARCHING:
      |                                            ^~~~~~~~~
src/esphome/components/awox_mesh/mesh_connection.cpp: In member function 'void esphome::awox_mesh::MeshConnection::connect_to(esphome::awox_mesh::FoundDevice*)':
src/esphome/components/awox_mesh/mesh_connection.cpp:53:51: error: 'SEARCHING' is not a member of 'esphome::esp32_ble_tracker::ClientState'
   53 |   this->set_state(esp32_ble_tracker::ClientState::SEARCHING);
      |                                                   ^~~~~~~~~
In file included from src/esphome/components/awox_mesh/awox_mesh_mqtt.h:9,
                 from src/esphome/components/awox_mesh/awox_mesh.h:15,
                 from src/esphome/components/awox_mesh/awox_mesh.cpp:7:
src/esphome/components/awox_mesh/mesh_connection.h: In member function 'virtual void esphome::awox_mesh::MeshConnection::set_state(esphome::esp32_ble_tracker::ClientState)':
src/esphome/components/awox_mesh/mesh_connection.h:121:44: error: 'SEARCHING' is not a member of 'esphome::esp32_ble_tracker::ClientState'
  121 |       case esp32_ble_tracker::ClientState::SEARCHING:
      |                                            ^~~~~~~~~
In file included from src/esphome/components/awox_mesh/awox_mesh_mqtt.h:9,
                 from src/esphome/components/awox_mesh/awox_mesh_mqtt.cpp:14:
src/esphome/components/awox_mesh/mesh_connection.h: In member function 'virtual void esphome::awox_mesh::MeshConnection::set_state(esphome::esp32_ble_tracker::ClientState)':
src/esphome/components/awox_mesh/mesh_connection.h:121:44: error: 'SEARCHING' is not a member of 'esphome::esp32_ble_tracker::ClientState'
  121 |       case esp32_ble_tracker::ClientState::SEARCHING:
      |                                            ^~~~~~~~~
Compiling .pioenvs/esphome-web-26d444/src/esphome/components/esp32/gpio.cpp.o
src/esphome/components/awox_mesh/awox_mesh_mqtt.cpp: In lambda function:
src/esphome/components/awox_mesh/awox_mesh_mqtt.cpp:90:58: warning: 'ArduinoJson::V742HB22::JsonObject ArduinoJson::V742HB22::JsonObject::createNestedObject(const TString&) [with TString = std::__cxx11::basic_string<char>]' is deprecated: use obj[key].to<JsonObject>() instead [-Wdeprecated-declarations]
   90 |           JsonObject connection = root.createNestedObject("connection_" + std::to_string(i));
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from .piolibdeps/esphome-web-26d444/ArduinoJson/src/ArduinoJson.hpp:39,
                 from .piolibdeps/esphome-web-26d444/ArduinoJson/src/ArduinoJson.h:9,
                 from src/esphome/components/json/json_util.h:11,
                 from src/esphome/core/string_ref.h:11,
                 from src/esphome/core/entity_base.h:5,
                 from src/esphome/components/mqtt/mqtt_component.h:10,
                 from src/esphome/components/awox_mesh/awox_mesh_mqtt.cpp:10:
.piolibdeps/esphome-web-26d444/ArduinoJson/src/ArduinoJson/Object/JsonObject.hpp:218:14: note: declared here
  218 |   JsonObject createNestedObject(const TString& key) {
      |              ^~~~~~~~~~~~~~~~~~
src/esphome/components/awox_mesh/awox_mesh_mqtt.cpp: In lambda function:
src/esphome/components/awox_mesh/awox_mesh_mqtt.cpp:178:53: warning: 'ArduinoJson::V742HB22::JsonObject ArduinoJson::V742HB22::JsonObject::createNestedObject(TChar*) [with TChar = const char]' is deprecated: use obj[key].to<JsonObject>() instead [-Wdeprecated-declarations]
  178 |           JsonObject color = root.createNestedObject("color");
      |                              ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
.piolibdeps/esphome-web-26d444/ArduinoJson/src/ArduinoJson/Object/JsonObject.hpp:211:14: note: declared here
  211 |   JsonObject createNestedObject(TChar* key) {
      |              ^~~~~~~~~~~~~~~~~~
src/esphome/components/awox_mesh/awox_mesh_mqtt.cpp: In lambda function:
src/esphome/components/awox_mesh/awox_mesh_mqtt.cpp:212:59: warning: 'ArduinoJson::V742HB22::JsonObject ArduinoJson::V742HB22::JsonObject::createNestedObject(TChar*) [with TChar = const char]' is deprecated: use obj[key].to<JsonObject>() instead [-Wdeprecated-declarations]
  212 |           JsonObject device_info = root.createNestedObject(MQTT_DEVICE);
      |                                    ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
.piolibdeps/esphome-web-26d444/ArduinoJson/src/ArduinoJson/Object/JsonObject.hpp:211:14: note: declared here
  211 |   JsonObject createNestedObject(TChar* key) {
      |              ^~~~~~~~~~~~~~~~~~
src/esphome/components/awox_mesh/awox_mesh_mqtt.cpp: In lambda function:
src/esphome/components/awox_mesh/awox_mesh_mqtt.cpp:234:59: warning: 'ArduinoJson::V742HB22::JsonObject ArduinoJson::V742HB22::JsonObject::createNestedObject(TChar*) [with TChar = const char]' is deprecated: use obj[key].to<JsonObject>() instead [-Wdeprecated-declarations]
  234 |           JsonObject device_info = root.createNestedObject(MQTT_DEVICE);
      |                                    ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
.piolibdeps/esphome-web-26d444/ArduinoJson/src/ArduinoJson/Object/JsonObject.hpp:211:14: note: declared here
  211 |   JsonObject createNestedObject(TChar* key) {
      |              ^~~~~~~~~~~~~~~~~~
src/esphome/components/awox_mesh/awox_mesh_mqtt.cpp: In lambda function:
src/esphome/components/awox_mesh/awox_mesh_mqtt.cpp:256:59: warning: 'ArduinoJson::V742HB22::JsonObject ArduinoJson::V742HB22::JsonObject::createNestedObject(TChar*) [with TChar = const char]' is deprecated: use obj[key].to<JsonObject>() instead [-Wdeprecated-declarations]
  256 |           JsonObject device_info = root.createNestedObject(MQTT_DEVICE);
      |                                    ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
.piolibdeps/esphome-web-26d444/ArduinoJson/src/ArduinoJson/Object/JsonObject.hpp:211:14: note: declared here
  211 |   JsonObject createNestedObject(TChar* key) {
      |              ^~~~~~~~~~~~~~~~~~
src/esphome/components/awox_mesh/awox_mesh_mqtt.cpp: In lambda function:
[... 151 lines cut ...]
      |              ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
.piolibdeps/esphome-web-26d444/ArduinoJson/src/ArduinoJson/Object/JsonObject.hpp:168:8: note: declared here
  168 |   bool containsKey(const TString& key) const {
      |        ^~~~~~~~~~~
src/esphome/components/awox_mesh/awox_mesh_mqtt.cpp:585:23: warning: 'bool ArduinoJson::V742HB22::JsonObject::containsKey(const TString&) const [with TString = char [11]; typename ArduinoJson::V742HB22::detail::enable_if<ArduinoJson::V742HB22::detail::IsString<TString>::value, int>::type <anonymous> = 0]' is deprecated: use obj[key].is<T>() instead [-Wdeprecated-declarations]
  585 |   if (root.containsKey("color_temp")) {
      |       ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
.piolibdeps/esphome-web-26d444/ArduinoJson/src/ArduinoJson/Object/JsonObject.hpp:168:8: note: declared here
  168 |   bool containsKey(const TString& key) const {
      |        ^~~~~~~~~~~
src/esphome/components/awox_mesh/awox_mesh_mqtt.cpp:597:23: warning: 'bool ArduinoJson::V742HB22::JsonObject::containsKey(const TString&) const [with TString = char [7]; typename ArduinoJson::V742HB22::detail::enable_if<ArduinoJson::V742HB22::detail::IsString<TString>::value, int>::type <anonymous> = 0]' is deprecated: use obj[key].is<T>() instead [-Wdeprecated-declarations]
  597 |   if (root.containsKey("effect")) {
      |       ~~~~~~~~~~~~~~~~^~~~~~~~~~
.piolibdeps/esphome-web-26d444/ArduinoJson/src/ArduinoJson/Object/JsonObject.hpp:168:8: note: declared here
  168 |   bool containsKey(const TString& key) const {
      |        ^~~~~~~~~~~
src/esphome/components/awox_mesh/awox_mesh_mqtt.cpp:617:23: warning: 'bool ArduinoJson::V742HB22::JsonObject::containsKey(const TString&) const [with TString = char [6]; typename ArduinoJson::V742HB22::detail::enable_if<ArduinoJson::V742HB22::detail::IsString<TString>::value, int>::type <anonymous> = 0]' is deprecated: use obj[key].is<T>() instead [-Wdeprecated-declarations]
[... cut ...]
      |       ~~~~~~~~~~~~~~~~^~~~~~~~~
.piolibdeps/esphome-web-26d444/ArduinoJson/src/ArduinoJson/Object/JsonObject.hpp:168:8: note: declared here
  168 |   bool containsKey(const TString& key) const {
      |        ^~~~~~~~~~~
*** [.pioenvs/esphome-web-26d444/src/esphome/components/awox_mesh/mesh_connection.cpp.o] Error 1
*** [.pioenvs/esphome-web-26d444/src/esphome/components/awox_mesh/awox_mesh.cpp.o] Error 1
*** [.pioenvs/esphome-web-26d444/src/esphome/components/awox_mesh/awox_mesh_mqtt.cpp.o] Error 1``

fsaris, 2025-09-17:
#128 is a attempt to fix the issue. But wasn't able yet to test is.

Dillton, 2025-09-18:
I will try today when I get home. 😊

hosebei, 2025-09-18:
The issue-127 branch compiles on my configuration. the device boots up normally and works as far as I can consider.

hosebei, 2025-09-18:
I have to take back my "it works". it seems to not connect to the bulbs. I tested the wrong method I now have seen. sorry for the false information.

Dillton, 2025-09-18:
Evening guys. Sorry I am today totally stuck at work so won't be able to test tonight. But I see I don't have to if I don't want to break all lights based on @hosebei feedback.

PR #128 (2025-09-20): Make compatible with Esphome 2025.9
Resolves #127
commit 2a4440ac (2025-09-20): Merge pull request #128 from fsaris/issue-127

Make compatible with Esphome 2025.9

Resolves #127
closed 2025-09-20 by fsaris